espressif/arduino-esp32 #8479: External flash

savagerex, 2023-08-02:
### Board

ESP32 S3

### Device Description

External Flash - W25Q256JV

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

i use these lib, but these lib have problem.
these lib as below.

https://github.com/PaulStoffregen/SerialFlash
https://github.com/adafruit/Adafruit_SPIFlash/tree/master

Can ESP team do lib for external flash(W25Q256JV)???

### Sketch

```cpp
NONE
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-02:
arduino-esp32 uses the spi flash libraries from [esp-idf](https://github.com/espressif/esp-idf/tree/master/components/spi_flash).  If you need support for a particular chip, you should [ask upstream](https://github.com/espressif/esp-idf/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc).

SuGlider, 2023-08-02:
@savagerex - I remember that your project uses the ESP32-S3-Mini M0N8 with 8MB Flash, no PSRAM.

One possible way to solve it, would be to use EEPROM Arduino Library within the S3 module flash.
It uses NVS IDF functions. It may be possible to create a separated NVS partition with, for instance, 4MB, and modify the EEPROM library to use such partition.

By this way the Winbond external module wouldn't be necessary anymore, but it would use the S3 module flash. 
It may also be more cost effective to use a S3-Mini M0N16 instead and get 16MB flash for everything the project may need. 

Would it work for you?

savagerex, 2023-08-03:
i know if i esp32 s3 of flash.
if i use "SPIFFS" lib.
i have tested it, it's no problem.

but we still need external flash. because external flash(W25Q256JV) is 32M byte.

ESP32 S3 of flash only have 16MB. it's not enough.

we need external flash to store data.

SuGlider, 2023-08-03:
In that case, what is the issue that you have when using these libraries: 
SerialFlash from Paul and the Adafruit_SPIFlash?
Could you describe it and give me some examples about how to reproduce the issue in each one of them?

SuGlider, 2023-08-03:
I'll also need more details about how the SPI Flash chip is connected to the ESP32-S3 (connected pins).

savagerex, 2023-08-04:
@SuGlider 
below is i use lib ( https://github.com/PaulStoffregen/SerialFlash).

External Flash total is 32 M byte.
Everytime i creat file 2M. after 16 times, Write Flash Full. there is error ( "There was an error opening the file for writing")

below is the sample code.
******************************************

#include "SPI.h"
#include <SerialFlash.h>

#define SensorPowerPin 17

#define MOSI              35
#define MISO              37
#define SCK               36
#define FCSPIN            34
//#define CSPIN           21  // Arduino 101 built-in SPI Flash

#define SPI_FREQUENCY           40 * 1000000    //10

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  3        //30
RTC_DATA_ATTR int bootCount = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(SensorPowerPin, OUTPUT);
  digitalWrite(SensorPowerPin, HIGH);
  delay(15);
  
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  bootCount++;

    SPI.begin(SCK, MISO, MOSI, FCSPIN);
    //SPI2.begin(SCK, MISO, MOSI);
    SPI.setDataMode(SPI_MODE0);
    SPI.setBitOrder(MSBFIRST);
    SPI.setFrequency(SPI_FREQUENCY);

    if (!SerialFlash.begin(SPI , FCSPIN)) {
    while (1) {
      Serial.println(F("Unable to access SPI Flash chip"));
      delay(1000);
       }
     }
      //We start by formatting the flash...
      uint8_t id[5];
      SerialFlash.readID(id);
      Serial.printf("ID: %02X %02X %02X\n", id[0], id[1], id[2]);
      delay(500);
      Serial.print("bootCount = ");
      Serial.println(bootCount);

      delay(1000);
      Serial.println("Flash Ready");

      xTaskCreatePinnedToCore(taskTwo, "TaskTwo", 30000, NULL, 2, NULL, 1);

}

void taskTwo( void * parameter ) {
   disableCore1WDT();

   SerialFlashFile flashFile;

   while(SerialFlash.createErasable("123.txt", 2000000))  //2000000
   {
       Serial.println("Flash Creat Error");
       delay(2000);
   }
   
   flashFile = SerialFlash.open("123.txt");

   if (!flashFile) {
     Serial.println("There was an error opening the file for writing");    //when when flash is full, there is error.
     while(1)
     {
        Serial.print("Boot = ");
        Serial.println(bootCount);
        delay(3000);
     }
     return;
   }

   for (;;) {
            Serial.println("DelayTime : 1000");
            delay(1000);
            flashFile.close();

            flashFile = SerialFlash.open("123.txt");
            flashFile.erase();
            delay(500);
            SerialFlash.remove("123.txt");

            Serial.println("Task2 Going to sleep now");
            delay(100);
            esp_deep_sleep_start();
   }
   vTaskDelete( NULL );
   
}

void loop() {
  // put your main code here, to run repeatedly:

}

SuGlider, 2023-08-04:
@savagerex 
A possible alternative could be to write the `ext_flash_fatfs` IDF example in order to make it compile within ESP32 Arduino.

I have created a "translation" from the IDF example to Arduino 2.0.9 (see below), please try it.
This example is in https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/ext_flash_fatfs
This IDF example has been tested with Winbond W25Q32 SPI Flash chip.

It uses the default SPI VSPI pins. Please refer to the readme file for more information about changing it.
https://github.com/espressif/esp-idf/blob/release/v4.4/examples/storage/ext_flash_fatfs/README.md
 
``` cpp
/* Example of FAT filesystem on external Flash.
   This example code is in the Public Domain (or CC0 licensed, at your option.)

 

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.

 

   This sample shows how to store files inside a FAT filesystem.
   FAT filesystem is stored in a partition inside SPI flash, using the
   flash wear levelling library.
*/

 

#include "esp_flash.h"
#include "esp_flash_spi_init.h"
#include "esp_vfs_fat.h"

 

static const char *TAG = "example";

 

// Handle of the wear levelling library instance
static wl_handle_t s_wl_handle = WL_INVALID_HANDLE;

 

// Mount path for the partition
const char *base_path = "/extflash";

 

static esp_flash_t* example_init_ext_flash(void);
static const esp_partition_t* example_add_partition(esp_flash_t* ext_flash, const char* partition_label);
static void example_list_data_partitions(void);
static bool example_mount_fatfs(const char* partition_label);
static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes);

 

void setup(){
  my_app_main();
}

 

void loop() {
}

 

void my_app_main(void)
{
[... 171 lines cut ...]
    }
    return true;
}

 

static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes)
{
    FATFS *fs;
    size_t free_clusters;
    int res = f_getfree("0:", &free_clusters, &fs);
    assert(res == FR_OK);
    size_t total_sectors = (fs->n_fatent - 2) * fs->csize;
    size_t free_sectors = free_clusters * fs->csize;

 
    // assuming the total size is < 4GiB, should be true for SPI Flash
    if (out_total_bytes != NULL) {
        *out_total_bytes = total_sectors * fs->ssize;
    }
    if (out_free_bytes != NULL) {
        *out_free_bytes = free_sectors * fs->ssize;
    }
}
```

lbernstone, 2023-08-04:
Note that once you have a partition labeled, you can mount it with the arduino tools by providing that partition label to the begin call.
https://github.com/espressif/arduino-esp32/blob/master/libraries/FFat/src/FFat.h#L31
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/src/SPIFFS.h#L27

savagerex, 2023-08-04:
@SuGlider 

Question 1
if i write flash , as below.

flashFile.write(DataBufWrite, 120); 

how to modify in this sample code from IDF ???

Question 2
if i read flash , as below.

flashFile.read(DataBufRead, 40500);

how to modify in this sample code from IDF ???

Question 3

how to delete the file ???

savagerex, 2023-08-09:
@SuGlider 

why i , Serial.println(DataBufRead[100]);     
the DataBufRead[100] is 0.
it should be 1.

below is code.
*************************

#include "esp_flash.h"
#include "esp_flash_spi_init.h"
#include "esp_vfs_fat.h"

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  3        //30
RTC_DATA_ATTR int bootCount = 0;

static const char *TAG = "example";

// Handle of the wear levelling library instance
static wl_handle_t s_wl_handle = WL_INVALID_HANDLE;

// Mount path for the partition
const char *base_path = "/extflash";

static esp_flash_t* example_init_ext_flash(void);
static const esp_partition_t* example_add_partition(esp_flash_t* ext_flash, const char* partition_label);
static void example_list_data_partitions(void);
static bool example_mount_fatfs(const char* partition_label);
static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes);

uint8_t* DataBufWrite;  
uint8_t* DataBufRead;

 

void setup(){
  Serial.begin(115200);

  DataBufWrite = (uint8_t*)ps_calloc(1000000, sizeof(uint8_t));   //6 * 27000 * 3 = 486000
  DataBufRead = (uint8_t*)ps_calloc(1000000, sizeof(uint8_t));

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  bootCount++;
  
  my_app_main();

  Serial.print("bootCount = ");
  Serial.println(bootCount);

  xTaskCreatePinnedToCore(taskTwo, "TaskTwo", 30000, NULL, 2, NULL, 1);
}

void taskTwo( void * parameter ) {
   disableCore1WDT();

    for (;;) {
               LoopFlash();
               Serial.println("Task2 Going to sleep now");
               delay(1000);
               esp_deep_sleep_start();
    }
    vTaskDelete( NULL );
}

 

void loop() {
}

[... 218 lines cut ...]
        return false;
    }
    return true;
}

 

static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes)
{
    FATFS *fs;
    size_t free_clusters;
    int res = f_getfree("0:", &free_clusters, &fs);
    assert(res == FR_OK);
    size_t total_sectors = (fs->n_fatent - 2) * fs->csize;
    size_t free_sectors = free_clusters * fs->csize;

 
    // assuming the total size is < 4GiB, should be true for SPI Flash
    if (out_total_bytes != NULL) {
        *out_total_bytes = total_sectors * fs->ssize;
    }
    if (out_free_bytes != NULL) {
        *out_free_bytes = free_sectors * fs->ssize;
    }
}

savagerex, 2023-08-10:
OK, Thanks

savagerex, 2023-08-11:
@SuGlider 

i use ESP32 WROVER, but my flash is connect HSPI.
i modify the code as below.
but it can not work.

there is the debug message.
********************************************
E (593) memspi: no response

bootCount = 2
Task2 Going to sleep now
E (593) memspi: no response

bootCount = 3
Task2 Going to sleep now
E (593) memspi: no response

bootCount = 4
Task2 Going to sleep now

*************************************************************
#include "esp_flash.h"
#include "esp_flash_spi_init.h"
#include "esp_vfs_fat.h"

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  3        //30
RTC_DATA_ATTR int bootCount = 0;

static const char *TAG = "example";

// Handle of the wear levelling library instance
static wl_handle_t s_wl_handle = WL_INVALID_HANDLE;

// Mount path for the partition
const char *base_path = "/extflash";

static esp_flash_t* example_init_ext_flash(void);
static const esp_partition_t* example_add_partition(esp_flash_t* ext_flash, const char* partition_label);
static void example_list_data_partitions(void);
static bool example_mount_fatfs(const char* partition_label);
static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes);

uint8_t* DataBufWrite;
uint8_t* DataBufRead;

void setup() {
  Serial.begin(115200);

  DataBufWrite = (uint8_t*)ps_calloc(1000000, sizeof(uint8_t));   //6 * 27000 * 3 = 486000
  DataBufRead = (uint8_t*)ps_calloc(1000000, sizeof(uint8_t));

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  bootCount++;

  my_app_main();

  Serial.print("bootCount = ");
  Serial.println(bootCount);

  xTaskCreatePinnedToCore(taskTwo, "TaskTwo", 30000, NULL, 2, NULL, 1);
}

void taskTwo( void * parameter ) {
  disableCore1WDT();

  for (;;) {
    LoopFlash();
    Serial.println("Task2 Going to sleep now");
[... 188 lines cut ...]
  esp_err_t err = esp_vfs_fat_spiflash_mount(base_path, partition_label, &mount_config, &s_wl_handle);
  if (err != ESP_OK) {
    ESP_LOGE(TAG, "Failed to mount FATFS (%s)", esp_err_to_name(err));
    return false;
  }
  return true;
}

static void example_get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes)
{
  FATFS *fs;
  size_t free_clusters;
  int res = f_getfree("0:", &free_clusters, &fs);
  assert(res == FR_OK);
  size_t total_sectors = (fs->n_fatent - 2) * fs->csize;
  size_t free_sectors = free_clusters * fs->csize;

  // assuming the total size is < 4GiB, should be true for SPI Flash
  if (out_total_bytes != NULL) {
    *out_total_bytes = total_sectors * fs->ssize;
  }
  if (out_free_bytes != NULL) {
    *out_free_bytes = free_sectors * fs->ssize;
  }
}

savagerex, 2023-08-14:
OK!!  Thanks

VojtechBartoska, 2023-08-18:
Can we consider this as solved and close this ticket?

Endorfin35, 2023-09-12:
@SuGlider 
> IDF örneğinden Arduino 2.0.9'a bir "çeviri" oluşturuldu (aşağıya bakın), lütfen deneyin. [Bu örnek https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/ext_flash_fatfs](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/ext_flash_fatfs)

I tried this example successfully. I want to use the functions in the Fatfs library when writing my own functions related to filing.

https://github.com/espressif/arduino-esp32/blob/master/libraries/FFat/examples/FFat_Test/FFat_Test.ino

However, I'm a little confused. In the example you gave, we had already installed the Fat system. Is it OK to add it back to the Fatfs.h library? Or will Fatfs.h only be effective on internal flash?

lbernstone, 2023-09-12:
Please don't hijack threads.
listdir() is not a built in function.  AFAIK, the test example should work just fine with an external flash mounted into arduino FS.

Endorfin35, 2023-09-13:
Sorry for the confusion.

The example works just fine. I have no problem with this.

I want to ask something else. When I do not connect the Wp and Hold pins, the flash works without any problems. I want to use the io connections defined on these pins for another purpose. How can I make the code release these pins?

GPIO22WP
GPIO21HD