espressif/arduino-esp32 #8480: ESP32 S2 I2C Bus speed changes unexpectedly from 400KHz to 5KHz

tchilton, 2023-08-02:
### Board

ESP32 S2

### Device Description

Vanilla ESP32 S2 (DIL header) development board, although problem originally found in custom PCB with various I2C devices including an I2C bus MUX (PCA9547) and 4 remote boards, which may or may not be installed. 

The problem can be demonstrated just on the vanilla ESP32 S2 development board with one I2C device present (the bus mux). I expect it will be the same for an empty I2C bus.

### Hardware Configuration

SDA=8, SCL=9, pull up resistors on I2C bus
Logic analyser on I2C bus.

### Version

v2.0.10

### IDE Name

Arduino IDE 2.1.1

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

On an ESP32 S2 device, when attempting to write to an I2C device via the Wire library to a device which is not currently present on the bus, after 10 such transactions, the bus speed changes unexpectedly from the configured 400KHz to 5KHz. It does not matter if the 10 transactions are sequential to each other, or interspersed with other valid I2C traffic.

Additionally, 5KHz is not a standard I2C bus speed and its also far too slow for any effective communications

Expected behaviour is that The I2C bus should remain at the configure bus speed set via Wire.setClock() for all I2C transactions.

### Sketch

```cpp
// Demo code to show uncommanded I2C bus data rate change on ESP32 S2 when using I2C and attempting to talk to 
// a device which is not currently on the bus
// Initial cycles are at 400KHz as expected, however this changes to 5KHz after a short duration

// Bug experienced on ESP32 S2 chip. Works as expected on ESP£2 DevKit V1 and Atmel MEGA2560 boards

// Change happens after exactly 10 attempts to write to non-existent devices
// It doesn't matter if other successful I2C commands are issued during, or after the non-existent device writes.

// Attempting to reset the bus speed has no impact, unless you select a different bus speed (i,e, 1M), then switch back to the desired speed (400K)

// For I2C return values, see https://www.arduino.cc/reference/en/language/functions/communication/wire/endtransmission/

#include <Arduino.h>
#include <Wire.h>

// I2C pins
#define I2C_SDA                          8      // I2C Data pin
#define I2C_SCL                          9      // I2C Clock pin

#define I2C_CLKRATE_400K            400000      // Speed of I2C bus 400KHz
#define I2C_CLKRATE_1M             1000000      // Speed of I2C bus 1MHz
#define I2C_MAX_RETRIES                  2      // Number of times we will retry a failed I2C write instruction
#define TEST_CYCLES                     30      // How many test cycles to run

#define I2C_ID_TALK_TO                0x60      // The target device, which is not present
#define I2C_BUS_MUX                   0x71      // An 8 channel I2C bus mux - PCA9547, which is present

void write_mux(int);

void setup() {
  Serial.begin(115200);
  Serial.println("Starting up");
  
  if (! Wire.begin(I2C_SDA, I2C_SCL)) {         // I2C bus setup
    Serial.println("Failed to configure I2C bus");                
  }
   Wire.setClock(I2C_CLKRATE_400K);             // Set the I2C bus rate
}

void loop() {
  uint8_t  i2cError,i2cRetry;

  Serial.println("Main");
  for (int i=0 ; i< TEST_CYCLES ; i++) {
    i2cRetry = 0;
    do {
      Wire.beginTransmission(I2C_ID_TALK_TO);   
      Wire.write(0x00);                         // Some payload data - will never get sent as the device is not present
      Wire.write(0x01);
      i2cError = Wire.endTransmission();        // An error here indicates device is not present / bus congestion, etc.
      if (i2cError == 2) break;                 // Device is not currently present, so no point in retrying
    } while (i2cError && i2cRetry++ < I2C_MAX_RETRIES);   // Other errors, retry

    if (i2cError) {   
      Serial.print("I2C Error ");
      Serial.print(i2cError);
      Serial.print(" Retries= ");
      Serial.println(i2cRetry);
    }

    delayMicroseconds(10);    
    // Workaround to the problem - change bus speed to something else, then back to desired speed
    Wire.setClock(I2C_CLKRATE_1M);                 // Set the I2C bus rate
    Wire.setClock(I2C_CLKRATE_400K);                 // Try and reset clock rate

//    write_mux(i);                                  // Now try and talk to another I2C device - mixed comms
  }
       
  // Give a bit of time to the chip, just in case something is happening in the background
  yield();      
  delayMicroseconds(100);   

  // Workaround to the problem
  // Just resetting to 400K fails, but changing speed, then changing back, does reset the bus speed
  // Wire.setClock(I2C_CLKRATE_1M);                 // Set the I2C bus rate
  // Wire.setClock(I2C_CLKRATE_400K);               // Try and reset clock rate

  write_mux(0x01);                                  // After failure, try and talk to an existing device

  Serial.println("Test complete, press reset");
  while (1) {
  }
}

// Write to the I2C bus mux - PCA9547
void write_mux(int channel) {
  uint8_t  i2cError,i2cRetry;
  
  i2cRetry = 0;
  do {
    Wire.beginTransmission(I2C_BUS_MUX);
    Wire.write(channel & 0x07);               // Select the MUX channel
    i2cError = Wire.endTransmission();
    if (i2cError == 2) break;         
  } while (i2cError && i2cRetry++ < I2C_MAX_RETRIES);

  if (i2cError) {   
    Serial.print("Bus MUX I2C Error ");
    Serial.print(i2cError);
    Serial.print(" Retries= ");
    Serial.println(i2cRetry);
  }
}
```

### Debug Message

```plain
No debug output, device continues to operate correctly in other areas
```

### Other Steps to Reproduce

1. Configure IDE for ESP32 S2 device
2. Compile sample code
3. Observe bus speed change unexpectedly from 400KHz to 5KHz using a logic analyser or digital scope after 10 successful I2C transfers at the expected speed. An example capture of this is provided [I2C Bus Defect Saelae Capture.txt](https://github.com/espressif/arduino-esp32/files/12241615/I2C.Bus.Defect.Saelae.Capture.txt) rename this to a .sal file and open in Saelae logic 2 software (https://www.saleae.com/downloads/) . Problem is visible from 12ms into the capture

4. Optionally change target board and prove expected bus protocol on other chips - I've tested on both ESP32 DevKit 1 and Atmega 2560 boards, so looks like this is specific to ESP32 S2

5. Workaround, un-comment the two lines 76 and 77 to change the bus speed and repeat test.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-04:
@me-no-dev PTAL, thanks!

thorrak, 2023-09-06:
I was able to track this down to some kind of change that happened between v2.0.6 and v2.0.7 of the Arduino framework -- which coincidentally is also when the framework changed from ESP-IDF v4.4.3 to v4.4.4.

This issue still exists in v2.0.10 which is based on v4.4.5, as well as v2.0.12. 

The way this bug manifested on my side is in I2C communication becoming excruciatingly slow, blocking the controller from doing much else. I was able to resolve my issue via the workaround you cited:
```
  // Workaround to the problem
  // Just resetting to 400K fails, but changing speed, then changing back, does reset the bus speed
  // Wire.setClock(I2C_CLKRATE_1M);                 // Set the I2C bus rate
  // Wire.setClock(I2C_CLKRATE_400K);               // Try and reset clock rate
```

...but admittedly, this seems less than ideal if it can be avoided!

tchilton, 2023-09-06:
Thanks @thorrak. I wonder if its something to do with retries where comms has failed and as a result, buffers fill up - although new comms are still able to be sent.

The problem is that unless you have a logic analyser, you are not going to spot this problem, other than further congestion where there isn't the bus bandwidth you think you have.

Is there a change log of the IDF and Arduino frameworks that might allow us to break down a bit further where this problem may be ?

thorrak, 2023-09-06:
> Thanks @thorrak. I wonder if its something to do with retries where comms has failed and as a result, buffers fill up - although new comms are still able to be sent.
>
> The problem is that unless you [...]

Yeah -- for me, I was seeing a (VERY low bandwidth, I think!) LCD screen take forever to refresh in v2.0.7 when it worked fine in 2.0.6. I have access to an S2 with an attached hardware debugger, so I could track it down to this [specific line of code taking forever to run](https://github.com/espressif/arduino-esp32/blame/master/cores/esp32/esp32-hal-i2c.c#L176) but there's nothing there that indicates a specific cause. 

Assuming that the bug is in ESP-IDF and not Arduino, then the full changelog is [here](https://github.com/espressif/esp-idf/compare/v4.4.3...v4.4.4), with the release notes [here](https://github.com/espressif/esp-idf/releases/tag/v4.4.4).

tchilton, 2023-09-06:
Thanks for the info. I've done a quick compare of all I2C related topics and the related deltas to the code. There are only a couple and of those, nothing sticks out as a likely cause. (with my very limited knowledge of the underlying code). However, if you said it worked in the older versions, then we need to break this down as caused by IDF or caused by Arduino. I'd guess IDF is more likely as its closer to the hardware. Can we binary chop the list of changed items for I2C and see which ones / one is the likely cause ? I don't know how to do that.

astrogene1000, 2023-11-04:
Hello all, any forward progress on this?
I have an S2 talking to three targets addr's 14/15 and 16.
If any error is thrown the SCL drops to (as measured on logic analyzer) 833hz, starts at 100khz
In the attached successful communications seen to addr 14 but a failure to addr 15 and the clock drops down and never recovers.

Thanks!

Gene
[image]

tchilton, 2023-11-04:
Use the workaround I found. Set the bus speed to something else, then set it back to what you want, then try and do the I2C transaction on the bus. I've just got an i2cWorkaround function that does this until its resolved. See details above.

void i2cWorkaround() {
    Wire.setClock(I2C_CLKRATE_1M);                 // Set the I2C bus rate to something else
    Wire.setClock(I2C_CLKRATE_400K);                 // Try and reset clock rate to the speed you want
}

I have heard nothing on this defect being worked on in recent months.

astrogene1000, 2023-11-04:
Thanks tchilton for the workaround.

Pity all is good on WROOM's and S3's but trying to make use of a mini S2 readily available all with same pinouts because only need 4 pins total from the ESP when using I2C (SCL/SDA?GND?V3.3).

My s2 is in control of 8 stepper motors controlled by 3 different Pololu stepper drivers (4/2/2 motors each Z/X/Y), the app itself is under control of a windows application but is constantly talking to all targets regardless of the windows application.

I do not have bus integrity problems, I have a test bed with only one of the three targets available and I change the address on the target based on what axis I am working on.

This is all compounded by a long wire I2C in the actual system, with appropriate pullup;s and an LTC4311 sitting on SCL/SDA host side.

I2c errors are checked for every operation to the targets and a retry is done if I2C returns anything but OK so I will add your workaround and see how it goes!

Thanks again!

Gene

tchilton, 2023-11-04:
You are welcome and I'm glad someone else can confirm the bug I found :-)

The lack of presence of all 8 motor drivers is probably whats driving the fault for you, its the same for me. I have a design with 4 remote modules and some could be disconnected and others haven't been built yet.

I too only had problems on the ESP32-S2 so far. I'm going to upgrade to ESP32-S3's soon as I'm getting tight on FLASH space and want to add Bluetooth to my design, but the different footprint and pinout means new PCB's and a lot more time and cash for the new PCB's and components.

If you are running long distance on I2C, this might be a problem, its designed to be fairly short bus lengths. Take a look at the PCA9615 from NXP - https://www.nxp.com/docs/en/data-sheet/PCA9615.pdf, this is a differential bus extender. I'm using these in my design and they work well. Differential signalling would be good if you have motors involved, they are always electrically noisy, plus offcuts of network cable are always available.

I'm also using a PCA9547, which is an 8 channel I2C bus multiplexer, so I can fan out to multiple external modules with different I2C devices present. Each output channel drives into a PCA9615, one local on the main board and one remote in the far end module, this allows me to segregate all the busses and minimise risk if one has a problem. This approach might help for you too ?

astrogene1000, 2023-11-05:
Thanks again tchilton!

The workaround did 'solve' the issue!

My I2C run is less than 7 feet but will have the three targets and yes, all the motor noise at the targets. The motors do not run much, total travel with stepper leadscrews is maybe 4000 steps for Z and  3000 steps for x/y and only during adjustment phase, 99% of time will be no motion.

Plus only looking for 100k not 400k rate :-)

On the bench I have a single target/motor but about 25ft of CAT5E with SDA and GND on one twisted pair and SCL on a single wire of a separate pair. The code is basically self healing for errors, the retry helps some but every new talk to a target goes through the whole error checking process. So far have run 48 hours and target saw 2 errors and esp I2C error response saw more than 2.  Rough counting, there are about 3400 I2C xmit or rcv transactions/second constant traffic.
[image]

Gene

tchilton, 2023-11-05:
I'm glad it worked for you too. Its not a long term solution, but works OK in the mean-time until it gets fixed. 

If you want to check for successful I2C comms, have you seen this page 
https://www.arduino.cc/reference/en/language/functions/communication/wire/endtransmission/

Response code 2 is the one that says if the target is not present, since in the normal I2C comms, the client pulls the bus low to acknowledge the request, if its not there, clearly it doesn't do that.

If I get this response in my code, I don't bother trying any other comms to that device on that cycle in the code as that board is not present. In my system, boards can be hot plugged in / rebooted separately to the main controller in the head unit.

iharshadec, 2023-11-06:
I am not sure if my issue is exactly same but I was using I2C ADC chip and I was able to read the data correctly on ESP Arduino version 1.0.6 but when I updated to latest one 2.0.14 I see singal is not same as it should be.
Looks like something is changed in I2C speed or clock or something like that.

tchilton, 2023-11-07:
@iharshadec. Can you provide a bit more information please.

Which type of processor are you using ? 
What are you seeing for the signal when good vs bad / how are you measuring this, scope, logic analyser, etc ?
Do you have bus integrity problems (noise, long I2C bus, missing pull-up resistors etc)
Are you trying to talk to missing devices on the bus as that seems to be the trigger 
What response codes is the Arduino endTransmission() return ?

@thorrak mentioned that this came in on the 2.0.6 and its still not fixed, so it might be part of the problem you have. 

Have you tried the workaround to see if it resolves the issue for you ?

@me-no-dev , do you need any more info / do you know when this might be investigated / resolved ?

iharshadec, 2023-11-07:
@tchilton 
I am using ESP32-PICO-D4. I am getting scope tomorrow and I will post with more details.
there is no hardware issue as the same firmware if I downgrade my board manager to 1.0.4 or 1.0.6 it works perfectly.

So may not be exact this but I see something wrong with I2C here.

me-no-dev, 2023-11-07:
Current `Wire` implementation is a very light wrapper around ESP-IDF. Have you noticed any similar issue reports being reported there? As far as I can see, this issue sounds to be ESP32-S2 specific? Or did you also find the same problem elsewhere?

@iharshadec I2C driver changed from 1.0.x to 2.0.x. We used to have our own I2C driver and now we use the ESP-IDF one. With adding more chips, it makes no sense to support two separate I2C drivers.

astrogene1000, 2023-11-07:
I wonder if this is languishing because of 3.0.0 but it has been around for a long time if it does go back to 2.0.6 to 2.0.7 after Dec last year ?
[https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4](url)

It is annoying and agreed it must be deep in IDF as for me only the S2 variant is plagued by it.
The reset of frequency shown by tchilton above is functional.
I use it the same way, outer function checks and if error reset frequency and if still fails on retry ignores inner functions.
Inner functions succumb to the delay for one retry if it happens and gets reset in outer function.

iharshadec, 2023-11-07:
> @iharshadec I2C driver changed from 1.0.x to 2.0.x. We used to have our own I2C driver and now we use the ESP-IDF one. With adding more chips, it makes no sense to support two separate I2C drivers.

So,
does that mean need to change I2C in arduino library level? Is there any link you can share so I can change my I2C calls to ESP-IDF ones?

tchilton, 2023-11-07:
@iharshadec If you are using v2.x, then you already are and that's probably why the problem is happening. 
I must admit that I was thinking of doing the same test on native ESP-IDF, but I've not got that far yet due to time constraints.

If you want to go native ESP-IDF and not use the Arduino framework, then the following links will be handy.

The tech manual for the ESP32-S2 - https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf
The list of peripherals - https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32s2/api-reference/peripherals/index.html
The I2C peripheral - https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/i2c.html

Native ESP-IDF requires Visual Studio Code (free editor) and looks to be the best route, but needs some re-work to existing code. I'm working towards this but haven't made the switch fully yet. 

I'm currently using bits of ESP-IDF from within Platform IO (PIO), so there is a bit of re-learning to do. Its also worth investigating the FreeRTOS kernel in the ESP32's, this makes things a lot easier for more complex projects with its separate processes, timers and interrupt handling - all with no libraries. Obviously you are trading a level of lock-in to ESP32 platform by doing this.

Personal view from my couple of months using VS Code / PIO,  is a lot faster in terms of builds and provides far better control on configs (dev/prod/multiple upload methods - all within platformio.ini), plus JTAG support, debugging, separate JTAG download and serial console for debugging, etc.

iharshadec, 2023-11-07:
@tchilton 
No, I do not want to go away totally from Arduino IDE as there are many libraries I am using right now can not get converted into ESP-IDF easily.

I was just thinking as it's related to ADC Chip read and it's I2C Code is written my way like this.

```
	Wire.beginTransmission(tuAddr);
	Wire.write(0x02); // Points to Threshold Low
	Wire.write(0x00); // MSB
	Wire.write(0x00); // LSB
	Wire.endTransmission();

	Wire.beginTransmission(tuAddr);
	Wire.write(0x03); // Points to Threshold High
	Wire.write(0x80); // MSB
	Wire.write(0x00); // LSB
	Wire.endTransmission();
```

So if is there anything I can change here and this Wire use ESP-IDF I2C wrapper and not use Arduino one?
If this can fix my issue as I told earlier that when I move back to Board manager 1.0.6 with ESP32-PICO-D4 it has no issue but with latest one (2.0.14) something is messing up I2C which results into missing samples.

tchilton, 2023-11-07:
@iharshadec. I have the same challenge re libraries. However, when I looked in more detail, some are already present - TFT displays for example and I ended up far less dependant than I initially thought. Even switching to VS Code, PIO and keeping the Arduino framework within them makes for a much nicer and faster compiling development experience.

To move things forwards. First, you need to be checking the return value for Wire.endTransmission() to see what it tells you - see previous posts for info on that. We also still need to confirm that you have the same issue, per my queries earlier. A scope or logic analyser trace will help there. I'm guessing that the ESP32-PICO-D4 is newer than the ESP32-S2 - given its got the larger RAM of the ESP32-S3 ?

As to converting just one part to ESP-IDF, this is where I am right now, some bits are native Arduino, other bits are ESP-IDF (timers, interrupts, tasks, GPIO, Preferences, OTA, etc). I believe that I get a lot of FLASH back once I flip completely to ESP-IDF and the Arduino frameworks are no longer compiled into my binary. I'll be looking at this later in the year when I get more free time.

As to how to convert to ESP-IDF for just I2C, take a look at the last link I sent before (I2C peripheral), it shows the resource and the code to initialise it and the functions to control the bus. I would guess that the whole sketch should use one or the other methods (Arduino or ESP-IDF) so that you don't cause problems for the other framework / difficult to find issues.

To get going on ESP-IDF, you might want to start on something easier though, perhaps GPIO - see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html so that you see how the IDF works and can get the right include files in etc.

I will take a look at the v3.0 framework beta that @astrogene1000 mentioned and see what that does (probably in a VM so I don't screw up my environment) and then I'll take a crack at the ESP-IDF I2C and repeat the test and see what happens.

It would be good if others could confirm what they see on ESP32-S3 and other newer chips so that we can determine if this is only affecting ESP32-S2's or if its likely to be fixed in v3.0, etc.

iharshadec, 2023-11-07:
@tchilton 
ESP32-PICO-D4 is newer -- I am not sure but it's same as ESP32-D0WD just has inbuilt Flash so Single chip and still same as ESP32-D0WD

tchilton, 2023-11-08:
According to Wikipedia (https://en.wikipedia.org/wiki/ESP32), the ESP32-PICO-D4 has two cores, so it can't be based on the ESP32-S2 as that only has one, which suggests this is not impacting one CPU generation only.  Given that we have all been impacted by 2.x software, I'm going to guess that its a software bug and focus my efforts to narrow that down a bit.

For clarity, I've summarised the reports so far in this thread.
Confirmed Impact
- ESP32-S2 - 2 reports. Mine is a Saola-1 
- ESP32-PICO-D4

Not Impacted 
- ESP32-Devkit-1. Being older, I wonder if this was using 1.x / older ESP-IDF behind the scenes ?

@thorrak - what was the CPU you had problems with. You say you have access to an S2, but imply this is not the one you had the fault with.

@astrogene1000 - can you confirm that you tested on an ESP32-S3 ?

thorrak, 2023-11-08:
I apologize if there was any implication otherwise -- It was definitely an ESP32-S2 that I encountered issues with. I have no issues with the (original) ESP32.

tchilton, 2023-11-08:
@thorrak. No problem and thanks for clarifying. I've updated the table above with the info.

tchilton, 2023-11-08:
I've logged a new defect against 3.0.0 for the incorrect I2C bus speed in 3.0.0Alpha, its #8855, see https://github.com/espressif/arduino-esp32/issues/8855

astrogene1000, 2023-11-08:
@tchilton 
Yes, confirmed tested on ESP-S3 DevKitC-1 v1.6 and no CLK issues seen.
I also tried on a PICO-D4, no issues there either
And sorry about broken link to blog post
https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4

With an 80X reduction, this looks interesting, but same value back to 2.0.1
From i2c_ll.c
hardware\esp32\2.0.14\tools\sdk\esp32s2\include\hal\esp32s2\include\hal
#define I2C_LL_CLK_SRC_FREQ(src_clk)  (((src_clk) == I2C_SCLK_REF_TICK) ? **1000*1000 : 80*1000*1000);** // Another clock is APB clock
But then for S3
hardware\esp32\2.0.14\tools\sdk\esp32s3\include\hal\esp32s3\include\hal
#define I2C_LL_CLK_SRC_FREQ(src_clk)  (((src_clk) == I2C_SCLK_RTC) ? 8*1000*1000 : 40*1000*1000); // Another clock is XTAL clock
And legacy ESP32
hardware\esp32\2.0.14\tools\sdk\esp32\include\hal\esp32\include\hal
#define I2C_LL_CLK_SRC_FREQ(src_clk)  (80*1000*1000)

From S2 soc_caps.h
#define SOC_I2C_SUPPORT_REF_TICK   (1)
#define SOC_I2C_SUPPORT_APB        (1)

again i2c_ll.c from S2
static inline void i2c_ll_set_source_clk(i2c_dev_t *hw, i2c_sclk_t src_clk)
{
    hw->ctr.ref_always_on = (src_clk == I2C_SCLK_REF_TICK) ? 0 : 1;
}

And finally from
https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-reference/peripherals/i2c.html

I2C_SCLK_APB
I2C source clock from APB, 80M

I2C_SCLK_REF_TICK
I2C source clock from REF_TICK, 1M

tchilton, 2023-11-08:
Nice investigative work @astrogene1000 thank you. :-)

I think the code above has some italics that didn't make sense initially and I was struggling to see what they meant until I looked at the header files. Perhaps use the <> code option in the menu bar above next time to insert source code and stop GitHub  mangling the important info.

I agree with the 80:1 suggestion but I'm not figuring why the values are in the source code. @me-no-dev Please advise and see if this is an easy fix now that @astrogene1000 has done the heavy lifting here. Is this simply an overflow issue on whatever takes I2C_LL_CLK_SRC_FREQ ?

I've got a couple of ESP32-S3's on order at the moment,. I'll be able to test my main board with them and some adapter boards I made that do ESP32-S2 <-> ESP32-S3 conversions, it allows S2's to be used in S3 designed boards or S3's to be used in S2 designed boards.

I note that you say that the PICO-D4 works fine, yet @iharshadec  is having problems with the same board. Can you offer any advice on your differences that makes it work ?

Looks like we are collectively getting close to the source of the problem and I'm thinking its just a code issue in the defines above,  rather than a hardware issue at this point.

@me-no-dev - can we give this a little push to get to closure please. Any of us are happy to test updates.

astrogene1000, 2023-11-09:
@tchilton 
No joy here or maybe I did not hit the correct place(for sure!)
In esp32-hal-i2c.c they make a distinction for S3 so I extended it
`#if CONFIG_IDF_TARGET_ESP32S3
        if (clk == I2C_SCLK_RTC) { // RTC clock for s3 is unaccessable now.
            continue;
        }
#endif
`
And my poke and try adds no matter the core type:
` if (clk == I2C_CLK_LIMIT_REF_TICK) { 
            continue;
        }
        if (clk == I2C_CLK_LIMIT_XTAL) { 
            continue;
        }
` 

But alas no joy for the S2

I will re-load the PICO-D4 tomorrow and check but in my code the error checking i2c error code responses is heavy so the single retries may hide anything through my 25ft longwire.
 
Gene

astrogene1000, 2023-11-09:
Well my try at use of the <> code button got me what looks like the above
, plain text version sorry:
esp32-hal-i2c.c

#if CONFIG_IDF_TARGET_ESP32S3
        if (clk == I2C_SCLK_RTC) { // RTC clock for s3 is unaccessable now.
            continue;
        }
#endif

        if (clk == I2C_CLK_LIMIT_REF_TICK) { // ESJNK
            continue;
        }
        if (clk == I2C_CLK_LIMIT_XTAL) { // ESJNK
            continue;
        }

TD-er, 2023-11-09:
@astrogene1000 
When trying to include code, you can use 3x a backtick on a new line before and after your multiline code block.
And if you want to do it really fancy, you can also include `c++` right after the first 3 backticks.
```c++
#if CONFIG_IDF_TARGET_ESP32S3
    if (clk == I2C_SCLK_RTC) { // RTC clock for s3 is unaccessable now.
       continue;
    }
#endif

    if (clk == I2C_CLK_LIMIT_REF_TICK) { // ESJNK
        continue;
    }
    if (clk == I2C_CLK_LIMIT_XTAL) { // ESJNK
        continue;
    }
```
Like this:
[image]

me-no-dev, 2023-11-09:
@tchilton I am following the thread with interest and I am looking for a break that would give some direction towards resolving this. Source clocks are a good direction. If the clock is changed and dividers are not updated, then resulting frequency will be fraction of the original one, equal to the division of both clocks. 80x/40x difference would point to switching from APB/XCLK to REF_TICK. I can not explain what or why would the clock switch at this point, but nothing is lost. If issue persists on 3.0.0, then chances for getting it fixed are even better.

I am willing to overlook the PICO-D4 issue reported for a bit more, because if I2C was broken on ESP32 since 2.0.x, the internet would have exploded already. 95%+ of the used boards are ESP32 based.

tchilton, 2023-11-09:
@me-no-dev. I accept that I2C is very commonly used, but if people don't look at the bus or do not notice the unexpected speed change as their sketch is only using a low data rate, then how many would even know that there is a problem ?
The cases above, each person realised there was a problem and went digging and ended up here.

A bit more searching and it seems this this problem has been experienced by others, see this posting on the ESP32.com site - 
https://esp32.com/viewtopic.php?t=33934  Their analysis seems to show the same things we have determined the 80:1 change and a likely clock source change.  I've added a link there to this page.

V3.0.0 Alpha2 is still impacted, but now has two different bugs that impact speed, this one and another that increases it at boot to multi-megabit rates. The latter will cause a lot more problems as not all devices can go that fast and additionally, the workaround of updating the clock to something else and then back no longer works.  I'm guessing that for defect https://github.com/espressif/arduino-esp32/issues/8855, the clocks are not initialised correctly at the start, whereas with this problem, something is altering the clock source / divider when the platform is running.

I can see section 2.2 of the technical manual for the ESP32-S2 is the clock sources (link already posted above) and the only two clock sources are APB_CLK and REF_TICK which @astrogene1000 has already identified in the header files.

I'll have more of a read and see if we can read these clock registers and see what their values are at boot and when this problem occurs. A quick Google shows a macro called DPORT_REG_READ and there is a lot of info on the data structures for i2c_ll.h, so I'm wondering if I can read back register contents to show that its changed. I'll have a play.

me-no-dev, 2023-11-09:
> I accept that I2C is very commonly used, but if people don't look at the bus or do not notice the unexpected speed change as their sketch is only using a low data rate, then how many would even know that there is a problem ?

I did not dismiss the issue, just rather will not yet focus on it. You will be surprised how many people look at the clocks :)

tchilton, 2023-11-09:
OK, I've been digging and have found out a lot more about the ESP32, so handy learnings. Pretty much all of this is in the Technical Manual for the device which I should have spent more time reading before.

I can now access the registers for the various devices and the following code has been added to dump the I2C register values to see what it yields about the state of the peripheral before and after the test run.

Add the following into the incudes and defines sections, this provides the base IO address for Peripheral Bus 1 and 2 (PERIBUSx) and the offsets for both I2C0 and I2C1.  PERIBUS2 is used as section 1.3.5.2 says that bus1 does some speculative reads that would just add uncertainty that we don't need

```
#include "soc/dport_access.h"             // https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/dport_access.h

// All IO is based on offsets from the Base Addresses, see section 5.13 of the ESP32-S2 technical manual
// See section 1.3.1 of ESP32S2 Technical Manual/ Peripheral Bus base addresses
// See 1.3.5.2, PERIBUS1 does speculative reads, PERBUS2 doesn't
#define PERIBUS1                0x3F400000      // Address of IO base Peripheral Bus 1 PERI BUS 1
#define PERIBUS2                0x60000000      // Address of IO base Peripheral Bus 2 PERI BUS 2

// I2C0 and I2C1 are defined in 19.7 of the technical manual
#define I2C0_BASE_OFFSET        0x00013000      // I2C0 is at 0x0001_3000 0x0001_3FFF
#define I2C1_BASE_OFFSET        0x00027000      // I2C1 is at 0x0002_7000 0x0002_7FFF
```

Then add the following new function above setup(); This gets the register contents and dumps them out

```
// Dump the registers from the I2C device to see if that provides any info on what is happening
void dump_regs(const char * msg) {
  Serial.print("Register Dump : ");
  Serial.println(msg);
  
  uint32_t  iobase = PERIBUS2 | I2C0_BASE_OFFSET;     // Base address is bus address, plus device's IO offset
  Serial.printf("I2C IOBase : 0x%08X\n", iobase);     // 0x6001 3000 according to Section 19.7, Table 90 in ESP32S2 Tech manual

  // Register addresses from Section 19.8 Register summary, ESP32-S2 Technical manual
  // Only printing out those that look relevant to the problem
  Serial.printf("I2C_SCL_LOW_PERIOD_REG       : 0x%08X\n",DPORT_REG_READ(iobase | 0x0000));  // Low level width of clock
  Serial.printf("I2C_SCL_HIGH_PERIOD_REG      : 0x%08X\n",DPORT_REG_READ(iobase | 0x0038));  // High level width of clock
  Serial.printf("I2C_SCL_ST_TIME_OUT_REG      : 0x%08X\n",DPORT_REG_READ(iobase | 0x0098));  // SCL status time out register
  Serial.printf("I2C_SCL_MAIN_ST_TIME_OUT_REG : 0x%08X\n",DPORT_REG_READ(iobase | 0x009c));  // SCL main status time out register
  Serial.printf("I2C_SCL_STRETCH_CONF_REG     : 0x%08X\n",DPORT_REG_READ(iobase | 0x00a4));  // SCL clock stretching configuration register
  Serial.printf("I2C_SR_REG                   : 0x%08X\n",DPORT_REG_READ(iobase | 0x0008));  // I2C Status register
  Serial.printf("I2C_FIFO_ST_REG              : 0x%08X\n",DPORT_REG_READ(iobase | 0x0014));  // I2C FIFO Status register
  
  Serial.println("");
  Serial.println("");
}

```

Then, 
After the line Serial.println("Main"); add in
`dump_regs("Start of Main"); ` 

Before the line   Serial.println("Test complete, press reset"); add in
` dump_regs("End of run"); ` 

Compile and run. The output from this is as follows 

```
15:51:45.217 -> I2C IOBase = 0x60013000
15:51:45.217 -> I2C_SCL_LOW_PERIOD_REG       : 0x00000063
15:51:45.217 -> I2C_SCL_HIGH_PERIOD_REG      : 0x000C0034
15:51:45.217 -> I2C_SCL_ST_TIME_OUT_REG      : 0x00000100
15:51:45.217 -> I2C_SCL_MAIN_ST_TIME_OUT_REG : 0x00000100
15:51:45.217 -> I2C_SCL_STRETCH_CONF_REG     : 0x00000000
15:51:45.217 -> I2C_SR_REG                   : 0x00008000
15:51:45.217 -> I2C_FIFO_ST_REG              : 0x0E800000

15:51:45.419 -> Register Dump : End of run
15:51:45.419 -> I2C IOBase = 0x60013000
15:51:45.419 -> I2C_SCL_LOW_PERIOD_REG       : 0x00000063
15:51:45.419 -> I2C_SCL_HIGH_PERIOD_REG      : 0x000C0034
15:51:45.419 -> I2C_SCL_ST_TIME_OUT_REG      : 0x00000100
15:51:45.419 -> I2C_SCL_MAIN_ST_TIME_OUT_REG : 0x00000100
15:51:45.419 -> I2C_SCL_STRETCH_CONF_REG     : 0x00000000
15:51:45.419 -> I2C_SR_REG                   : 0x77008001
15:51:45.419 -> I2C_FIFO_ST_REG              : 0x0E800000
```

The register purposes / values can be seen in the technical manual, section 19.8. 
I've only added those that look like they might be related to this and ignored the rest for clarity. Similarly, I've only probed I2C0, not I2C1 as I am not using both busses in that design

It looks like the peripheral its self is not the problem. The values are identical except for the status register and digging into the bit meanings looks OK, its reflecting the status at the end of transmission.

Next up, I'm going to dig more into the clock registers / sources and their values. I couldn't see anything in section 19.8 that indicated the clock sources

astrogene1000, 2023-11-09:
To close the loop on S3 and PICO-D4, startup communications
First addr 14 and gets response, then addr 15 no response and finally addr 16 and no response
No reset's of CLK

Pico-D4
[image]

S3 DevKitC-1
[image]

And finally a 10 second capture for PICO-KIT-D4 where I pull SDA at target side
The number 3 trace is toggled when addr 14 comms returns an I2C error
[image]

And a close up of the error toggling every three (addr 14 toggles, 15 and 16 do not toggle the bit)
[image]

The steady state case with no induced error is 1824 I2C transfers per second to/from(combined 1824) addr 14 and 228 fails to addr 15 and 16 combined

I do not see much wrong with the PIC-D4 even through my 25ft longwire with Adafruit extender in circuit.

tchilton, 2023-11-09:
@astrogene1000. For completeness, which version of esp32 was installed in Boards manager 2.0.11, 2.0.14 etc 

@iharshadec - have you had a chance to get a scope / Logic Analyser onto things yet. Would be good to confirm what is different in your case as it would seem that something else may be at play here for you. 

Could it be for example that the native ESP-IDF libraries are faster than the older Arduino ones and this results in trying to talk to your ADC faster than it can run ? 
What is the I2C bus speed you have configured ?
What is the ADC you are using ?
What is its maximum sample frequency ?
Did you check the endTransmission() response codes ?
I also reviewed your code, all your I2C code is writing, I'd expect mostly reads for an ADC chip
Does the workaround work for you ?

I think that until we can get more info, we need to consider this a different problem to the one reported on this issue, given the lack of info and confirmation from others that they are not seeing the same defect on the PICO-D4

tchilton, 2023-11-10:
A further update following some testing on defect #8855. The current GitHub master branch no longer contains the bus speed setting issue and now mostly works, except testing at 1MHz, which resulted in a 909KHz clock rate and another defect where PlatformIO builds against the master GitHub branch fail to install properly

This clock rate defect remains in place in the current V3.0 alpha code release, however the workaround still operates now that the bus clock rate can be correctly controlled.

tchilton, 2023-11-10:
@me-no-dev, @astrogene1000 
I think I've found the culprit of the problem, it is as was suggested above, clock source / divider changes.

I was reading a bit more on the clock source selection and sections 19.3.3 in the tech manual details the clock source selection via the I2C_REF_ALWAYS_ON bit. 0=REF_TICK, 1=APB_CLK, both are shown in section 2.2.1, Clock Overview, similarly, the clock sources for each peripheral is listed in section 2.2.4, peripheral clocks and each clock is defined in the sections below that.

The I2C_REF_ALWAYS_ON bit is present in the I2C_CTR_REG (0x0004) register, so I added this to the dump_regs function and re-tested
```
// Dump the registers from the I2C device to see if that provides any info on what is happening
void dump_regs(const char * msg) {
  Serial.print("Register Dump : ");
  Serial.println(msg);
  
  uint32_t  iobase = PERIBUS2 | I2C0_BASE;     // Base address is bus address, plus device's IO offset
  Serial.printf("I2C IOBase : 0x%08X\n", iobase);     // 0x6001 3000 according to Section 19.7, Table 90 in ESP32S2 Tech manual

  // Register addresses from Section 19.8 Register summary, ESP32-S2 Technical manual
  // Only printing out those that look relevant to the problem
  Serial.printf("I2C_CTR_REG                  : 0x%08X\n",DPORT_REG_READ(iobase | 0x0004));  // Transmission Setting / Clock source
  Serial.printf("I2C_SCL_LOW_PERIOD_REG       : 0x%08X\n",DPORT_REG_READ(iobase | 0x0000));  // Low level width of clock
  Serial.printf("I2C_SCL_HIGH_PERIOD_REG      : 0x%08X\n",DPORT_REG_READ(iobase | 0x0038));  // High level width of clock
  Serial.printf("I2C_SCL_ST_TIME_OUT_REG      : 0x%08X\n",DPORT_REG_READ(iobase | 0x0098));  // SCL status time out register
  Serial.printf("I2C_SCL_MAIN_ST_TIME_OUT_REG : 0x%08X\n",DPORT_REG_READ(iobase | 0x009c));  // SCL main status time out register
  Serial.printf("I2C_SCL_STRETCH_CONF_REG     : 0x%08X\n",DPORT_REG_READ(iobase | 0x00a4));  // SCL clock stretching configuration register
  Serial.printf("I2C_SR_REG                   : 0x%08X\n",DPORT_REG_READ(iobase | 0x0008));  // I2C Status register
  Serial.printf("I2C_FIFO_ST_REG              : 0x%08X\n",DPORT_REG_READ(iobase | 0x0014));  // I2C FIFO Status register
  
  Serial.println("");
  Serial.println("");
}
```
If I then re-run without the workaround in place, then before the run, the value in 
I2C_CTR_REG is : 0x00000813, but after running, the value is : 0x00000013, therefore bit 11 in the register (I2C_REF_ALWAYS_ON) is changing from 1 to 0 and thus the clock is changing from APB_CLK to REF_TICK with the resulting change in data rate on the I2C bus. 

Repeating with the workaround in place results in the 0x00000813 value being present at the end. 

So, guess the next challenge is to find out what is writing to the I2C_CTR_REG and flipping that bit in error

me-no-dev, 2023-11-10:
@tchilton thanks for testing with master! Now let's finalize the issue so that I can build a test report for the IDF team.

Can you guys provide me with:
- Minimal example code to just reproduce the issue (really minimal)
- Environment requirements for the bug to appear (chip name, i2c device not connected, holding SDA, etc.)
- Workaround: code, when to apply, does it need to be reapplied

I will then turn this into ESP-IDF project and send it to them for evaluation (after confirming reproduction in ESP-IDF)

TD-er, 2023-11-10:
Just an idea... can you also check to see if the CPU and APB frequency remains the same?
I've been reading about clock selection on other platforms like the ESP32-C2 and ESP32-C6.
The technical reference guide suggested these clock divider values and clock sources may change when some peripherals no longer require a high frequency APB clock.
So maybe the S2 also has something similar.

tchilton, 2023-11-11:
I was intrigued by the defect now I know that the register is changing unexpectedly and I thought, hey I've got an [ESP-Prog](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/other/esp-prog/user_guide.html) and the JTAG interface is already, so what can I find. I've now learned just enough to be dangerous with GDB, but have found something interesting that I think gives a good start into what the root cause is.

I'm using the same test code detailed above, but am working within PlatformIO on Visual Studio Code, I've got ESP IDF 4.4 selected using the manual install method detailed in https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation, so platform_packages is set to idf-release/v4.4. I can't go newer than this as it doesn't download and install properly on the newer v5.1 or master branch at the moment.

my platformio.ini has a section as follows, which gives the frameworks, esp-prog for upload and debug and COM3 for debugging :

```[env:esp32-s2-saola-1-git-master-jtag]
platform = https://github.com/platformio/platform-espressif32.git
; platform = espressif32
board = esp32-s2-saola-1
framework = arduino
; should be on master, but its buggy atm, also v5.1 is buggy, others work
;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#release/v2.x
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#idf-release/v4.4
;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#idf-release/v5.1
upload_protocol = esp-prog
;upload_port = COM3
monitor_speed = 115200
build_type = debug

; Debugging via JTAG, see https://community.platformio.org/t/esp32-debugging-setup-and-troubleshooting/28648
debug_tool = esp-prog
; to stop in setup rather than in main
debug_init_break = tbreak setup
```
The platform= and platform_packages= can be removed and the commented out platform= line re-inserted for standard vanilla Platform IO / ESP-IDF / Arduino / JTAG configuration

If I start debugging, then go to Memory and add a watch for the same address (add 0x60013004 with a size of 8 bytes) and also add a Watch expression of *(0x60013005) which is the specific byte I'm interested in, then I see the value initialise with the 0x00000813 value (or just 0x08 in the watch). This value changes at the point in the code where i=9 (10th loop) and the call to Wire.endTransmission() is invoked. Digging a bit further, this is where it actually starts transmitting the data.

I also managed to get a breakpoint onto the i2cSetClock function, thinking that if something else was calling it, then I should get a breakpoint, however this never fires except on initial setup, which makes sense. 

What I really wanted though was a breakpoint on the address that is changing, but I couldn't achieve this in the user interface, however there is a debug console window that gives the command line and you then have native GDB access :-) 

After a bit of googling and testing, I found the watch and awatch commands which I set on 0x60013004, but I was getting nothing, other than my code doing the register dumps. Defeated late last night, I gave up and got some sleep. 

This morning, I realised my mistake. There are two paths to peripherals - one via PERIBUS1 and one via PERIBUS2 and I was only using PERIBUS2 because of PERIBUS1 doing speculative reads (see above). So I've re-run the tests today with both active and I've found out what's happening.

This meant adding another Watch expression, so I've now got the following expressions, which are just the individual bytes and are not word aligned, its just a standard pointer de-reference
WATCH : 
```
*(0x60013005)
*(0x3F413005)
```

Then, under MEMORY, these have to be word aligned 
```
0x60013004+8
0x3F413004+8
```
Note memory is added by setting the address (top centre box), hitting return, then 8 in the next box for 8 bytes, you can't just type the expression directly.

For the next bit, to get to the full debug session, build and compile as detailed above, start a debugging session via run, start debugging. Once the debugger is connected after a few seconds, switch to the debug console (at the bottom) and type the following commands  to set write watch hardware breakpoints and to see which breakpoints are set. 

```
watch *0x3F413004
watch *0x60013004
info b
```
Optionally replace watch with awatch and you get read and write, but that resulted in a more breaks, which was odd as the register in question is a control register, not a status register.

The output of the breakpoint print is as follows.
```
Num     Type           Disp Enb Address    What
2       breakpoint     keep y   0x4008302c in i2cSetClock at ....../.platformio/packages/framework-arduinoespressif32@src-5839966d996d63b391b36d77e8e64919/cores/esp32/esp32-hal-i2c.c:247
	breakpoint already hit 1 time
9       breakpoint     keep y   0x400817c9 in loop() at src/main.cpp:101
10      breakpoint     keep y   0x40081829 in loop() at src/main.cpp:138
12      hw watchpoint  keep y              *0x3F413004
13      hw watchpoint  keep y              *0x60013004
[... 139 lines cut ...]

```
Thread 
1 "loopTask" received signal SIGTRAP, Trace/breakpoint trap.
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
[... cut ...]
@me-no-dev, hopefully this gives a clear view on what is happening and I guess the only question is what is calling the i2c_ll_master_init within the HAL. Given its in the HAL, this might explain why its only impacting the ESP32-S2 too.

The line in question is 
```0x4002517d in i2c_ll_trans_start (hw=0x3f413000) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s2/include/hal/i2c_ll.h:490```

Do you still want the simple code that you requested or does this provide more info to the dev team ?
I'm happy to do further testing if you want me to.

tchilton, 2023-11-11:
I've just found out about the backtrace (bt) command in GDB, which gives the stack trace of how its getting here.
```
> bt

#0  0x4008b6b5 in i2c_ll_set_data_mode (rx_mode=I2C_DATA_MODE_MSB_FIRST, tx_mode=I2C_DATA_MODE_MSB_FIRST, hw=0x3f413000) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s2/include/hal/i2c_ll.h:372
#1  i2c_hal_master_init (hal=<optimized out>, i2c_num=0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/i2c_hal.c:196
#2  0x400848c6 in i2c_hw_fsm_reset (i2c_num=0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s2/include/hal/i2c_ll.h:949
#3  0x4008569a in i2c_master_cmd_begin (i2c_num=0, cmd_handle=<optimized out>, ticks_to_wait=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2c.c:1506
#4  0x40083009 in i2cWrite (i2c_num=<optimized out>, address=96, buff=0x3ffca6a4 "", size=2, timeOutMillis=50) at C:/Users/tim/.platformio/packages/framework-arduinoespressif32@src-5839966d996d63b391b36d77e8e64919/cores/esp32/esp32-hal-i2c.c:176
#5  0x40081e77 in TwoWire::endTransmission (this=0x3ffc2304 <Wire>, sendStop=true) at C:/Users/tim/.platformio/packages/framework-arduinoespressif32@src-5839966d996d63b391b36d77e8e64919/libraries/Wire/src/Wire.cpp:453
#6  0x40081f34 in TwoWire::endTransmission (this=0x3ffc2304 <Wire>) at C:/Users/tim/.platformio/packages/framework-arduinoespressif32@src-5839966d996d63b391b36d77e8e64919/libraries/Wire/src/Wire.cpp:642
#7  0x400817f0 in loop () at src/main.cpp:105
#8  0x40083bb0 in loopTask (pvParameters=<optimized out>) at C:/Users/tim/.platformio/packages/framework-arduinoespressif32@src-5839966d996d63b391b36d77e8e64919/cores/esp32/main.cpp:50

```
So, it looks like the I2C Write starts, calls i2c_master_cmd_begin which then calls i2c_hw_fsm_reset. The next question is Why is it doing that.

me-no-dev, 2023-11-11:
very good work @tchilton ! Minimal sketch is still required for me to turn into IDF code and submit to the IDF team. Though your explanation and finding cover up well enough what is actually happening.

astrogene1000, 2023-11-11:
@tchilton  Not sure about i2c_hw_fsm_reset
I first took out
esp32s2\include\hal\esp32s2\include\hal\i2c_ll.h

static inline void i2c_ll_master_fsm_rst(i2c_dev_t *hw)
{
 **//hw->ctr.fsm_rst = 1;**
}

No change
Then   changed
\esp32s2\include\soc\esp32s2\include\soc\soc_caps.h

// FSM_RST only resets the FSM, not using it. So SOC_I2C_SUPPORT_HW_FSM_RST not defined.
//ESP32-S2 support hardware clear bus
**// ORIG #define SOC_I2C_SUPPORT_HW_CLR_BUS  (1)**
**#define SOC_I2C_SUPPORT_HW_FSM_RST (1)**

Same slow clock after two tries, one to addr 15, second to addr 16 then slow

Then totally took out both
**// ORIG #define SOC_I2C_SUPPORT_HW_CLR_BUS  (1)**
**// ADDED #define SOC_I2C_SUPPORT_HW_FSM_RST (1)**

And no I2C at all :-)

tchilton, 2023-11-11:
I guess FSM -> Finite State Machine, so resetting it if the code thinks the I2C subsystem is hung is probably a reasonable strategy, but in our cases, its failing to send because the target device is not present and not because the bus is hung. If something is reset, it should put it back to the expected state, not to a different state - i.e. different clock source / bus speed etc.

I've been looking a bit more into the debugger and code too.
I've found that you can do a "bt -full -no-filters" in GDB and get the variable contents on the stack. I've looked through them and many are <optimised out> and not visible, others are not particularly helpful to the diagnosing the problem - pointers to control blocks etc. I could dump these then would have to unpack  them etc, so would need to know which ones I cared about before spending that time.

I've also looked at the source code i2c.c, i2c_master_cmd_begin is interesting, it says in lines 1478 to 1524

```
    // Wait event bits
    i2c_cmd_evt_t evt;
    while (1) {
        TickType_t wait_time = xTaskGetTickCount();
        if (wait_time - ticks_start > ticks_to_wait) { // out of time
            wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
        } else {
            wait_time = ticks_to_wait - (wait_time - ticks_start);
            if (wait_time < I2C_CMD_ALIVE_INTERVAL_TICK) {
                wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
            }
        }
        // In master mode, since we don't have an interrupt to detective bus error or FSM state, what we do here is to make
        // sure the interrupt mechanism for master mode is still working.
        // If the command sending is not finished and there is no interrupt any more, the bus is probably dead caused by external noise.
        portBASE_TYPE evt_res = xQueueReceive(p_i2c->cmd_evt_queue, &evt, wait_time);
        if (evt_res == pdTRUE) {
            if (evt.type == I2C_CMD_EVT_DONE) {
                if (p_i2c->status == I2C_STATUS_TIMEOUT) {
                    // If the I2C slave are powered off or the SDA/SCL are connected to ground, for example,
                    // I2C hw FSM would get stuck in wrong state, we have to reset the I2C module in this case.
                    i2c_hw_fsm_reset(i2c_num);
                    clear_bus_cnt[i2c_num] = 0;
                    ret = ESP_ERR_TIMEOUT;
                } else if (p_i2c->status == I2C_STATUS_ACK_ERROR) {
                    clear_bus_cnt[i2c_num]++;
                    if (clear_bus_cnt[i2c_num] >= I2C_ACKERR_CNT_MAX) {
                        clear_bus_cnt[i2c_num] = 0;
                        i2c_hw_fsm_reset(i2c_num);
                    }
                    ret = ESP_FAIL;
                } else {
                    ret = ESP_OK;
                }
                break;
            }
            if (evt.type == I2C_CMD_EVT_ALIVE) {
            }
        } else {
            ret = ESP_ERR_TIMEOUT;
            // If the I2C slave are powered off or the SDA/SCL are connected to ground, for example,
            // I2C hw FSM would get stuck in wrong state, we have to reset the I2C module in this case.
            i2c_hw_fsm_reset(i2c_num);
            clear_bus_cnt[i2c_num] = 0;
            break;
        }
    }
```
So, the logic seems good - can't communicate so attempt some recovery steps. This is in a generic i2c module, not specific to ESP32-S2 (i.e. in the HAL or in ESP32S2 specific folders) then the logic must work on other CPU's in the family, or they handle it differently.

I'm struggling to get lower than this as "i2c_hw_fsm_reset (i2c_num=0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s2/include/hal/i2c_ll.h:949" at least in the source I have for IDF 4.4 doesn't have i2c_hw_fsm_reset at that line, it has i2c_ll_get_scl_clk_timing. I've checked some local copies and what's on GitHub in the IDF-4.4 branch (https://github.com/espressif/esp-idf/tree/release/v4.4), but still no joy.

I can't get any closer with GDB either. I've tried setting breakpoints when the PC reaches the parent functions address on the stack trace (GDB break *<address>) but I end up not catching it or it drops into Assembly Language, which I'm not particularly motivated to want to dig into. Its been a long time since I've been in Assembly Language or Microcode.

My next step is to produce the minimalistic test code as requested.

tchilton, 2023-11-11:
@me-no-dev. As requested please find attached a simplified piece of code to show the defect, was written in PIO using the Arduino framework, it is using stock versions of the espressif32 platform and no platform_packages, just to show its against the current stable code, even though we know its also impacting as far forwards as I can get with the master / v5.1 bug on PIO atm.

Much of the code block are comments and defines to detail what's happening. you may remove these if you wish

Note that you will need a working I2C device on the bus and will need to update the write_mux function to talk correctly to that device. I just happen to have an I2C Bus multiplexer on my board, this talks to other busses on other modules in my design. In this case, though, its function is just to generate real data on the bus which you can then probe with the logic analyser.

The fault happens when a number of writes have happened to a device which is not present. It appears from all the analysis above, the IDF is counting these and then figuring it has a bus problem and is attempting to reset the bus, but in doing so, it returns the I2C peripheral device to a different state with the wrong clock after that reset. 

As a suggestion, it should be looking at both successful and unsuccessful communication, with successful communication showing that the bus is not hung and resetting any error counters. This should work in 90% of other cases if you look at the details from others on this bug report. The 10% are those attempting to talk to empty or badly terminated busses, or buggy code that is not talking to the intended device,  but those are completely different problems which your code can't fix.

To see the problem, stick a Scope or Logic Analyser on the I2C pins and measure the clock rate as the code runs.

To use the workaround, un-comment the line and re-compile
```//#define USE_WORKAROUND 1```

Test code is as follows 
```
// Demo code to show un-commanded I2C bus data rate change on ESP32-S2 when using I2C and attempting to talk to a device which is not currently on the bus
// Initial cycles are at 400KHz as expected, however this changes to 5KHz after a short duration 
// The cause is the clock source changing in I2C_CTRL_REG, I2C_REF_ALWAYS_ON bit gets cleared, moving the clock from APB_CLK to REF_TICK
// See https://github.com/espressif/arduino-esp32/issues/8480 for detail on the bug and analysis on its cause

// Bug experienced on ESP32-S2 chip. Works as expected on ESP32 DevKit V1, reported by others to work properly on ESP32-S3 and PICO-D4
// In my case, change happens after exactly 10 attempts to write to non-existent devices, in others, it can be shorter (no of bytes in payload ?)
// It doesn't matter if other successful I2C commands are issued during, or after the non-existent device writes. So, 
// the IDF is not seeing good transmissions as a working bus, it only seems to be counting failed ones
// Attempting to reset the bus speed to the original value has no impact, unless you select a different bus speed (i,e, 1M) first, then switch back to the desired speed (400K)

#include <Arduino.h>
#include <Wire.h>
#include "soc/dport_access.h"                     // To dump register contents

#define I2C_SDA                          8        // I2C Data pin
#define I2C_SCL                          9        // I2C Clock pin

#define I2C_CLKRATE_400K            400000        // Speed of I2C bus 400KHz
#define I2C_CLKRATE_1M             1000000        // Speed of I2C bus 1MHz
#define TEST_CYCLES                    100        // How many test cycles to run

#define I2C_ID_TALK_TO                0x60        // The target device, which is not present
#define I2C_BUS_MUX                   0x71        // An 8 channel I2C bus mux - PCA9547, which is present -- change to the device you are testing against

// All IO is based on offsets from the Base Addresses, see section 5.13 of the ESP32-S2 technical manual
// See section 1.3.1 of ESP32S2 Technical Manual/ Peripheral Bus base addresses
// See 1.3.5.2, PERIBUS1 does speculative reads, PERBUS2 doesn't, hence I'm using PERIBUS2
#define PERIBUS2                0x60000000        // Address of IO base Peripheral Bus 2 PERI BUS 2

// I2C0 and I2C1 are defined in 19.7 of the technical manual
#define I2C0_BASE               0x00013000        // I2C0 registers are at 0x0001_3000 0x0001_3FFF

void write_mux(int);                              // Test function to write to a device that is always present on the bus.

// Dump the registers from the ESP32 I2C device to see what its doing
void dump_regs(const char * msg) {
  Serial.print("Register Dump : ");
  Serial.println(msg);
  
  uint32_t  iobase = PERIBUS2 | I2C0_BASE;        // Base address is bus address, plus device's IO offset
  Serial.printf("I2C IOBase : 0x%08X\n", iobase); // 0x6001 3000 according to Section 19.7, Table 90 in ESP32S2 Tech manual

  // Register addresses from Section 19.8 Register summary, ESP32-S2 Technical manual
  // Only printing out those that are relevant to the problem
  // Problem is visible in I2C_CTR_REG, the clock source is changed, hence the problem
  Serial.printf("I2C_CTR_REG                  : 0x%08X\n",DPORT_REG_READ(iobase | 0x0004));  // Transmission Setting / Clock source
  Serial.printf("I2C_SCL_LOW_PERIOD_REG       : 0x%08X\n",DPORT_REG_READ(iobase | 0x0000));  // Low level width of clock
[... cut ...]
  Serial.println("Test complete, press reset");
  while (1) ;
}

// Write to the I2C bus mux - PCA9547
// You will need another device you can talk successfully to, so you can generate some data
// Change this function to talk to that device.
void write_mux(int channel) {
    Wire.beginTransmission(I2C_BUS_MUX);
    Wire.write(channel & 0x07);                    // Select the MUX channel
    Wire.endTransmission();
}
```
Once this is converted into an IDF bug, can you post its ID on this page please.

tchilton, 2023-11-12:
@TD-er re your comment 

> Just an idea... can you also check to see if the CPU and APB frequency remains the same? I've been reading about clock selection on other platforms like the ESP32-C2 and ESP32-C6. The technical [...]

I've had a quick look at the clocks and there are loads of bits all over the place that do clock enabling, dividers and selection etc, so without something more specific such as which registers you re referring to in which functional area, this is going to be difficult and time consuming, additionally, given what has been discussed in the last couple of days and the clock will still be needed by the ongoing I2C transactions, I think that is a very unlikely cause, particularly as we know that the bit that selects the clock for I2C is being flipped by some I2C bus specific bus recovery logic.

TD-er, 2023-11-12:
@tchilton 
I totally agree.
I also agree with your idea that it might be some baked-in I2C bus recovery algorithm.

I can give you a perfectly normal use case however where you might run into such issues where N devices may not reply and thus perhaps trigger such behavior and that is performing an I2C bus scan.

So maybe that would make your test program even simpler?

In my project (ESPEasy) I had some "recovery" algorithm in place to free a stuck I2C bus and had to disable this when migrating to IDF5.1 code.
So I can confirm the I2C code in the IDF has changed in such a way that it was absolutely not "happy" with what I tried to do here.
I will hook up a logic analyzer too to all supported ESP32's I have (C2/C3/C6/S2/S3/ESP32-classic) to see if others may show similar behavior when trying to address a non-existing I2C device.

TD-er, 2023-11-12:
I can confirm just scanning for I2C devices is enough to mess up the clock frequency on an ESP32-S2

[image]

10 scans at 100 kHz and then it went to 1.25 kHz.
That's 1/80th so it seems like it may have switched clocks from 80 MHz (APB ???) to 1 MHz

Tested on ESP32-S2FNR2
Reported APB frequency is 80 MHz.

tchilton, 2023-11-12:
@TD-er, Yes, that is consistent with what others are seeing, the clock is changing from APB_CLK to REF_TICK.

You mentioned you had other devices (C2/C3/C6/S2/S3/ESP32-classic), what does the same test do on those ? Is it only ESP32-S2 that is affected ? Could you provide test results please ?

As to simplifying the test case, mine is based on a real life use case, where I have an I2C bus multiplexer that maps one of up to 8 remote I2C busses onto the main bus that the ESP32-S2 controls. I'm only using 4 of these, they are in turn connected via I2C differential drivers to the remote modules that either just have additional I2C devices or an ESP32's or Raspberry Pi's on them acting as I2C slaves. In my case a Front Panel with User Interface controls on it, power monitoring (Solar generation and battery state / charge / discharge) and some control modules or Machine Learning modules that detects Chickens, Rats, Eggs and Stones (As not an egg)

Not all will be installed at all times, and they may be removed, reset, get a firmware update or crash at any time, but the main unit must continue to operate as it manages the welfare of some chickens. This is how I found the problem. My test case is therefore based on the PCB's I have at hand. It also has the I2C bus scanning, but mine is enhanced to map the base bus on the main board and each of the remote busses.

From reading the ESP IDF code, it appears that its only detecting "failure to transmit" as "buffer not empty" and not seeing the difference between a dead bus or a dead device on the bus. Its also not measuring successful transmission as a positive indicator of a functioning bus which is therefore not dead and doesn't need resetting. 

What I would expect to happen is that a failure to send simply means no device present and the requested transaction fails and is removed from the buffers. This is easy to detect as the client pulls SDA low to perform an ACK. If there is no device, nothing pulls the bus low and this is a NACK, hence the Wire.endTransaction() response code 2

TD-er, 2023-11-12:
> You mentioned you had other devices (C2/C3/C6/S2/S3/ESP32-classic), what does the same test do on those ? Is it only ESP32-S2 that is affected ? Could you provide test results please ?

Already tested with the C3 and that one doesn't show this behavior.

I will try to test all others later this evening to see if this simple test case of performing an I2C bus scan will lower the I2C clock frequency.

I have all variantions of ESP32 present on my desk as I'm working hard on implementing support for IDF5.1 for my project.

TD-er, 2023-11-12:
All platforms checked and only the S2 is showing this behavior.

tchilton, 2023-11-12:
@TD-er. Thanks for confirming the test results, very helpful in narrowing this down. So, all contributions confirm that this is an ESP32-S2 specific bug.

I've spent a bit more time checking the data sheets between ESP32-S2 and ESP32-S3. The S3 doesn't have the clock selection in the I2C_CTR_REG and indeed it doesn't have the same clock sources as the S2, it uses XTAL_CLK and RC_FAST_CLK, both of which are in register - 0x0054 as opposed to The S2's sources of APB_CLK and REF_TICK in I2C_CTR_REG

So, I guess the question for Espressif is what is the difference in the I2C FSM reset logic between the ESP32-S2 and ESP32-S3 in regard to the handling of the I2C_CTR_REG (0x0004) on the S2 and the I2C_CLK_CONF_REG(0x0054) on the S3 when the FSM reset is performed. I'd guess that they are simply forgetting to select the correct clock source in the FSM reset on the ESP32-S2. I also think the logic of queue >0 is a bad principle for bus hang / reset processing.

One other thing I noted whilst reading the data sheets is that the ESP32's only support 100KHz and 400KHz data rates. Anything above 400KHz is not supported, which probably explains the issue I found on #8855 where the 1Mbps rate is actually 909KHz, this is a shame as I2C standards also support data rates of 1MBps and 3.4Mbps too. Hopefully support will be added in later firmware and hardware for this.

@me-no-dev. Hopefully you now have far more info than you need to get this replicated into ESP-IDF and ultimately resolved.

Thanks to all for the helpful contributions to diagnosing this problem. I've learned a lot.

TD-er, 2023-11-12:
Please also double (triple) check the macros operating on registers like the ones in `soc.h` 
In the last few days I have found several of them simply being incorrect.
Not yet for the S2, but there were a few for C2/C6/C3 which would result in incorrect address translations.
For example `REG_SPI_BASE(i)` was wrong on those.
So I wouldn't be surprised when others may be wrong also in the respective include files for each platform as quite a few of them seem to be changed for IDF5.1

me-no-dev, 2023-11-13:
So problem is in `i2c_ll_master_init()` in ESP32-S2's `i2c_ll.h`.  I can suggest the following edit and I will also propose that to the IDF team:
```cpp
static inline void i2c_ll_master_init(i2c_dev_t *hw)
{
    typeof(hw->ctr) ctrl_reg;
    uint32_t ref_always_on = ctrl_reg.ref_always_on;
    ctrl_reg.val = 0;
    ctrl_reg.ms_mode = 1;
    ctrl_reg.sda_force_out = 1;
    ctrl_reg.scl_force_out = 1;
    ctrl_reg.ref_always_on = ref_always_on;
    hw->ctr.val = ctrl_reg.val;
}
```

tchilton, 2023-11-13:
@me-no-dev That sounds good.  Thanks.

I've just compared to the v4.4.5 IDF source I have and its adding in the missing ref_always_on, which sounds good. I don't recall seeing i2c_ll_master_init in the stack trace though ?

A secondary question though is about the reset logic. Is it really necessary to reset the I2C module just because a slave device is not present ? Shouldn't the stack attempt to transmit and simply discard if there is an implicit NACK on the bus (As the client hasn't actively ACK'd the request since its not present). Also only detecting failure isn't necessarily right either, successful transmission of other frames should indicate that the bus is operating OK, hence doesn't need to be reset. This though would be more appropriate for the I2C module i2c_master_cmd_begin. 

Happy to test when any updates to the IDF come available.

me-no-dev, 2023-11-13:
@tchilton such questions better be asked in IDF's repo. I am not all that intimate with the reason about all chips, but we did have issues with ESP32 and also had to implement very similar approach in order to keep the bus running.
You did not see `i2c_ll_master_init` in the trace, because it was inlined :)

tchilton, 2023-11-13:
Thanks for the info. I'll have a look at the IDF repo and comments on that for I2C to get any history / I might make the suggestion there.
 
Inline would explain things. I had read that in the source code too :-)

TD-er, 2023-11-13:
An algorithm to keep the I2C bus from getting stuck is really useful as it does happen occasionally that the I2C bus gets stuck.

The I2C bus can operate on just about any clock frequency, from nearly DC to the max (100 kHz / 400 kHz, etc)
This is a really nice feature, but it also makes it hard to implement simple guards like a timeout.

So what can happen is that a device on the I2C bus interprets some glitch as a clock pulse and thus remains out of sync with the master.
You can't easily get out of such a situation as you don't know which slave device is hanging.
But it is keeping either the data or clock pin low, preventing other devices from using the bus.

So it can happen in any situation where an I2C bus is used.

In ESPEasy I use a very simple trick to get out of such a locked-up bus by switching SDA and SCL, starting the I2C bus again and then switch them back and start the I2C bus again.

iharshadec, 2023-11-24:
I tried one thing here -- ESP32 board
ADS1115 chip and I put constant 2.2v on channel now look at the profile.
Old version 1.0.6 is very close to 2.21v and not much change.. Same hardware and same environment but arduino version 2.0.14 and look at the change in reading. it's surely missing in I2C execution as sample is getting missed here looks like.

2.0.14
[image]

1.0.6
[image]

[6 more replies]

closed 2024-09-29 by tchilton