espressif/arduino-esp32 #8481: Upload error

bepitama, 2023-08-03:
### Board

Arduino Nano ESP32

### Device Description

Arduino Nano ESP32 connected via usb.

### Hardware Configuration

irrelevant

### Version

latest master (checkout manually)

### IDE Name

IDE 2.1.1

### Operating System

Linux Mint 21.1 Cinnamon

### Flash frequency

irrelevant

### PSRAM enabled

yes

### Upload speed

irrelevant

### Description

The error comes out when you try to send any sketch.
It seems that this problem is present, as well as on Linux also installed on Raspberry, also on IOs.
With the Windows 10 OS the problem does not exist. Everything works without errors.

### Sketch

```cpp
The error also occurs when nothing is connected. Even just trying to send the blink example program.
```

### Debug Message

```plain
dfu-util: Cannot open DFU device 2341:0070 found on devnum 3 (LIBUSB_ERROR_ACCESS)
dfu-util: No DFU capable USB device available
Upload Failed: Error while uploading: exit status 74
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-03:
This seems like you have not selected the correct board.  Under Tools, choose the esp32 dev module. The port should then be /dev/ttyUSBX (or /dev/ttyACMX in rpi).

me-no-dev, 2023-08-03:
> This seems like you have not selected the correct board. Under Tools, choose the esp32 dev module. The port should then be /dev/ttyUSBX (or /dev/ttyACMX in rpi).

He is using the Arduino.CC new board, which is using DFU for uploads.

> dfu-util: Cannot open DFU device 2341:0070 found on devnum 3 (LIBUSB_ERROR_ACCESS)

@bepitama this error looks like permissions error on UNIX. The OS does not give you access to the USB device. There were some lines that need to be added to linux configs to give you access to those devices. Quick Google search should produce some helpful results. Something like [this](https://askubuntu.com/questions/595896/how-to-allow-software-access-to-any-usb-devices)

tormodvolden, 2023-08-03:
> There were some lines that need to be added to linux configs to give you access to those devices.

For Linux udev rules examples, see also:
https://sourceforge.net/p/dfu-util/dfu-util/ci/master/tree/doc/60-dfuse.rules

bepitama, 2023-08-03:
It takes a clear and easy example tutorial. I looked at the page you indicated but it is not enough to understand what to do. Or this file should be added automatically when you install Arduino Nano ESP32 board in IDE.
Giuseppe

bealsj, 2023-08-04:
I had the same issue.  To fix the issue, I downloaded this [script](https://github.com/arduino/ArduinoCore-mbed/blob/main/post_install.sh), made it executable, opened the terminal and changed the directory to the download location of the script, and ran `sudo post_install.sh`.  When I reopened the Arduino IDE I was able to successfully send the blink script to the new board.  This process essentially does what was suggested previously but hopefully in a way that is clear and easy to follow.

bepitama, 2023-08-04:
It worked. Thank you very much. I love Open Source and Free Software because, in this environment, there are real people who share knowledge. I love you.
Thank you very much.
Giuseppe

closed 2023-08-18 by VojtechBartoska

tormodvolden, 2023-08-18:
> Or this file should be added automatically when you install Arduino Nano ESP32 board in IDE.

Indeed. Is this the wrong project, @VojtechBartoska ?

VojtechBartoska, 2023-08-21:
@tormodvolden I'm not sure, if I understand you. What do you mean?

tormodvolden, 2023-08-21:
> @tormodvolden I'm not sure, if I understand you. What do you mean?

The udev rules file should be added by some project so that the user doesn't have to do it manually. So I wonder why this report was closed as "solved". I asked whether this project is the right one because I am unsure about who is delivering what in this case. I see that https://github.com/arduino/arduino-esp32 is a downstream git repo from you, and if the Arduino Nano ESP32 support is added there and should not be upstreamed that is maybe where this bug report belongs (they don't have an issue tracker at GitHub though). Meanwhile there is a PR there to add this file. So maybe everything will be fine. However, if you are also providing board setups using DFU you should probably ship similar files.

me-no-dev, 2023-08-22:
Since we do not offer DFU boards and support for that is added only by Arduino.cc, I suggest that you contact them in their fork and let them know about your issue. If they find a fix, they will upstream it here and make it available for all.

tormodvolden, 2023-08-22:
Thanks, all fine then. I'll follow the PR on their fork.

Jubei-VA, 2024-02-01:
Why is this marked as solved? This is occurring to my nano esp32 on a stock Ubuntu 22.04 install. This needs to be documented as a known issue.

dss010101, 2024-03-09:
im seeing same issue trying to upload from VS Code.
It seems this fix as to be done on the arduino nano directly, is that correct?  Is there a way to ssh into the nano esp32 from my windows 11 laptop to fix this?

Jubei-VA, 2024-03-09:
> im seeing same issue trying to upload from VS Code. It seems this fix as to be done on the arduino nano directly, is that correct? Is there a way to ssh into the nano esp32 from my windows 11 laptop to fix this?

No it needs to be done on the host (the computer), not the Arduino. On Windows you should not have to worry about this issue. You are probably running into another issue.

dss010101, 2024-03-09:
> No it needs to be done on the host (the computer), not the Arduino. On Windows you should not have to worry about this issue. You are probably running into another issue.

i tried both Arudino Editor and VSCode, i get the same error when trying to upload
[image]

dss010101, 2024-03-09:
for my case i had to install the Espressif libraries, not the Arduino ones for ESP32 boards

JebKerman91, 2024-05-06:
> I had the same issue. To fix the issue, I downloaded this [script](https://github.com/arduino/ArduinoCore-mbed/blob/main/post_install.sh), made it executable, opened the terminal and changed the [...]

I too encountered the same issue with an Arduino Nano ESP32 while trying to upload to it from my laptop running Fedora 40. Running the script fixed the issue.

doctor-TZ, 2024-07-26:
> > > im seeing same issue trying to upload from VS Code. It seems this fix as to be done on the arduino nano directly, is that correct? Is there a way to ssh into the nano esp32 from my windows 11 laptop to fix this?我尝试从 VS Code 上传时遇到同样的问题。看来这个修复是直接在 arduino nano 上完成的，对吗？有没有办法从我的 Windows 11 笔记本电脑 ssh 进入 Nano esp32 来解决这个问题？
> > 
> > 
> > No it needs to be done on the host (the computer), not the Arduino. On Windows you should not have to worry about this issue. You are probably running into another issue.不，它需要在主机（计算机）上完成，而不是在 Arduino 上完成。在 Windows 上您不必担心这个问题。您可能遇到了另一个问题。
> 
> i tried both Arudino Editor and VSCode, i get the same error when trying to upload.我尝试了 Arudino Editor 和 VSCode，在尝试上传时遇到相同的错误。 I did have to install this driver in order for the port to be recognized: 我确实必须安装此驱动程序才能识别端口： https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads
> 
> ![image](https://private-user-images.githubusercontent.com/855627/311457070-6f5ac852-245a-4e89-9100-eb4366817412.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5NjM0MjcsIm5iZiI6MTcyMTk2MzEyNywicGF0aCI6Ii84NTU2MjcvMzExNDU3MDcwLTZmNWFjODUyLTI0NWEtNGU4OS05MTAwLWViNDM2NjgxNzQxMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyNlQwMzA1MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNzQzOWFhMWMwMTlkZjkwYzE5NTIxYjRhMTI5YTIwOWM5MGQ0YTQzNzU2NmIyNzA3MDQ5ZmExYTkwOTJhZWQ4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-O1iRzwsv-Uo7HotBOxRPM91h6v-BAZrhNerbIPaTkY)
> 
> UPDATE: it seems that if select Arduino Nano (Arduino AVR Boards) instead of 'Arduino ESP32 Boards', its able to find the port and attempt the upload, but taking forever. been stuck here for several minutes:更新：似乎如果选择 Arduino Nano（Arduino AVR 板）而不是“Arduino ESP32 板”，它能够找到端口并尝试上传，但需要很长时间。被困在这里几分钟了：
> 
> ```
> Using Port                    : COM3
> Using Programmer              : arduino
> Overriding Baud Rate          : 115200
> ```
> 
> RESOLVED for me issue: for my case i had to install the Espressif libraries, not the Arduino ones for ESP32 boards已解决我的问题：对于我的情况，我必须安装 Espressif 库，而不是 ESP32 板的 Arduino 库

I searched a lot of answer, and I found this is the really and only useful one. Thank you!

DrakerDG, 2024-10-20:
sudo +x post_install.sh./post_install.sh

> > I had the same issue. To fix the issue, I downloaded this [script](https://github.com/arduino/ArduinoCore-mbed/blob/main/post_install.sh), made it executable, opened the terminal and changed the directory to the download location of the script, and ran `sudo ./post_install.sh`. When I reopened the Arduino IDE I was able to successfully send the blink script to the new board. This process essentially does what was suggested previously but hopefully in a way that is clear and easy to follow.
> 
> Worked for me! Thanks

Thanks a lot!  It works for me too!

In my case it was necessary to first apply the following command: sudo +x post_install.sh./post_install.sh

DuckersMcQuack, 2025-10-28:
> I had the same issue. To fix the issue, I downloaded this [script](https://github.com/arduino/ArduinoCore-mbed/blob/main/post_install.sh), made it executable, opened the terminal and changed the [...]

Thank you! Been trying to get this blasted waveshare esp32-s3-nano to read for hours now lol.

Also, for others who gets "command not found" to run it, `chmod +x post_install.sh`, then `sudo ./post_install.sh`