espressif/arduino-esp32 #8482: Increase in HEAP usage on 2.0.11 (framework 4.4.5)

stuartpittaway, 2023-08-03:
### Board

ESP32 DEVKITC 

### Device Description

DEVKITC as used in DIYBMS project https://github.com/stuartpittaway/diyBMSv4/tree/master/ControllerCircuit

### Hardware Configuration

Yes, but doesn't make a difference to this issue

### Version

other

### IDE Name

PlatformIO

### Operating System

Win 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921000

### Description

After upgrading from 2.0.9 release to the 2.0.11 release, the memory heap usage has significantly increased by about 30%

There are no other changes to the code base, except for the re-compile with the new 2.0.11 release.

I've also tested with 2.0.10 and that was worse than 2.0.11 release with Min free Heap reported as 49772.

Values returned by call to ```ESP.getMinFreeHeap()``` which calls ```heap_caps_get_minimum_free_size(MALLOC_CAP_INTERNAL)```.

On 2.0.9
```
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.9

RAM:   [==        ]  19.6% (used 64192 bytes from 327680 bytes)
Flash: [========= ]  90.5% (used 1601337 bytes from 1769472 bytes)

SDK Version: v4.4.4
Min free Heap: 90612
Free heap: 114728
Heap size: 291396
```

On 2.0.11
```
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.11

RAM:   [==        ]  19.5% (used 63812 bytes from 327680 bytes)
Flash: [========= ]  90.7% (used 1604529 bytes from 1769472 bytes)

SDK Version: v4.4.5
Min free Heap: 61776
Free heap: 78944
Heap size: 256136
```

### Sketch

```cpp
ESP.getMinFreeHeap();
```

### Debug Message

```plain
No debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-03:
Confirmed. Using this code:
``` cpp
  Serial.printf("Total Heap Size: %d\n",ESP.getHeapSize());
  Serial.printf("Available Heap: %d\n",ESP.getFreeHeap());
  Serial.printf("Lowest level of heap since boot: %d\n",ESP.getMinFreeHeap());
  Serial.printf("Largest block of heap that can be allocated at once: %d\n",ESP.getMaxAllocHeap());
```

All SoC results with empty sketch:

C3 with 2.0.11
Total Heap Size: 323620
Available Heap: 301508
Lowest level of heap since boot: 296528
Largest block of heap that can be allocated at once: 278516

C3 with 2.0.9
Total Heap Size: 331848
Available Heap: 309700
Lowest level of heap since boot: 304720
Largest block of heap that can be allocated at once: 286708

================================================================
ESP32 with 2.0.11
Total Heap Size: 340052
Available Heap: 314656
Lowest level of heap since boot: 311272
Largest block of heap that can be allocated at once: 110580

ESP32 with 2.0.9
Total Heap Size: 374708
Available Heap: 349208
Lowest level of heap since boot: 343664
Largest block of heap that can be allocated at once: 114676

================================================================
ESP32S2 with 2.0.11
Total Heap Size: 254052
Available Heap: 233796
Lowest level of heap since boot: 228816
Largest block of heap that can be allocated at once: 204788

ESP32S2 with 2.0.9
Total Heap Size: 252324
Available Heap: 232068
Lowest level of heap since boot: 227088
Largest block of heap that can be allocated at once: 204788

================================================================
ESP32S3 with 2.0.11
Total Heap Size: 400508
Available Heap: 375104
Lowest level of heap since boot: 369956
Largest block of heap that can be allocated at once: 335860

ESP32S3 with 2.0.9
Total Heap Size: 399832
Available Heap: 374384
Lowest level of heap since boot: 369236
Largest block of heap that can be allocated at once: 335860

SuGlider, 2023-08-03:
@VojtechBartoska @me-no-dev --
FYI:
I have seen a number of issues related to IDF 4.4.5, including I2S and other memory problems, such as this one, that breaks sketched made with IDF 4.4.4.

me-no-dev, 2023-08-22:
@SuGlider I found the problem. It's this: https://github.com/espressif/arduino-esp32/pull/8243

Reverting it fixes the memory issue.

closed 2023-08-23 by me-no-dev

Jason2866, 2023-08-23:
Great finding!!

Tasmota Arduino fork metrics

```
Tasmota Arduino 2.0.11
Total Heap Size: 356236
Available Heap: 331996
Lowest level of heap since boot: 328364
Largest block of heap that can be allocated at once: 114676

Tasmota Arduino 2.0.11 build 1530 (with the fix)
Total Heap Size: 403840
Available Heap: 369628
Lowest level of heap since boot: 364848
Largest block of heap that can be allocated at once: 122868
```