espressif/arduino-esp32 #8484: Interrupt not working w/ ESP32-C3-MINI-1

abhi-sachdeva, 2023-08-03:
### Board

ESP32-C3

### Device Description

[image]

### Hardware Configuration

This is just a simulator, but on the actual device build, I have 1 button connected from a boost converter to the 3v3 of the eESP32-C3-MINI-1 (which will control the initial power on of the module and start a BLE scan, then fall asleep). The second button is from GPIO3 to GND (which will control the wake-up from deep sleep).

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

115200

### Description

I am simply unable to get the button to wake-up the ESP32-C3. My code runs up until after the module falls goes into deep sleep, then it fails to wake it up. Not sure if it is in my breadboard setup or code (I am thinking code). I have tried using a lot of different forums and documentation for guidance but no such luck yet. Any help would be greatly appreciated. Thank you!

### Sketch

```cpp
#define INTERRUPT_PIN 3
RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);

  //pinMode(LED, OUTPUT);
 // digitalWrite(LED, LOW);

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  print_wakeup_reason();

  Serial.println("Starting");
  delay(2000);
  
  for(int i=3; i!=0; i--){
      Serial.println(i);
      delay(1000);
  }

  Serial.println("Going to sleep now");

  esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN, ESP_GPIO_WAKEUP_GPIO_HIGH);
  esp_deep_sleep_start();

  Serial.println("This will never be printed");
}

void loop() { }

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:
      Serial.println("Wakeup caused by external signal using RTC_IO");
      break;
    case ESP_SLEEP_WAKEUP_EXT1:
      Serial.println("Wakeup caused by external signal using RTC_CNTL");
      break;
    case ESP_SLEEP_WAKEUP_TIMER:
      Serial.println("Wakeup caused by timer");
      break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
      Serial.println("Wakeup caused by touchpad");
      break;
    case ESP_SLEEP_WAKEUP_ULP:
      Serial.println("Wakeup caused by ULP program");
      break;
    default:
      Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
      break;
  }
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2370
entry 0x403ce000
Boot number: 1
Wakeup was not caused by deep sleep: 0
Starting
3
2
1
Going to sleep now
```

### Other Steps to Reproduce

I am using the Wokwi simulator since my Dev kit has not arrived yet.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-03:
It has worked with this code:

``` cpp
  Serial.println("Going to sleep now");

  esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN, ESP_GPIO_WAKEUP_GPIO_HIGH);
  gpio_set_direction((gpio_num_t)INTERRUPT_PIN, GPIO_MODE_INPUT);  // <<<=== Add this line
  esp_deep_sleep_start();

  Serial.println("This will never be printed");
}
```

abhi-sachdeva, 2023-08-03:
> It has worked with this code:
> 
> ```c++
>   Serial.println("Going to sleep now");
> 
>   esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN, ESP_GPIO_WAKEUP_GPIO_HIGH);
>   gpio_set_direction((gpio_num_t)INTERRUPT_PIN, GPIO_MODE_INPUT);  // <<<=== Add this line
>   esp_deep_sleep_start();
> 
>   Serial.println("This will never be printed");
> }
> ```

I added that line of code in the simulator and it still failed to wake up. Could it be an issue with the breadboard or Wokwi simulator? (I did go back and add esp_sleep.h library as well and no luck)

SuGlider, 2023-08-03:
>I added that line of code in the simulator and it still failed to wake up. Could it be an issue with the breadboard or Wokwi simulator? (I did go back and add esp_sleep.h library as well and no luck)

I tested it with the real ESP32-C3. It works with the actual SoC. 
Maybe the simulator can't do it right.
What is this project URL (in Wokwi)?

SuGlider, 2023-08-03:
@abhi-sachdeva - I just created a Wokwi project as described in this issue.
Indeed, it doesn't work with Wokwi.

But with the real C3, it works correctly **with and without** that Line that I mentioned.
Therefore, the code you have presented shall work fine anyway.

abhi-sachdeva, 2023-08-04:
> @abhi-sachdeva - I just created a Wokwi project as described in this issue. Indeed, it doesn't work with Wokwi.
> 
> But with the real C3, it works correctly **with and without** that Line that I mentioned. Therefore, the code you have presented shall work fine anyway.

I apologize for the delay in my response, but Thank you so much! I really appreciate you taking the time out of your day to help me debug this :)