espressif/arduino-esp32 #8491: Unable to install library

joedavid-tuni, 2023-08-04:
### Board

ESP32 Dev Board NodeMCU (ESP32-WROOM-32)

### Device Description

Plain Module

### Hardware Configuration

None

### Version

v2.0.11

### IDE Name

Arduino IDE 2.1.2

### Operating System

Ubuntu 20.04

### Flash frequency

Not sure

### PSRAM enabled

yes

### Upload speed

I dont use an upload speed as this issue is about installing the library

### Description

I am unable to install the latest  esp32 library for the Arduino 2 IDE. The same [issue](https://github.com/espressif/arduino-esp32/issues/7574) was raised but no solution was found.

### Sketch

```cpp
No Sketch really. The problem is with installing the library
```

### Debug Message

```plain
Downloading packages
esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0
esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0
esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0
esp32:xtensa-esp-elf-gdb@11.2_20220823
esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0
esp32:riscv32-esp-elf-gdb@11.2_20220823
esp32:openocd-esp32@v0.11.0-esp32-20221026
esp32:esptool_py@4.5.1
esp32:mkspiffs@0.2.3
esp32:mklittlefs@3.0.0-gnu12-dc7f933
arduino:dfu-util@0.11.0-arduino5
esp32:esp32@2.0.10
Installing esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0
Failed to install platform: 'esp32:esp32:2.0.10'.
Error: 13 INTERNAL: Cannot install tool esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 91623424 != 94397702
```

### Other Steps to Reproduce

I also tried the library version 2.0.10

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-04:
@joedavid-tuni - I think that if you try again, it may work. 
If necessary remove the previous installation and then try to install it again.

joedavid-tuni, 2023-08-04:
Well, it was a completely unrelated problem, i guess. My /home directory bloated from an Arduino log file of 39 GB. There was no place to install the library to begin with. Once i deleted the log file, and cleared up space the library installed correctly.

closed 2023-08-04 by joedavid-tuni