espressif/arduino-esp32 #8492: 'esp_wpa2_config_t' was not declared in this scope

Gabykhan, 2023-08-04:
### Board

WEMOS LOLIN32 Lite

### Device Description

None

### Hardware Configuration

HX711, DHT21, DHT22

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

?

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello,

I encounter an error message while other people who use the exact same configuration don't have this issue.
I tried reloading libraries and boards, to no avail.
The text in Arduino IDE appears in green, meaning that the function is unkown.
But I can't find the library where the 'esp_wpa2_config_t' comes from.
I'd be glad to get help from someone who encountered this issue.

Thanks!

### Sketch

```cpp
#include <WiFi.h>
#include "esp_wpa2.h"

void WIFIInit() {
  WiFi.disconnect();
  // If 802.1x mode
  if (strlen(WIFI_LOGIN) > 1) {
    Serial.println("Wifi 802.1x mode");
    WiFi.mode(WIFI_STA);
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)WIFI_LOGIN, strlen(WIFI_LOGIN));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)WIFI_PASSWORD, strlen(WIFI_PASSWORD));
    esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config settings to default
    esp_wifi_sta_wpa2_ent_enable(&config); //set config settings to enable function
    WiFi.begin(WIFI_SSID); //connect to wifi
  } else {
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  }

  short wifi_count = 0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    wifi_count++;
    if (wifi_count > 20) {
      if (ignore_wifi_error) {
        Serial.println("Can't connect to WiFi, ignoring internet related stuff...");
        return;
      } else {
        Serial.println("Can't connect to WiFi, disconnecting and resetting...");
        WiFi.disconnect();
        ESP.restart();
      }
    }
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

char* get_datetime() {
  struct tm timeinfo;
  if (!getLocalTime(&timeinfo)) {
    Serial.println("Failed to obtain time");
    return "unknown";
  }
  char * ret = (char*)malloc(128);
  *ret = '\0';
  sprintf(ret, "%d-%d-%d_%dh%d", timeinfo.tm_mday, timeinfo.tm_mon + 1, 1900 + timeinfo.tm_year, timeinfo.tm_hour, timeinfo.tm_min);
  return ret;
}
```

### Debug Message

```plain
In function 'void WIFIInit()':
wifi_functions.ino:17:5: error: 'esp_wpa2_config_t' was not declared in this scope
wifi_functions.ino:18:35: error: 'config' was not declared in this scope
Compilation error: 'esp_wpa2_config_t' was not declared in this scope
```

### Other Steps to Reproduce

I tried on two different computers, and one with a clean Arduino IDE install.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-04:
This code is taken from the [ESP-IDF example](https://github.com/espressif/esp-idf/blob/v4.4.5/examples/wifi/wifi_enterprise/main/wifi_enterprise_main.c).  It shows the correct way to do this using IDF 4.4.5

SuGlider, 2023-08-08:
@Gabykhan - These functions are part of WiFi Supplicant Module when using TLS.
Look at the examples that use it:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecureEnterprise/WiFiClientSecureEnterprise.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/HTTPClientEnterprise/HTTPClientEnterprise.ino

If you need to use it anyway, please change your code to:
``` cpp
void WIFIInit() {
  WiFi.disconnect();
  // If 802.1x mode
  if (strlen(WIFI_LOGIN) > 1) {
    Serial.println("Wifi 802.1x mode");
    WiFi.mode(WIFI_STA);
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)WIFI_LOGIN, strlen(WIFI_LOGIN));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)WIFI_PASSWORD, strlen(WIFI_PASSWORD));
    esp_wifi_sta_wpa2_ent_enable(); //set config settings to enable function
    WiFi.begin(WIFI_SSID); //connect to wifi
  } else {
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
 } 
```

SuGlider, 2023-08-08:
@Gabykhan - 

``` cpp
    esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config settings to default
    esp_wifi_sta_wpa2_ent_enable(&config); //set config settings to enable function
```

is used with Arduino Core 1.0.6 (2018/19), not with the most recent Arduino Core 2.0.11 (2023).

VojtechBartoska, 2023-08-18:
any updates @Gabykhan?

Gabykhan, 2023-09-19:
Thanks @SuGlider for your input, it's definitely the answer to this issue.
@VojtechBartoska The issue is solved, thanks!