espressif/arduino-esp32 #8494: BLE communication leads to WDT reset

vsky279, 2023-08-07:
### Board

Heltec LoRa 32 (v3) (ESP32-S3)

### Device Description

Heltec LoRa 32 (v3)

### Hardware Configuration

no

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

In project SoftRF (https://github.com/lyusupov/SoftRF) I am using the BLE communication to LXNAV device or to XCSoar on phone.
Random WDT resets occur when `m_semaphore.wait()` remains stuck.

Adjustment of all `m_semaphore.take()` with the same routine plus a timeout seem to solve the issue very well.

A simple replacement with of the following regexp solves the issue
Find what: `m_semaphore(.*?)\.take\("(.*?)"\);`
Replace with: `+ m_semaphore$1.take\(500, "$2"\);`

The following files are affected:
* [/arduino-esp32/libraries/BLE/src/BLEClient.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp)
* [/arduino-esp32/libraries/BLE/src/BLEDescriptor.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEDescriptor.cpp)
* [/arduino-esp32/libraries/BLE/src/BLECharacteristic.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.cpp)
* [/arduino-esp32/libraries/BLE/src/BLERemoteDescriptor.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteDescriptor.cpp)
* [/arduino-esp32/libraries/BLE/src/BLERemoteCharacteristic.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteCharacteristic.cpp)
* [/arduino-esp32/libraries/BLE/src/BLEScan.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEScan.cpp)
* [/arduino-esp32/libraries/BLE/src/BLEServer.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEServer.cpp)
* [/arduino-esp32/libraries/BLE/src/BLEService.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEService.cpp)

I can submit a PR if it was welcome.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

scripter-co, 2023-10-10:
what's the reason for the 500ms? seems like this is more of a patch to the issue than fixing the problem?

vsky279, 2023-10-10:
There is no specific reason for 500ms. I wanted to give enough time to BLE to finish its task and still not to end in WDT reset.

I definitely agree that this is patch rather than fix of the root cause.

I wonder if the patch will work for you.

vsky279, 2023-11-05:
I realized this is not a solution. I'm closing the issue.

closed 2023-11-05 by vsky279