espressif/arduino-esp32 #8495: Problem with PSRAM

samuelatceon, 2023-08-07:
### Board

ESP32-WROVER-IE-N16R8

### Device Description

this is a development board that is planned to be used in conjunction with the SIM7600CE LTE chip, and CAN and RS 485 chips for other communication purposes.  

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

when uploading a simple hello world program to the esp32-wrover-ie-n16r8, it accepts the new software, however upon restating, on the aerial monitor i get this message E (88) psram: PSRAM ID read error: 0x7ca9ae06, this adress has been seen to change at times. 

ive given a screen shot of the current settings that i have been using. 

[image]

### Sketch

```cpp
I am just using a simple "hello World!" program.,
```

### Debug Message

```plain
https://user-images.githubusercontent.com/141648517/258825997-43e0a0d5-100b-48ba-8589-9b4affef6d13.png
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-08-07:
this suggests an issue with the PSRAM itself (or the module). Does it work if you set the frequency to 40MHz? Have you tried another board/module?

podaen, 2023-08-08:
Or you try to map more than 4MB.

samuelatceon, 2023-08-08:
> this suggests an issue with the PSRAM itself (or the module). Does it work if you set the frequency to 40MHz? Have you tried another board/module?

Thank you for your suggestion, i have another board, but with the same issue.

samuelatceon, 2023-08-08:
> Or you try to map more than 4MB.

do you have any information on how to do that, cause ive tried looking for the boards.txt file through the file manager without any luck?

samuelatceon, 2023-08-08:
[image]
this seems to be the main issue, i thought my making sure PSRAM in the tools page is enabled would have ensure this.

closed 2023-08-08 by samuelatceon

podaen, 2023-08-08:
> [image] this seems to be the main issue, i thought my making sure PSRAM in the tools page is enabled would have ensure this.

He dit that. Look at this description.

> > Or you try to map more than 4MB.
> 
> do you have any information on how to do that, cause ive tried looking for the boards.txt file through the file manager without any luck?

Are you trying to do it your self? PSRAM is a separate chip on your Esp board. It has range of datablocks you can write in. If you try to access without that range you will get random values after.

samuelatceon, 2023-08-08:
> @samuelatceon - What Arduino Core version is being used?
im not entirely sure what you mean by core version 
[image]

samuelatceon, 2023-08-08:
> I see now. You have selected the wrong board. Please take "ESP Dev Module". I know it is weired because there are profiles like "ESP Wrover module" that doesn't have PSRAM and there is "ESP32 Wrover kit (all versions)" that didn't work for me too.

unfortunately, changing the board hasnt made any changes to outcome 
[image]

samuelatceon, 2023-08-09:
please refer to this thread #8505