espressif/arduino-esp32 #8497: Will there be a partition table using the 32Mb of ESP32-S3 Dev-board

thomaspapad94, 2023-08-07:
### Related area

Partition table

### Hardware specification

ESP32-S3-DevKitC-1-N32R8V

### Is your feature request related to a problem?

Hello, I'm currently working on developing a rather intricate codebase for an ESP32-S3 microcontroller. My aim is to fully leverage the capabilities of the board. I'm utilizing the PlatformIO environment for coding. While I managed to get my board operational through some research, I encountered a hurdle in utilizing the entire 32Mb of flash memory effectively.

### Describe the solution you'd like

I would greatly appreciate it if you could provide a default partition table for the 32Mb flash memory. Currently, there are existing options available for 16Mb flash boards, and having a similar default configuration for the larger memory capacity would be immensely helpful.

### Additional context

Here is my current platformio.ini configuration, which is functional:

```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_upload.flash_size = 16MB
board_build.partitions = large_spiffs_16MB.csv
monitor_speed = 115200
board_build.flash_mode = dio
build_flags = 
    -DBOARD_HAS_PSRAM 
    -mfix-esp32-psram-cache-issue
board_build.arduino.memory_type = opi_opi
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-08-07:
This partition is getting to the size where I would not recommend using SPIFFS.

thomaspapad94, 2023-08-07:
I came across information suggesting that the Arduino-ESP32 library might incorporate support for both 16MB and 32MB ESP32-S3 boards in a forthcoming v3.x version. Could anyone provide additional confirmation on this matter? Furthermore, if these developments are accurate, is there any available information regarding the anticipated release date for this version?

thomaspapad94, 2023-08-07:
> This partition is getting to the size where I would not recommend using SPIFFS.

Thanks for the reply.
Why shouldn't I use SPIFFS? To be honest that's why I need the extra space..

lbernstone, 2023-08-07:
SPIFFS is not a hierarchical file system.  The directories are just a slash in the string name of the file, and the performance on large spaces is subpar, particularly if you are fragmenting the space at all.
LittleFS is recommended nowadays if you want to optimize for file integrity.  FAT has better seek/replace performance.

PR #8498 (2023-08-29): Added a partition table for 32MB ESP32-S3 (4.8/4.8/24)
*By completing this PR sufficiently, you help us to review this Pull Request quicker and also help improve the quality of Release Notes*

### Checklist
1. [x] Please provide specific title of the PR describing the change, including the component name (*eg. „Update of Documentation link on Readme.md“*)
2. [x] Please provide related links (*eg. Issue which will be closed by this Pull Request*)
3. [x] Please **update relevant Documentation** if applicable
4. [x] Please check [Contributing guide](https://docs.espressif.com/projects/arduino-esp [... cut]
closed 2023-08-29 by me-no-dev