espressif/arduino-esp32 #8499: Serial bug in multitasking

PM04290, 2023-08-08:
### Board

ESP32-S3

### Device Description

ESP32-S3 Devkit

### Hardware Configuration

Simple GPS card (GY-GPSV3-NEO) connected on Serial1
Same problem with FTDI (USB/UART) on same serial
Crash when serial is not connected

### Version

v2.0.11

### IDE Name

Arduino IDE 1.8.19

### Operating System

Window 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The code below is part of my large code but it show the problem 
When Serial is used in separate task, the ISR timer not working

- ISR timer at 1Hz just printing 1Hz done.
- In separate task, i have defined 2 possibilities : OK (led blinking), KO (Serial reading)

NB: if serial is not connected, it run few time (4second), and after there is UART_MUTEX crash (but not easy to have backtrace to show it)

### Sketch

```cpp
#include <Arduino.h>
#include <HardwareSerial.h>

#define SerialGPS Serial1

hw_timer_t* isr1Hz = NULL;
portMUX_TYPE Mux1Hz = portMUX_INITIALIZER_UNLOCKED;
volatile bool top1Hz = false;

void IRAM_ATTR on1Hz()
{
  portENTER_CRITICAL_ISR(&Mux1Hz);
  top1Hz = true;
  portEXIT_CRITICAL_ISR(&Mux1Hz);
}

//#define RUN_OK
#define RUN_KO

void manageGPS(void * parameter)
{
  static int n = 0;
  for (;;) // infinite loop for multitasking
  {
#ifdef RUN_OK
    digitalWrite(RGB_BUILTIN, n);
    n = 1 - n;
    delay(500);
#endif

#ifdef RUN_KO
    if (SerialGPS.available() > 0)
    {
      char c = SerialGPS.read();
      //Serial.print(c);
    }
#endif
  }
}

void setup()
{
  Serial.begin(115200);
  while (Serial.availableForWrite() == false)
  {
    delay(50);
  }

  // define 1Hz ISR
  isr1Hz = timerBegin(0, 80, true);
  timerAttachInterrupt(isr1Hz, &on1Hz, true);
  timerAlarmWrite(isr1Hz, 1000000, true);
  timerAlarmEnable(isr1Hz);

  SerialGPS.begin(9600, SERIAL_8N1, 18, 17);
  // start GPS parser
  xTaskCreate(manageGPS, "gps", 8192, NULL, 2, NULL);
  delay(200);

  Serial.println("Init done");
}

void loop()
{
  if (top1Hz)
  {
    // release indicator
    portENTER_CRITICAL(&Mux1Hz);
    top1Hz = false;
    portEXIT_CRITICAL(&Mux1Hz);

    Serial.println("1Hz");
  }
}
```

### Debug Message

```plain
nothing with serial connected because of bug,
but this when serial is disconnected

Init done
1Hz
1Hz
1Hz
1Hz
E (10133) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10133) task_wdt:  - IDLE (CPU 0)
E (10133) task_wdt: Tasks currently running:
E (10133) task_wdt: CPU 0: gps
E (10133) task_wdt: CPU 1: IDLE
E (10133) task_wdt: Aborting.

abort() was called at PC 0x42009330 on core 0

```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-08-08:
this is not a serial issue. You are running a busy loop in your GPS task and are not allowing other tasks with lower priority to run.

If you are reading from a GPS, I suggest that you use something like `String line = SerialGPS.readStringUntil('\n');` to allow other tasks to run in a mean time. You could also do something like:
```cpp
if(!SerialGPS.available()){
  delay(3); // allow other tasks to run
} else {
  while(SerialGPS.available()){
    char c = SerialGPS.read();
    // ....
  }
}
```

PM04290, 2023-08-08:
Thanks for response,
loop is not another task ? what is the priority of loop ?
when two tasks have the same priority, is there "balancing" between the two task?

and what about the serial link not connected issue?

me-no-dev, 2023-08-08:
> loop is not another task ? what is the priority of loop ?

Loop runs on Core 1, where WDT is disabled.

> when two tasks have the same priority, is there "balancing" between the two task?

You can read more [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html)

> and what about the serial link not connected issue?

Cause probably by your busy loop as well. Try fixing that first.

PM04290, 2023-08-08:
i have pinned mamageGPS on core 0 and it solve my ISR (loop) problem, but when Serial is disconnected, i still have the same problem of crash

PM04290, 2023-08-08:
with your code exemple it run without crash (with serial disconnected), not multitasking friendly, but run...
you can close issue.

closed 2023-08-08 by me-no-dev

SuGlider, 2023-08-08:
I also would suggest using `SerialGPS.onReceive(calbackFunc)` to achieve the same result with no need of a separated task. `void callbackFunc(void)` will called whenever data is received by `SerialGPS`.

Take a look at the example of 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/OnReceive_Demo/OnReceive_Demo.ino