espressif/arduino-esp32 #8501: Arduino ESP core install using VS

podaen, 2023-08-08:
### Related area

Installation

### Hardware specification

No

### Is your feature request related to a problem?

I was looking in the Arduino documentation how to install Arduino-ESP using they GitHub menu in Visual Studio, but I couldn't find. Could it be that this is missing? I did this in the past and it was working. 

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

Last time because of the missing documentation and because the document tells me to use bash what the VS Git doesn't have, I tried GIT and it worked. But now I have some certificate errors using the GitHub extension in VS, because I uninstalled GIT and the VS GitHub is looking for certificate bundles in that folder. 

[image]

This is not normal, because I was trying to install it with the build in GitHub and all settings are properly filled in to VS settings. I was planning to reset my PC because I have too many issues with signatures in to the entire MS Windows. Git is actually based on the Linux kennel and I don't want to mix those two different application up anymore.

### Describe the solution you'd like

I want to be using only Visual Studio and the tools that is now builtin. Could it please be documentation how to do this, other than using GIT, Arduino IDE, ... So I can clone this repo to my repositories. I would appreciated it!

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-08-10:
Hi @podaen , we officially support Arduino IDE and provide troubleshooting for that, but if you figure out how to get this feature of VS going, we will accept contribution to the documentation.

me-no-dev, 2023-08-10:
BTW this looks like you have an SSL issue with your computer

podaen, 2023-08-11:
> Hi @podaen , we officially support Arduino IDE and provide troubleshooting for that, but if you figure out how to get this feature of VS going, we will accept contribution to the documentation.

Ok. Thanks! I will do in here.

https://github.com/espressif/arduino-esp32/blob/master/docs/source/installing.rst

The only difference is that 'Git Bash'.

> - open a `Git Bash` session pointing to ``[ARDUINO_SKETCHBOOK_DIR]/hardware/espressif/esp32`` and execute ```git submodule update --init --recursive``` 

This is done automatically by the Git in Visual Studio using this file

https://github.com/espressif/arduino-esp32/blob/master/.gitmodules

Because the IDF bin is presents when I look in to the folder

D:\Code\VisualStudio\2022\repos\Arduino\hardware\espressif\esp32\tools\sdk

> BTW this looks like you have an SSL issue with your computer

Yes!  I see, it is a connection issue. Probably not related to signing issues. I found on the internet how to solve this by edit the config file in that folder. But it will not be a solution for me because I don use GIT anymore in Windows and I probably reconfigured the root when I install GIT. 

I found out that I added the environment pads in Windows for Git. 

[image]

Not sure it will solve it, but I need to reboot windows now...

podaen, 2023-08-11:
The git module in VS still crashes, Reinstalling Visual Studio will probably solve it!

@me-no-dev Thanks for your answer!

closed 2023-08-11 by podaen

podaen, 2023-08-11:
Uninstall and install it again didn't solve it for me. Setting the Cryptographic network provider did!

[image]

podaen, 2023-09-01:
The problem was that there where two different version of mingw-w64-x86_64-curl-winssl and VS and git are not using the same.

So I tried it an other way, by using Mingw and a git plugin and do it from there. But now my python didn't work in that terminal. So I did it manually double clicking on the py.exe. After opening VS and link the paths I got this,

[image]

> CMake Error in CMakeLists.txt:
>   No CMAKE_CXX_COMPILER could be found.
> 
>   Tell CMake where to find the compiler by setting either the environment

I think VS is looking for the wrong compiler. Can you tell me where the ESP compiler or is it just python that let me down?