espressif/arduino-esp32 #8503: why the microSD card disconnect when connect webserver to esp32

TahaNokia, 2023-08-09:
### Board

ESP32S

### Device Description

[image]

### Hardware Configuration

#define SCK  18  
#define MISO 19  
#define MOSI 23  
#define CS  5    

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

11520

### Description

I am using the microSD card module with esp32s, it is well work but when connect the esp32s to server wifi in the same code the micrSD card disconnect. Can you tell me why and how to Solve it.

### Sketch

```cpp
#include <WiFi.h> //Built-in #include <ESP32WebServer.h> //https://github.com/Pedroalbuquerque/ESP32WebServer download and place in your Libraries folder #include <ESPmDNS.h>

#include "CSS.h" //Includes headers of the web and de style file #include <SD.h> #include <SPI.h>

#define SCK 18 //48 #define MISO 19 //36 #define MOSI 23 //35 #define CS 5 //37

ESP32WebServer server(80);

#define servername "FHD33" //Define the name to your server...

bool SD_present = false; //Controls if the SD card is present or not

/********* SETUP **********/

void setup(void) {
Serial.begin(115200); WiFi.softAP("FHD", "123456789"); //Network and password for the access point genereted by ESP32 ////
 if (!MDNS.begin(servername)) {
Serial.println("Error setting up MDNS responder!"); ESP.restart(); }

Serial.print("Initializing SD card...");

if (!SD.begin(CS)) { 
Serial.println("Card failed or not present, no SD Card data logging possible..."); 
SD_present = false; 
} 
else { 
Serial.println(F("Card initialised... file access enabled...")); 
SD_present = true; 
}

server.begin();

Serial.println("HTTP server started"); }

/********* LOOP **********/

void loop(void) { 
server.handleClient(); //Listen for client connections
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
Initializing SD card...Card failed or not present, no SD Card data logging possible...
HTTP server started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-09:
@TahaNokia - Does it work correctly when the sketch is simplified to only use the SDCard (no WiFi)?

TahaNokia, 2023-08-09:
@SuGlider - Yes, it does work without wifi

SuGlider, 2023-08-09:
@P-R-O-C-H-Y - Can you please take a look at this issue? Thanks!

P-R-O-C-H-Y, 2023-08-15:
Hi @TahaNokia, I have no issues to run both WIFI + SD card as your example.
I just did not use any of external libraries so I edited:
`ESP32WebServer server(80);` to `WebServer server(80);` 
and included:
```
#include <WebServer.h>
#include <ESPmDNS.h>
```

Log from serial output:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Initializing SD card...Card initialised... file access enabled...
HTTP server started
```

VojtechBartoska, 2023-08-18:
any updates @TahaNokia?

VojtechBartoska, 2023-08-21:
Closing as expired, if needed you can reopen. Thanks.

closed 2023-08-21 by VojtechBartoska

KobeOne2, 2024-08-18:
I encountered a similar issue, and it appears that it is only not working when launching an access point:

```WiFi.softAP("FHD", "123456789");```

The SD card functions well as long as no AP is used. Do you have any idea what's going on here?