espressif/arduino-esp32 #8504: how to reference or define the reset pin

NoahSTEMStuff, 2023-08-09:
### Board

adafruit esp32 huzzah feather breakout board

### Device Description

module on breakout board, connected to an adafruit pressure sensor (Adafruit MPRLS)

### Hardware Configuration

adafruit pressure sensor (Adafruit MPRLS)

### Version

v2.0.11

### IDE Name

arduino ide, thonny

### Operating System

windows

### Flash frequency

no idea

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to define the reset pin for my adafruit pressure sensor (Adafruit_MPRLS).
I am trying to do something like:
#define RESET_PIN   -1 // set to any GPIO pin # to hard-reset on begin()
#define EOC_PIN    -1  // set to any GPIO pin to read end-of-conversion by pin
Adafruit_MPRLS mpr = Adafruit_MPRLS(RESET_PIN, EOC_PIN);

but I do not know what to put instead of -1 since there is no gpio pin for the reset pin on the esp32

### Sketch

```cpp
/*!
 * @file mprls_simpletest.ino
 *
 * A basic test of the sensor with default settings
 * 
 * Designed specifically to work with the MPRLS sensor from Adafruit
 * ----> https://www.adafruit.com/products/3965
 *
 * These sensors use I2C to communicate, 2 pins (SCL+SDA) are required
 * to interface with the breakout.
 *
 * Adafruit invests time and resources providing this open source code,
 * please support Adafruit and open-source hardware by purchasing
 * products from Adafruit!
 *
 * Written by Limor Fried/Ladyada for Adafruit Industries.  
 *
 * MIT license, all text here must be included in any redistribution.
 *
 */
 
#include <Wire.h>
#include "Adafruit_MPRLS.h"

// You dont *need* a reset and EOC pin for most uses, so we set to -1 and don't connect
#define RESET_PIN   rst // set to any GPIO pin # to hard-reset on begin()
#define EOC_PIN    -1  // set to any GPIO pin to read end-of-conversion by pin
Adafruit_MPRLS mpr = Adafruit_MPRLS(RESET_PIN, EOC_PIN);

void setup() {
  Serial.begin(115200);
  Serial.println("MPRLS Simple Test");
  if (! mpr.begin()) {
    Serial.println("Failed to communicate with MPRLS sensor, check wiring?");
    while (1) {
      delay(10);
    }
  }
  Serial.println("Found MPRLS sensor");
}

void loop() {
  float pressure_hPa = mpr.readPressure();
  Serial.print("Pressure (hPa): "); Serial.println(pressure_hPa);
  Serial.print("Pressure (PSI): "); Serial.println(pressure_hPa / 68.947572932);
  delay(1000);
}
```

### Debug Message

```plain
there is no message since I do not know what to put, so I did not run anything
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-09:
@NoahSTEMStuff - The default is -1 for both. Just Wire the I2C and VCC/GND.

You can read the tutorial in https://learn.adafruit.com/adafruit-mprls-ported-pressure-sensor-breakout

Try it..

NoahSTEMStuff, 2023-08-09:
sorry if this was not clear. I am able to get the default script to run. I was trying to see if it was possible to make the reset pin for the pressure sensor be the reset pin on the esp32 itself, and the reset pin for the esp32 does not have a gpio pin number, hence my confusion.

SuGlider, 2023-08-09:
Ok, actually you will not use the RESET pin from the ESP32.
You can pick any free GPIO from the ESP32 to wire it to the RESET pin in the MPRLS board and use this GPIO as the value for RESET_PIN.
The same for EOC.

The Adafruit library will initialize both GPIO and use them correctly.

SuGlider, 2023-08-14:
@NoahSTEMStuff - Please let me know if we can close this issue. Thanks.

closed 2023-08-17 by SuGlider