espressif/arduino-esp32 #8505: The Arduino library cannot detect the PSRAM of the ESP32-WROVER-IE-N16R8 module

Rainbow-Cai, 2023-08-09:
### Board

ESP32-WROVER-IE-N16R8

### Device Description

ESP32-WROVER-IE-N16R8 with our board

### Hardware Configuration

The PSRAM pins is not lead out

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

windows

### Flash frequency

40MHz or 80MHz all tested

### PSRAM enabled

yes

### Upload speed

115200

### Description

When i use the follows code to get PSRAM  informations, i can not get the PSRAM information, as follows: 

[image]

I am currently using the most up to date board package from Espressif,

And using stable release link

[image]

### Sketch

```cpp
#include <Arduino.h>
void setup() {

//Use this code to init the pSARM
  Serial.begin(115200);

  setCpuFrequencyMhz(160); 

   // 获取 Flash 大小（以字节为单位）
  uint32_t flashSize = ESP.getFlashChipSize();
  Serial.printf("Flash Size: %u bytes\n", flashSize);

  // 获取 PSRAM 的大小（以字节为单位）
  // uint32_t psramSize = ESP.getMaxAllocPsram();
  // Serial.printf("PSRAM Size: %u bytes\n", psramSize);

    // 获取 PSRAM 的大小（以字节为单位）
  uint32_t psramSize = ESP.getPsramSize();
  Serial.printf("PSRAM Size: %u bytes\n", psramSize);

  // 获取剩余内存
  uint32_t freeMemory = ESP.getFreeHeap();
  Serial.printf("Free Memory: %u bytes\n", freeMemory);

   // 获取总内存
  uint32_t Totalheap = ESP.getHeapSize();
  Serial.printf("Total heap: %u bytes\n", Totalheap);

   // 获取 PSRAM 剩余内存
  uint32_t FreePSRAM = ESP.getFreePsram();
  Serial.printf("Free PSRAM: %u bytes\n", FreePSRAM);
  

}

void loop() {}
```

### Debug Message

```plain
But when i use the esp-idf v5.0 firmware ， i can get the PSRAM informations.  I'm using the same module. 
![image](https://github.com/espressif/arduino-esp32/assets/62236328/5314248d-0807-4a91-a2d5-68c813374215)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

samuelatceon, 2023-08-09:
Here is a schematic of the connections, i dont think there is any issues with this. 

[image]

samuelatceon, 2023-08-09:
Also ran this code, the result can be shown below, it will continuously reboot no matter the configuration. 

`int n_elements = 1000;

void setup() {
// put your setup code here, to run once:
  `Serial.begin(115200)
  //Init
  if(psramInit()){
  Serial.println("\nPSRAM is correctly initialized");
  }else{
  Serial.println("PSRAM not available");
  }

  //Create an array of n_elements
  int available_PSRAM_size = ESP.getFreePsram();
  Serial.println((String)"PSRAM Size available (bytes): " + available_PSRAM_size);

  int *array_int = (int *) ps_malloc(n_elements * sizeof(int)); //Create an integer array of n_elements
  array_int[0] = 42;
  array_int[999] = 42; //We access array values like classic array

  int available_PSRAM_size_after = ESP.getFreePsram();
  Serial.println((String)"PSRAM Size available (bytes): " + available_PSRAM_size_after); // Free memory space has decreased
  int array_size = available_PSRAM_size - available_PSRAM_size_after;
  Serial.println((String)"Array size in PSRAM in bytes: " + array_size);

  //Delete array
  free(array_int); //The allocated memory is freed.
  Serial.println((String)"PSRAM Size available (bytes): " +ESP.getFreePsram());
}

void loop() {
  // put your main code here, to run repeatedly:

}
`

[image]

SuGlider, 2023-08-09:
@Rainbow-Cai / @samuelatceon - I don't have a WROVER-IE-N16-R8 here in order to test it. I just have a WROVER-E.
Could you please try using Arduino Core 2.0.9 instead and let me know the results?

Arduino Core 2.0.9 uses IDF 4.4.4
Arduino Core 2.0.10 and 2.0.11 use IDF 4.4.5.

Thanks!

SuGlider, 2023-08-10:
I have tried it with the ESP32-WROVER-E-N8R8 and I got no error.
This is the output:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (157) esp_core_dump_flash: Incorre⸮~⸮size of core dump image: 1559798519
[     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   455][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   482][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Flash Size: 8388608 bytes
PSRAM Size: 4192107 bytes
Free Memory: 309432 bytes
Total heap: 334636 bytes
Free PSRAM: 4191847 bytes
```

samuelatceon, 2023-08-10:
> I have tried it with the ESP32-WROVER-E-N8R8 and I got no error. This is the output:
>
> ```
> ets Jul 29 2019 12:21:46
>
> rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee [...]

Hi, what settings in your tools page did you use, just so i can recreate what youve done

SuGlider, 2023-08-10:
@samuelatceon - These are the settings within the Arduino IDE 1.8.15 and Arduino Core 2.0.11:

[image]

SuGlider, 2023-08-10:
@Rainbow-Cai @samuelatceon - I'm not able to replicate this issue.

Rainbow-Cai, 2023-08-11:
Hi  @samuelatceon 

 When is uesd the ESP32-WROVER-IE-N16R8 module for testing, it is work OK.
[image]

podaen, 2023-08-11:
Seems that there is a lot misconception about this module and different versions there are.

ESP32-WROVER-IE N8R4. (SuGlider)

N8 means 8MB of flash 
R4 means 4MB of PSRAM

ESP32-WROVER-IE-N16R8 (Rainbow-Cai and samuelatceon)

N16 means 16MB of flash
R8 means 8MB of PSRAM

> Options there are for Embedded Flash and PSRAM
> 
>     R2 2MB PSRAM (QSPI)
>     R8 8MB PSRAM (OPI)
>     V 1.8V SPI
> 
> See here! https://github.com/espressif/arduino-esp32/blob/master/docs/source/guides/tools_menu.rst#psram

> [image]

When I look to the photo that SuGlider posts, I see no PSRAM chip on the board, So for me it is embedded in to the Rover module, but it has only 4MB of PSRAM (should be default).

> [image]

I don't see any "option" like the documentation tells us where we can select OPI for 8MB of PSRAM. So what could it be... Probably in the code somewhere. Should be in the menu somehow. There are selected different flash modes for that board QIO, DIO, QOUT, DOUT, but that doesn't seems it!

@Rainbow-Cai It seems you are using only 4MB of the total 8MB PSRAM you have, but you have your 16MB of flash.

---------------------------------------------

I did some further research and found this in the board.txt for the esp32c3

> esp32s3.menu.PSRAM.disabled=Disabled
> esp32s3.menu.PSRAM.disabled.build.defines=
> esp32s3.menu.PSRAM.disabled.build.psram_type=qspi
> esp32s3.menu.PSRAM.enabled=QSPI PSRAM
> esp32s3.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM
> esp32s3.menu.PSRAM.enabled.build.psram_type=qspi
> esp32s3.menu.PSRAM.opi=OPI PSRAM
> esp32s3.menu.PSRAM.opi.build.defines=-DBOARD_HAS_PSRAM
> esp32s3.menu.PSRAM.opi.build.psram_type=opi

This I can not find for the Wrover

----------------------------------------

I think there must be more lines add to the Wrover, because for me only the "ESP32 Dev Module" was working... And what I found that is

> esp32.menu.JTAGAdapter.external.build.openocdscript=esp32-wrover-kit-3.3v.cfg
> 
> See here! https://github.com/espressif/arduino-esp32/blob/e1f14331f173a00a9062f616bc9a62c358b9076f/boards.txt#L652C4-L652C4

Strange, Because I connect directly to the Wrover board without using an external adapter. 

---------------------------------------

Here are the links to the Wrover Module/(boards) in board.txt and the related settings.

- ESP32 Wrover Module

https://github.com/espressif/arduino-esp32/blob/e1f14331f173a00a9062f616bc9a62c358b9076f/boards.txt#L951

for the ESP32 Wrover Module nothing can be found, still it could be embedded with 8MB of PSRAM.

- ESP32 Wrover Kit (all versions)

> esp32wroverkit.menu.PSRAM.enabled=Enabled
> esp32wroverkit.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
> esp32wroverkit.menu.PSRAM.disabled=Disabled
> esp32wroverkit.menu.PSRAM.disabled.build.defines=

https://github.com/espressif/arduino-esp32/blob/e1f14331f173a00a9062f616bc9a62c358b9076f/boards.txt#L1612

We should in have there something like `.menu.PSRAM.opi.` It cost me a couple of hours to figure this all out... But I think we know how to solve this. :-)

samuelatceon, 2023-08-11:
> Hi @samuelatceon
> 
> When is uesd the ESP32-WROVER-IE-N16R8 module for testing, it is work OK. [image]

Hi, 

when i use the same settings, im still getting the same result as previous>

[image]

podaen, 2023-08-11:
@Rainbow-Cai Would you like to you send me your shematic?  I will do the coding and testing if  your shematic is completely fine to me.

samuelatceon, 2023-08-11:
> Seems that there is a lot misconception about this module and different versions there are.
>
> ESP32-WROVER-IE N8R4. (SuGlider)
>
> N8 means 8MB of flash R4 means 4MB of PSRAM
>
> ESP32-WROVER-IE-N16R8 [...]

Hi, 

Ive tried this to the best of my ability in comparing esp32wrover to esp32s3 build.txt,  i created a new board by copy/pasting and changing the name at the start to make it identifiable, but i still get the same issue. 

esp32wrover.name=ESP32 Wrover Module SAM

esp32wrover.bootloader.tool=esptool_py
esp32wrover.bootloader.tool.default=esptool_py

esp32wrover.upload.tool=esptool_py
esp32wrover.upload.tool.default=esptool_py
esp32wrover.upload.tool.network=esp_ota

esp32wrover.upload.maximum_size=1310720
esp32wrover.upload.maximum_data_size=327680
esp32wrover.upload.flags=
esp32wrover.upload.extra_flags=

esp32wrover.serial.disableDTR=true
esp32wrover.serial.disableRTS=true

esp32wrover.build.tarch=xtensa
esp32wrover.build.bootloader_addr=0x1000
esp32wrover.build.target=esp32
esp32wrover.build.mcu=esp32
esp32wrover.build.core=esp32
esp32wrover.build.variant=esp32
esp32wrover.build.board=ESP32_DEV

esp32wrover.build.f_cpu=240000000L
esp32wrover.build.flash_size=16MB
esp32wrover.build.flash_freq=40m
esp32wrover.build.flash_mode=qio
esp32wrover.build.boot=qio
esp32wrover.build.partitions=default
esp32wrover.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
esp32wrover.build.defines=
esp32wrover.build.loop_core=
esp32wrover.build.event_core=
esp32wrover.build.psram_type=qspi
esp32wrover.build.memory_type={build.boot}_{build.psram_type}

esp32wrover.menu.PSRAM.disabled=Disabled
esp32wrover.menu.PSRAM.disabled.build.defines=
esp32wrover.menu.PSRAM.disabled.build.psram_type=qspi
esp32wrover.menu.PSRAM.enabled=QSPI PSRAM
esp32wrover.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM
esp32wrover.menu.PSRAM.enabled.build.psram_type=qspi
esp32wrover.menu.PSRAM.opi=OPI PSRAM
esp32wrover.menu.PSRAM.opi.build.defines=-DBOARD_HAS_PSRAM
esp32wrover.menu.PSRAM.opi.build.psram_type=opi

esp32wrover.menu.PartitionScheme.default=Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
esp32wrover.menu.PartitionScheme.default.build.partitions=default
esp32wrover.menu.PartitionScheme.defaultffat=Default 4MB with ffat (1.2MB APP/1.5MB FATFS)
esp32wrover.menu.PartitionScheme.defaultffat.build.partitions=default_ffat
esp32wrover.menu.PartitionScheme.default_8MB=8M with spiffs (3MB APP/1.5MB SPIFFS)
esp32wrover.menu.PartitionScheme.default_8MB.build.partitions=default_8MB
esp32wrover.menu.PartitionScheme.default_8MB.upload.maximum_size=3342336
esp32wrover.menu.PartitionScheme.minimal=Minimal (1.3MB APP/700KB SPIFFS)
esp32wrover.menu.PartitionScheme.minimal.build.partitions=minimal
esp32wrover.menu.PartitionScheme.no_ota=No OTA (2MB APP/2MB SPIFFS)
[... 51 lines cut ...]
esp32wrover.menu.UploadSpeed.460800.linux=460800
esp32wrover.menu.UploadSpeed.460800.macosx=460800
esp32wrover.menu.UploadSpeed.460800.upload.speed=460800
esp32wrover.menu.UploadSpeed.512000.windows=512000
esp32wrover.menu.UploadSpeed.512000.upload.speed=512000

esp32wrover.menu.DebugLevel.none=None
esp32wrover.menu.DebugLevel.none.build.code_debug=0
esp32wrover.menu.DebugLevel.error=Error
esp32wrover.menu.DebugLevel.error.build.code_debug=1
esp32wrover.menu.DebugLevel.warn=Warn
esp32wrover.menu.DebugLevel.warn.build.code_debug=2
esp32wrover.menu.DebugLevel.info=Info
esp32wrover.menu.DebugLevel.info.build.code_debug=3
esp32wrover.menu.DebugLevel.debug=Debug
esp32wrover.menu.DebugLevel.debug.build.code_debug=4
esp32wrover.menu.DebugLevel.verbose=Verbose
esp32wrover.menu.DebugLevel.verbose.build.code_debug=5

esp32wrover.menu.EraseFlash.none=Disabled
esp32wrover.menu.EraseFlash.none.upload.erase_cmd=
esp32wrover.menu.EraseFlash.all=Enabled
esp32wrover.menu.EraseFlash.all.upload.erase_cmd=-e

[image]

SuGlider, 2023-08-11:
@podaen @Rainbow-Cai 

>I don't see any "option" like the documentation tells us where we can select OPI for 8MB of PSRAM. So what could it be... Probably in the code somewhere. Should be in the menu somehow. There are selected different flash modes for that board QIO, DIO, QOUT, DOUT, but that doesn't seems it!
>We should in have there something like .menu.PSRAM.opi. It cost me a couple of hours to figure this all out... But I think we know how to solve this. :-)

Only ESP32-S3 has OPI. We are talking about the ESP32 SoC. It has no OPI and PSRAM is QIO.

SuGlider, 2023-08-11:
@Rainbow-Cai - Do you have other ESP32 modules with PSRAM there with you? This issue happen for all of them?

podaen, 2023-08-11:
> Only ESP32-S3 has OPI. We are talking about the ESP32 SoC. It has no OPI and its PSRAM is QIO/DIO.

Then why we see on all the wrovers 4MB of PSRAM?

lbernstone, 2023-08-11:
@podaen  The ESP32 does not have the ability to directly address more than 4MB psram, so you must use the himem library for 8MB.  The newer variants do not have this constraint.

podaen, 2023-08-11:
Never heard of it! Hmm...

https://github.com/espressif/arduino-esp32/blob/e1f14331f173a00a9062f616bc9a62c358b9076f/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32/himem.h

Can do something here in the Arduino Core

By making a config file like this

https://github.com/espressif/openocd-esp32/blob/4da79b3e11da3de16b393fe1415207f48be98a0c/tcl/board/esp32-wrover-kit-3.3v.cfg#L14

For each PSRAM value and write an Arduino lib that set this on the first run?

lbernstone, 2023-08-11:
No, it requires significant changes to the sdk libraries.  You would need to compile them for yourself.  If you need to use 8MB psram in arduino, you should use an ESP32-S3.

podaen, 2023-08-11:
Should or could? You are probably right. I would like the idea to have it!

SuGlider, 2023-08-11:
`Himem` is currently incompatible with the ESP32 Arduino Core. It conflicts with PSRAM automatic allocation in `malloc()`.

Therefore, the only way to use `Himem` is by using Arduino as IDF Component and by creating an appropriate sdkconfig that be compatible. It also would require disabling all PSRAM settings in Arduino Core.

SuGlider, 2023-08-11:
>ESP32-WROVER-IE-N16R8 with our board

@Rainbow-Cai @samuelatceon - using DevKits, it is not possible to reproduce this issue. 
I understand that you have a custom hardware/board. Could it be some power/electrical problem instead? 
Or an error in the PCB?

podaen, 2023-08-11:
> `Himem` is currently incompatible with the ESP32 Arduino Core. It conflicts with PSRAM automatic allocation in `malloc()`.

What I understand from this is those functions  need to have a 'start' and size in Himem...

podaen, 2023-08-11:
I did some research and I found this

> However, because of hardware limitations, this only works for up to 4 MiB of external memory. If you have, for instance, an ESP32-WROVER module with 8 MiB of PSRAM, you cannot use the upper 4 MiB of PSRAM this way. However, this memory is not wasted, using the Himem API (which essentially is a bank switching scheme for the upper memory regions), the upper 4 MiB is still usable.

So, to use the full 8MB for a classic ESP this is the only why.

> Therefore, the only way to use `Himem` is by using Arduino as IDF Component and by creating an appropriate sdkconfig that be compatible. It also would require disabling all PSRAM settings in Arduino Core.

I have seen this before with classic Bluetooth and I believe there is no way in doing this whiten the Arduino core. This can't be don in the Arduino core. And if we look back to what we have for the extreme N16R8 is 16MB of flash and 4MB of PSRAM. We only loose 4MB of PSRAM and still can use the Arduino core. So I would go for that!

I did the test with my board

Opening port
Port open
[   450][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
VMDPR_[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 260
[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 500260
[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 260
PSRAM Size: 4192123 bytes
Free Memory: 299992 bytes
Total heap: 325092 bytes
Free PSRAM: 4191863 bytes

using this shetch

```
#include <Arduino.h>

void logMemory() {
	log_d("Used PSRAM: %d", ESP.getPsramSize() - ESP.getFreePsram());
}

void setup() {
	logMemory();
	byte* psdRamBuffer = (byte*)ps_malloc(500000);
	logMemory();
	free(psdRamBuffer);
	logMemory();

	uint32_t psramSize = ESP.getPsramSize();
	Serial.printf("PSRAM Size: %u bytes\n", psramSize);
	uint32_t freeMemory = ESP.getFreeHeap();
	Serial.printf("Free Memory: %u bytes\n", freeMemory);
	uint32_t Totalheap = ESP.getHeapSize();
	Serial.printf("Total heap: %u bytes\n", Totalheap);
	uint32_t FreePSRAM = ESP.getFreePsram();
	Serial.printf("Free PSRAM: %u bytes\n", FreePSRAM);
}

void loop() {}
```

So I think, @samuelatceon has an electrical issue. And when try to write more in to the  'ESP Wrover Kit' doesn't give an error. It simply refuse to write in it.

Port open
[   450][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
VMDPR_[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 260
[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 260
[  4480][D][PSRAM.ino:10] logMemory(): Used PSRAM: 260
PSRAM Size: 4192123 bytes
Free Memory: 299992 bytes
Total heap: 325092 bytes
Free PSRAM: 4191863 bytes

And the same for the 'ESP Dev module' in the latest.

lbernstone, 2023-08-12:
Absolutely.  Nothing says you have to use every feature :)

VojtechBartoska, 2023-08-18:
Closing this as answered, if needed you can reopen. Thanks everyone for help!

closed 2023-08-18 by VojtechBartoska

me-no-dev, 2023-08-22:
What I see in the logs and conversation so far, leads me to believe that either something is connected to one of the PSRAM pins, or that the PSRAM chips are bad, or that something in the module was not soldered correctly (had that happen before). Even if PSRAM is 8MB, it will still start up and allow for the lower 4MB to be used as usual. It will not allow the use of the upper 4MB, but it will boot and pass checks for 4MB.