espressif/arduino-esp32 #8506: Programming ESP32-C3-MINI-1 chip

abhi-sachdeva, 2023-08-09:
### Board

ESP32-C3-MINI-1

### Device Description

Hello! I have been looking around at every resource I could to figure out how to program the esp32-c3-mini-1 chip. I realize I need a UART to usb bridge within m a programmer to accomplish this. However I am getting conflicting points about what pinouts I need to do so. I have been told by one person that all I need is  TXD, RXD, GND to program the chip. Another said that I need TXD, RXD, RTS, DRT, GND (RTS for a connection to enable and DRT for a connection to GPIO9). 

I am not sure which is correct for me to be able to flash a ESP-32-C3-MINI-1. I have found multiple programmers such as: 
https://www.analoglamb.com/product/esp32-esp82-programmer-ch340c-breakout/
https://www.walmart.com/ip/USB-To-TTL-Serial-Module-Serial-Converter-Module-300bps-1Mbps-Baud-Rate-Serial-Adapter-For-DIY-Teaching/1916205152?wmlspartner=wlpa&selectedSellerId=101296366

Any help on understanding what programmer I need is really appreciated, thanks!

### Hardware Configuration

GPIO9 is the strapping pin for the ESP32-C3 thus the connection of DRT (or in the case of analoglamb GPIO0) would be GPIO9

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

115200

### Description

I do not understand which programmer to obtain for flashing the esp32-c3-mini-1 chip. I want to be able to program via breadboard and minimize what I need to do on the pcb. 

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-09:
If you just need to transfer data across the uart for your external programmer, tx and rx will work fine and a manual jumper on gpio0. You will likely need to cycle power after uploading.
If you plan on using this to develop code and be regularly uploading firmwares (like a development board), you will want transistors on the rts and dts to help reset the device properly.  See the lower left corner of the [devkitc schematic](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf)

abhi-sachdeva, 2023-08-09:
> If you just need to transfer data across the uart for your external programmer, tx and rx will work fine and a manual jumper on the boot pin. You will likely need to cycle power after uploading. If you plan on using this to develop code and be regularly uploading firmwares (like a development board), you will want transistors on the rts and dts to help reset the device properly. See the lower left corner of the [devkitc schematic](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf)

So then would it just be best to purchase a ESP-PROG so that I can write across the UART while having it handle the reset on its own? I do plan to test and upload multiple sketches until I can get the device to function as intended. The other option was the analoglamb board too. I would want to steer away from modifying my pcb to accomodate programming since space is extremely limited

lbernstone, 2023-08-09:
For development you should buy a development board.
There's a lot of variables there.  Honestly, on a C3, you should be thinking about using the usb port for programming in production.  Much easier way to program, the port costs a couple cents, requires no extra hardware (beyond the power supply which you likely need anyhow), and may be useful to you in other ways.

abhi-sachdeva, 2023-08-09:
> For development you should buy a development board. There's a lot of variables there. Honestly, on a C3, you should be thinking about using the usb port for programming in production. Much easier [...]

I do have a ESP32-C3-DevKitM1, however, the dev kit cannot program a standalone chip. The chip itself is what will be integrated into the device, the dev kit is just a tool for creating a proof of concept on a breadboard. I agree it with your statement that there should be a usb for programming, but in the current iteration of the design, there simply is no space for one. Which is why I ask for a particular programmer that can be used to program the chip instead of me having to update the pcb to accomodate the space for a usb. Additionally, my power supply is meant for a single use scenario where the esp is asleep most of its life and wakes up once and it is exhausted. Thus, I cannot even accomodate a bigger power supply.

Since there is no way to use the usb - uart on the dev kit to program a stand alone chip, would you then recommend a ESP-PROG?

abhi-sachdeva, 2023-08-09:
> No, your original idea will work, just be much more of a difficult connection than a micro-usb. On the pre-USB devices, I typically had a set of programming pins that exposed only tx, rx, boot, en, 3v3 and gnd. It was 1.27pitch pin header, and I would plug that in to program. Worked fine with an external programmer, although mine was a cp2102 and had enough power to program the device, but not enough to run the wifi :) .

To reiterate completely just so I understand what you are saying: No need for the ESP-PROG. Simply a programmer such as https://www.analoglamb.com/product/esp32-esp82-programmer-ch340c-breakout/ or
https://www.walmart.com/ip/FT232RL-FTDI-USB-to-TTL-Serial-Adapter-Module-Compatible-for-Mini-Port-3-3V-5-5V/1513489607?wmlspartner=wlpa&selectedSellerId=16214
is better for my purposes. The pins I would be using are the 3v3 for power, GND, TX, RX, En, and a GPIO9 for the boot?

abhi-sachdeva, 2023-08-09:
I appreciate all your help! Thank you :)

closed 2023-08-09 by abhi-sachdeva