espressif/arduino-esp32 #8508: ESP32-S3 SPI spiAttach... spiDetach... function use wrong pins

UlliBien, 2023-08-10:
### Board

ESP32-S3

### Device Description

any

### Hardware Configuration

doesn't matter

### Version

v2.0.11

### IDE Name

Arduino-IDE

### Operating System

Windows 11

### Flash frequency

doesn't matter

### PSRAM enabled

yes

### Upload speed

doesn't matter

### Description

spiAttachMOSI attaches pin 35 instead of pin 11.
   `#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3` 
is wrong. Later is a separate statement for the ESP32-S3 with the correct pin assignment: 
   `#elif CONFIG_IDF_TARGET_ESP32S3`

spiDetachSCK, spiDetachMISO, spiDetachMOSI the same.

### Sketch

```cpp
doesn't matter
```

### Debug Message

```plain
doesn't matter
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-08-11:
@UlliBien I can confirm this, will open a PR with a fix.

UlliBien, 2023-08-11:
I again looked deeper to the code. In methods spiDetach... you should not replace default values -1. If there are default values the pin could not have been attached before (spiAttach.. replaces defaults). spiDetach... sets pin mode to INPUT. This may have unexpected effects. Just do nothing when pin has value -1.

P-R-O-C-H-Y, 2023-08-11:
In my opinion we can leave it as it is. It's been there for a long time without any complains.
In the new 3.0 version of Arduino-esp32 core this issue won't be existing as we implemented the peripheral manager and the pin can't be -1 anymore.
If you want to take a look, you can do it there: https://github.com/espressif/arduino-esp32/blob/esp-idf-v5.1-libs/cores/esp32/esp32-hal-spi.c

VojtechBartoska, 2023-08-14:
@P-R-O-C-H-Y Can we mark this as pending "to be covered" by Peripheral Manager?

P-R-O-C-H-Y, 2023-08-14:
@VojtechBartoska Main of the issue is being fixed by that linked PR. In 5.1-libs branch this is already fixed and even the second part (UlliBien's comment) is already fixed there. So probably we can mark it like that.

P-R-O-C-H-Y, 2023-08-14:
Closing this as fixed by #8512 and in 5.1 branch this issue no longer exists as commented before.

PR #8512 (2023-08-12): SPI - fix attach pin for ESP32-S3
## Description of Change
This PR fixes attaching/detaching some of the SPI pins for S3. Probably a leftover when new chip was added.
In 5.1-libs branch, this part of code no longer exists, so no issue there.

## Tests scenarios
checked agains variant if pins are correct.

## Related links
Closes #8508
closed 2023-08-14 by P-R-O-C-H-Y