espressif/arduino-esp32 #8510: ESP32-C3 Deep Sleep Wake-up

abhi-sachdeva, 2023-08-10:
### Board

ESP32-C3-DevKit-M1

### Device Description

[image]

The pull down resistor is 10kohms

### Hardware Configuration

GPIO3 is the interrupt pin being used as a wake-up source

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

115200

### Description

The ESP32-C3 will not wake up from deep sleep. I have reached out to other forums and have verified that the code works for them and is able to wake up the ESP32-C3. I do not know whether my problem lies in the breadboard or something else. I am using ESP32C3 Dev Module as the board and COM3 as the port on Arduino IDE. The serial output just states that "wakeup was not caused by deep sleep: 7" every time i push the button in attempt to wake the esp up. The boot number successfully increments after that. I will also provide the output (not serial monitor) below:

[image]

### Sketch

```cpp
#include "esp_sleep.h"
#include "driver/gpio.h"

#define INTERRUPT_PIN 3
RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  print_wakeup_reason();

  Serial.println("Starting");
  delay(2000);
  
  for(int i=3; i!=0; i--){
      Serial.println(i);
      delay(1000);
  }

  Serial.println("Going to sleep now");

  pinMode(INTERRUPT_PIN, INPUT);
  esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN, ESP_GPIO_WAKEUP_GPIO_HIGH);
  gpio_set_direction((gpio_num_t)INTERRUPT_PIN, GPIO_MODE_INPUT);  // <<<=== Add this line
  esp_deep_sleep_start();

  Serial.println("This will never be printed");
}

void loop() { }

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:
      Serial.println("Wakeup caused by external signal using RTC_IO");
      break;
    case ESP_SLEEP_WAKEUP_EXT1:
      Serial.println("Wakeup caused by external signal using RTC_CNTL");
      break;
    case ESP_SLEEP_WAKEUP_TIMER:
      Serial.println("Wakeup caused by timer");
      break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
      Serial.println("Wakeup caused by touchpad");
      break;
    case ESP_SLEEP_WAKEUP_ULP:
      Serial.println("Wakeup caused by ULP program");
      break;
    default:
      Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
      break;
  }
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Boot number: 1
Wakeup was not caused by deep sleep: 0
Starting
3
2
1
Going to sleep now
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Boot number: 2
Wakeup was not caused by deep sleep: 7
Starting
3
2
1
Going to sleep now
```

### Other Steps to Reproduce

Others have tried using my code and have said that it has worked for them, I do not know why it does not work for me! Also, taking away the resistor and wiring straight to ground brings up a message that says: "Not connected. Select a board and port to connect automatically."

Could it be an issue with a driver that I installed?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-10:
>wakeup was not caused by deep sleep: 7

7 is wakeup by GPIO.

Please update your sketch function:

``` cpp
void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:
      Serial.println("Wakeup caused by external signal using RTC_IO");
      break;
    case ESP_SLEEP_WAKEUP_EXT1:
      Serial.println("Wakeup caused by external signal using RTC_CNTL");
      break;
    case ESP_SLEEP_WAKEUP_TIMER:
      Serial.println("Wakeup caused by timer");
      break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
      Serial.println("Wakeup caused by touchpad");
      break;
    case ESP_SLEEP_WAKEUP_ULP:
      Serial.println("Wakeup caused by ULP program");
      break;
      // new wakeup reason for the RiscV ESP32Cx
      case ESP_SLEEP_WAKEUP_GPIO:
      Serial.println("Wakeup caused by a GPIO");
      break;
    default:
      Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
      break;
  }
}
```

abhi-sachdeva, 2023-08-10:
Great, that worked! If you have the time, could you please explain why that 1 line of code was needed/worked because I have not seen that function "ESP_SLEEP_WAKEUP_GPIO" before. How come the EXT0 case did not wake up the esp?

SuGlider, 2023-08-10:
@abhi-sachdeva - EXT0 and EXT1 are for ESP32, ESP32S2 and ESP32S3, by other hand GPIO is for the C3, that has no EXT0, EXT1. 
The C3 can set GPIO 0 to GPIO 5 for wakeup from deep sleep. One or more GPIOs can be set at the same time.

abhi-sachdeva, 2023-08-10:
That makes sense, thanks for the explanation! :)

closed 2023-08-10 by abhi-sachdeva