espressif/arduino-esp32 #8516: Template for switch case scenario.

miguelfernando55, 2023-08-12:
### Related area

a simple function that allows to run a cycle of various types of variables

### Hardware specification

esp32

### Is your feature request related to a problem?

Redundancy code.

### Describe the solution you'd like

I have strings, uint32_t, uint8_t, etc variables to save on preferences. Right now I have a lot of redundancy code. I would like you to make a function where I only need to put the name of the namespace, the type of the variable and the value.
Cheers

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

HareshPrajapati, 2023-08-28:
Certainly, you can create a function to streamline the process of saving different types of variables to the Preferences library on the ESP32. Here's an example of how you could achieve this:

```
// Function to save variables to Preferences
template <typename T>
void saveToPreferences(const char* namespaceName, const char* key, T value) {
    Preferences prefs;
    prefs.begin(namespaceName, false); // "false" indicates read-only mode
    prefs.put(key, value);
    prefs.end();
}
```

VojtechBartoska, 2023-08-28:
@me-no-dev can you please help with triage?

closed 2023-08-28 by VojtechBartoska