espressif/arduino-esp32 #8519: DRAM segment data does not fit - Xtensa linker error in core versions 2.0.6 through current, 2.1.11

jponko, 2023-08-13:
### Board

NodeMCU-32S or ESP32-DevKitC but MCU hardware  is not needed - linker problem

### Device Description

No hardware is needed this is a linker problem. You may choose either a NodeMCU-32S or ESP32-DevKitC as the ESP32 MCU.

### Hardware Configuration

No hardware is needed this is a linker problem. You may choose either a NodeMCU-32S or ESP32-DevKitC as the MCU.

### Version

v2.0.11

### IDE Name

Arduino IDE 2.1.1 Linux

### Operating System

NAME="Linux Mint" VERSION="21.2 (Victoria)" ID=linuxmint ID_LIKE="ubuntu debian" PRETTY_NAME="Linux Mint 21.2" VERSION_ID="21.2" HOME_URL="https://www.linuxmint.com/" SUPPORT_URL="https://forums.linuxmint.com/" BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/" PRIVACY_POLICY_URL="https://www.linuxmint.com/" VERSION_CODENAME=victoria UBUNTU_CODENAME=jammy

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Linker problem has existed since core version 2.0.6. Program compliles and works on version 2.0.5 but Xtensa Linker fails in versions 2.0.6 through current 2.0.11 with errors "DRAM segment data does not fit." as shown below: 
---------------------------------------------------
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino/sketches/F539CEA913CF86ABC96F797B7891026C/VFO_CTRL_V5.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 3536 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

### Sketch

```cpp
Source code can be retrieved from:
https://www.hackster.io/mircemk/universal-vfo-10-khz-160-mhz-with-retro-analog-scale-c7c5b6
```

### Debug Message

```plain
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino/sketches/F539CEA913CF86ABC96F797B7891026C/VFO_CTRL_V5.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/jponko/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 3600 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-08-22:
It is not a linker issue. Code size has increased (in later cores) and goes overboard.
Two possibilitys use older core or decrease your code size.

jponko, 2023-08-22:
Ok, thanks for the clarification. Hopefully that the increase in core size
is for a good reason.

closed 2025-01-15 by Parsaabasi