espressif/arduino-esp32 #8526: No serial input data in task / works in loop()

caipifrosch, 2023-08-14:
### Board

custom ESP32

### Device Description

simple hardware with ESP32-WROOM module, serial port connected to USB adapter

### Hardware Configuration

just connecting UART0

### Version

v2.0.11

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Serial input does work in loop(), but not when called from separate task. 

Serial.available() always returns 0 

### Sketch

```cpp
TaskHandle_t testTaskHandle = NULL;

#define TEST_TASK_STACK_SIZE (8*1024)

void Read_RS232(void)
{

	Serial.println("Serial task");	// I can see this!!! task is executed...

	if(Serial.available())
	{
		String command = Serial.readStringUntil('\n');
		if(command == "ON")
		{
			digitalWrite(RED_LED, LOW);
			Serial.println("LED is turned ON");
		}
		else
		if(command == "OFF")
		{
			digitalWrite(RED_LED, HIGH);
			Serial.println("LED is turned OFF"); 
		}
	}

}

void radioTask( void * pvParameters ) {

	for (;;) {

		yield();

		Read_RS232();  // does NOT work here!!!

		vTaskDelay(500);
	
	}

} // while(1)
	
void setup() {

	Serial.begin(19200);
	delay(10);

    pinMode(RED_LED, OUTPUT);
    digitalWrite(RED_LED, HIGH);
	
	xTaskCreatePinnedToCore(testTask, "testTask", TEST_TASK_STACK_SIZE, NULL, 3, &testTaskHandle,  1);
	
 }
 
void loop() {

	Read_RS232();	// works here!!!
	
	delay(100);

}
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

use any ESP32 board, compile, download and watch in terminal program

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-14:
@caipifrosch - I can't replicate the issue. 

You have 2 tasks: `setup()/loop() task` and `radioTask()`.
 Both call `Read_RS232()`
`loop()` has priority 1 (lower) - runs 10 times per second
`radioTask()` has priority 3 (higher) - runs 2 times per second

There is a way higher probability that `loop()->Read_RS232()` runs most of the time and consumes the UART0 (`Serial`) data.
Right after reading, the `Serial.available()` will return `Zero` until a new line is sent to the `Serial` interface.

I have tested this modificated sketch and it read from both, the Task and from the loop().  
This is the sketch and its output:

``` cpp
#define RED_LED 2
#define TEST_TASK_STACK_SIZE (8*1024)

void Read_RS232(bool requestFromTask) {
  if (Serial.available()) {
    if (requestFromTask) Serial.println("Serial read by the task");
    else Serial.println("Serial read by Loop()");
    String command = Serial.readStringUntil('\n');
    if (command == "ON") {
      digitalWrite(RED_LED, LOW);
      Serial.println("LED is turned ON");
    } else if (command == "OFF") {
      digitalWrite(RED_LED, HIGH);
      Serial.println("LED is turned OFF");
    }
  }
}

void radioTask( void * pvParameters ) {
  for (;;) {
    Read_RS232(true);
    vTaskDelay(500);
  }
}

void setup() {
  Serial.begin(115200);
  delay(10);

  pinMode(RED_LED, OUTPUT);
  digitalWrite(RED_LED, HIGH);

  xTaskCreatePinnedToCore(radioTask, "radioTask", TEST_TASK_STACK_SIZE,
                          NULL, 3, NULL,  1);
}

void loop() {
  Read_RS232(false);
  delay(100);
}
```

#### Ouput:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[    15][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Serial read by Loop()
LED is turned ON
Serial read by the task
LED is turned OFF
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned OFF
Serial read by Loop()
LED is turned OFF
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned ON
Serial read by Loop()
LED is turned ON
Serial read by the task
Serial read by the task
Serial read by Loop()
Serial read by Loop()
Serial read by Loop()
Serial read by Loop()
Serial read by Loop()
Serial read by the task
Serial read by the task
Serial read by Loop()
Serial read by Loop()
Serial read by the task
Serial read by the task
Serial read by Loop()
Serial read by the task
Serial read by Loop()
Serial read by Loop()
Serial read by the task
Serial read by Loop()
Serial read by Loop()
```

caipifrosch, 2023-08-15:
@SuGlider Thanks a lot for this detailed analysis and your suggestions!

Maybe my source code example on initial report was missleading, as both places for function call were uncommented.

Function call is intended to run from task only. Just for tests I used function call in either loop() OR radioTask. Other was commented.

Function call enabled in loop() will get me rx bytes as expected.

BUT: There are no rx bytes when calling function from task. Task is executed though: I can see the Serial.println("Serial task");

VojtechBartoska, 2023-08-18:
can we close this @caipifrosch?

caipifrosch, 2023-08-19:
Thanks a lot for your help. Must be something dumb at my end...

closed 2023-08-19 by caipifrosch