espressif/arduino-esp32 #8527: Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled. (W/ ESP32-C3)

abhi-sachdeva, 2023-08-14:
### Board

ESP32-C3-DevKit-M1

### Device Description

The dev kit is on a breadboard with 2 push buttons connected to pins 2 & 3. The breadboard also has a green, red, and blue led connected to pins 5, 6, & 7 respectively. The button attached to pin 3 is used to control the red and green led and wake up the esp32-c3 from deep sleep. The other button attached to pin 2 is used to initiate ble server and advertise itself. While this process is being undergone, a blue led illuminates. The blue led turns off after a successful connection has established (which also puts the esp32-c3 into deep sleep).

### Hardware Configuration

Resistors in series with the leds (220ohm) and a pull down resistor for both buttons (10kohm)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Getting an error as follows:
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

The reason this occured is because I tried to implement a function: void pairing()
This function simply uses a button to: if HIGH --> Blue LED turns on and starts BLE code 
if LOW --> Blue LED stays off and no BLE code is ran
The function utilizes a variable called buttonState2 and sets it = digitalRead(buttonPin2);

The expected behavior is that pressing the button would turn on the LED and start scanning and connection via BLE. Once connected, I implemented a callback such that the LED turns off once a successful connection has been made and the esp32-c3 goes to deep sleep. The callback has been created, not implemented yet as I am trying to fix the issue with the button first.

### Sketch

```cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <Wire.h>

const int buttonPin = 3;  // # for button pin for States
const int buttonPin2 = 2; // # for button pin for BLE
const int ledPin =  5;   // # for the Green LED pin
const int ledPin2 = 6;   // # for the Red LED pin
const int ledPin3 = 7;    // # for the Blue LED pin

// variables will change:
int buttonState = 0;     
int lastButtonState = 0;
int buttonState2 = 0;
int lastButtonState2 = 0;  
int ledState = 0;
RTC_DATA_ATTR int counter = 0; 
[... 174 lines cut ...]
3fca0bf0: 0x3fca0c40 0x3fca0d30 0x3fca0d24 0x00000000 0x3fca0c04 0xffffffff 0x3fca0c04 0x3fca0c04
3fca0c10: 0x00000000 0x3fca0c18 0xffffffff 0x3fca0c18 0x3fca0c18 0x00000000 0x00000014 0x0000000c

ELF file SHA256: d9c97c912b1f8644

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038200a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

### Other Steps to Reproduce

I tried using/initializing a different variable in the void pairing() function and that did not fix the issue as well. My understanding of this is that the variable is in an invalid memory location/cant be read.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-14:
@abhi-sachdeva - Load Access Fault usually happens when trying to read content of a NULL pointer.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/fatal-errors.html#instruction-access-fault-load-access-fault-store-access-fault

SuGlider, 2023-08-14:
It seems that BLE is not initiated and `loop()` is running in parallel.
I suggest to only use `pCharacteristic->` if it is not NULL, like in this:

``` cpp
void loop(){
  
  // read the state of the BLE pushbutton
  buttonState2 = digitalRead(buttonPin2);
  if (lastButtonState2 == HIGH && buttonState2 == LOW) {
    Serial.println("The button is pressed");
    digitalWrite(ledPin3, HIGH);
    if (pCharacteristic != NULL) {
      pCharacteristic->setValue("device_state_1_ready"); //add code for device 1 state ready once buttonPin1 is pushed/released
      pCharacteristic->notify();
    }
  } else if (lastButtonState2 == LOW && buttonState2 == HIGH) {
    Serial.println("The button is released");
    digitalWrite(ledPin3, LOW);
  }
  
  lastButtonState2 = buttonState2;
 // ... rest of the loop() code ...
```

abhi-sachdeva, 2023-08-15:
> It seems that BLE is not initiated and `loop()` is running in parallel. I suggest to only use `pCharacteristic->` if it is not NULL, like in this:
> 
> ```c++
> // ======>>>> Always initialize pointer with NULL
> BLEServer *pServer = NULL;
> BLEService *pService = NULL;
> BLECharacteristic *pCharacteristic = NULL;
> 
> void loop(){
>   
>   // read the state of the BLE pushbutton
>   buttonState2 = digitalRead(buttonPin2);
>   if (lastButtonState2 == HIGH && buttonState2 == LOW) {
>     Serial.println("The button is pressed");
>     digitalWrite(ledPin3, HIGH);
> 
>     // =======>>>> Make sure the pointer is not NULL before using it
>     if (pCharacteristic != NULL) {
>       pCharacteristic->setValue("device_state_1_ready"); //add code for device 1 state ready once buttonPin1 is pushed/released
>       pCharacteristic->notify();
>     }
>     // =======>>>> Make sure the pointer is not NULL before using it
> 
> 
>   } else if (lastButtonState2 == LOW && buttonState2 == HIGH) {
>     Serial.println("The button is released");
>     digitalWrite(ledPin3, LOW);
>   }
>   
>   lastButtonState2 = buttonState2;
>  // ... rest of the loop() code ...
> ```

So would you initialize the pointers (pCharacteristic, pServer, pService) as something other than null?

closed 2023-08-15 by abhi-sachdeva