espressif/arduino-esp32 #8529: MQTT Library

savagerex, 2023-08-15:
### Board

ESP32-WROVER-E

### Device Description

Sensor - iis3dwb
external flash

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

I use Mqtt library.( https://github.com/256dpi/arduino-mqtt ).

but this libray have different result between ESP32 WROVER and ESP32 S3.

if i declare in  ESP32 WROVER
as below.

WiFiClient net;
MQTTClient client(891100);

it use "PSRAM" about 891100.

ESP32 WROVER always reset.

if i reduce

WiFiClient net;
MQTTClient client(1000);

then ESP32 WROVER is work.
*********************************************************

if i declare in  ESP32 S3

WiFiClient net;
MQTTClient client(1782000);

it use "PSRAM" about 1782000.

it is work.

### Sketch

```cpp
#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21     ASUS-38
const char* pass = "12345678";   //wireless    12345678   1234567890   12345678

const char* mqtt_server = "192.168.50.129";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";

//uint8_t DataBufRead[18500];
uint8_t* DataBufRead;

WiFiClient net;
MQTTClient client(891100);  //1782000   ,  891100

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");

  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  //DataBufRead = (uint8_t*)ps_calloc(891100, sizeof(uint8_t));  //600
  
  WiFi.begin(ssid, pass);

  //DataBufRead = (uint8_t*)ps_calloc(600, sizeof(uint8_t));  //1700500

  // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported
  // by Arduino. You need to set the IP address directly.
  client.begin("192.168.50.129", net);
  //client.onMessage(messageReceived);

  connect();

  client.loop();
  delay(10);  // <- fixes some issues with WiFi stability

  if (!client.connected()) {
    connect();
  }
  

  client.loop();

  if (!client.connected()) {
    connect();
  }

  
  while(1);

  for (uint32_t i = 0; i<890999; i++)       // 1699999
  //for (uint32_t i = 0; i<161999; i++)
  //for (uint32_t i = 0; i<19999; i++)
  {
    DataBufRead[i] = 'a';
  }

  Serial.println(DataBufRead[0]);

   char *DataBuf = (char*)DataBufRead;
   bool PubData = client.publish(topic, DataBuf, 891000, false, 0);  //18000 //6000
   Serial.println(PubData);

   
   Serial.println("Mqtt Finish");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
hecking wifi....
connecting...Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014a25b  PS      : 0x00060130  A0      : 0x800d4c74  A1      : 0x3ffb2110  
A2      : 0xffffffff  A3      : 0x000d98dc  A4      : 0x00000010  A5      : 0x00000004  
A6      : 0x00000000  A7      : 0x00000fff  A8      : 0x3ffb2150  A9      : 0x00000000  
A10     : 0x000006ec  A11     : 0x00000000  A12     : 0x000006ec  A13     : 0x0000b2c0  
A14     : 0x00000008  A15     : 0x3ffc4550  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008a58c  LEND    : 0x4008a5a8  LCOUNT  : 0x00000000  

Backtrace: 0x4014a258:0x3ffb2110 0x400d4c71:0x3ffb2130 0x400d4a0a:0x3ffb2180 0x400d46ed:0x3ffb21d0 0x400d27b9:0x3ffb2240 0x400d282c:0x3ffb2260 0x400d5d9a:0x3ffb2290

ELF file SHA256: 7a8bd1733f79ec0a

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-15:
This forum is for issues with the code included here.  Issues with 3rd party libraries should be addressed on their site.  General questions should be asked at https://esp32.com

savagerex, 2023-08-16:
before we discuss this issue. (https://github.com/espressif/arduino-esp32/issues/8423#issuecomment-1642231644)

you mention that ESP32 WROVER only have 4MB PSRAM.

but i only declare 891000.

why memory not enough????

lbernstone, 2023-08-16:
I also mentioned that there is no reason to have the buffer size larger than your maximum mqtt message size.

SuGlider, 2023-08-17:
@savagerex - I have analyzed the issue with the MQTT Library from  https://github.com/256dpi/arduino-mqtt

I concluded that this issue is due to the fact that the MQTT Library uses `malloc()` in its class constructor `MQTTClient::MQTTClient(int bufSize)` and it happens when the PSRAM has not being already initialized  (before Arduino `app_main()` has even being executed). 

Therefore the `malloc()` always fails if the size of the buffer doesn't fit in the ESP32 RAM (not PSRAM).
This is an issue of the Library and it must be fixed within the Library code.

It is not related to ESP32 Arduino code.

Check the related code:

#### Constructor
https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.h#L99-L100
```cpp
  explicit MQTTClient(int bufSize = 128) : MQTTClient(bufSize, bufSize) {}
  MQTTClient(int readSize, int writeBufSize);
```

https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.cpp#L161-L167
``` cpp
MQTTClient::MQTTClient(int readBufSize, int writeBufSize) {
  // allocate buffers
  this->readBufSize = (size_t)readBufSize;
  this->writeBufSize = (size_t)writeBufSize;
  this->readBuf = (uint8_t *)malloc((size_t)readBufSize + 1);
  this->writeBuf = (uint8_t *)malloc((size_t)writeBufSize);
}
```

SuGlider, 2023-08-17:
My suggestion would be for you to fix it manually and create your own MQTT Library version based on the original code.

savagerex, 2023-08-17:
@SuGlider 

but this lib is work in esp32 s3.

if i declare 

WiFiClient net;
MQTTClient client(1782000);

SuGlider, 2023-08-17:
> @SuGlider
> 
> but this lib is work in esp32 s3.
> 
> if i declare
> 
> WiFiClient net; MQTTClient client(1782000);

Please demonstrate that it works with the ESP32-S3. 
Do you have any evidence of it?

I have one evidence that it also doesn't work with the ESP32-S3:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   151][I][MQTTClient.cpp:164] MQTTClient(): MQTT buffer size is: [891100]
[   151][E][MQTTClient.cpp:167] MQTTClient(): ====> MQTT READ BUF is NULL !!!
[   153][E][MQTTClient.cpp:170] MQTTClient(): ====> MQTT WRITE BUF is NULL !!!
[   397][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   439][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   473][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
checking wifi....[   567][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   588][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   589][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.104, MASK: 255.255.255.0, GW: 192.168.1.1

connecting...Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42074e5f  PS      : 0x00060d30  A0      : 0x8200690c  A1      : 0x3fcebe60  
A2      : 0xffffffff  A3      : 0x000d98dc  A4      : 0x00000010  A5      : 0x00000004  
A6      : 0x00000001  A7      : 0x00000fff  A8      : 0x3fcebea0  A9      : 0x00000000  
A10     : 0x00000691  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000008  A15     : 0x3fc9ae44  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x42074e5c:0x3fcebe60 0x42006909:0x3fcebe80 0x420066f4:0x3fcebed0 0x42006411:0x3fcebf20 0x42002c4d:0x3fcebf90 0x42002cc1:0x3fcebfb0 0x42007d32:0x3fcebfe0
```

savagerex, 2023-08-21:
@SuGlider 

As below is ESP32 S3 .

[image]

HamzaHajeir, 2023-08-21:
If it provides any help, you might checkout [H4AsyncMQTT](https://github.com/hamzahajeir/h4asyncmqtt).

SuGlider, 2023-08-23:
@savagerex - I've confirmed your point with the S3.
I used a OPI PSRAM ESP32-S3 and it has worked as you described.
But when using a QSPI PSRAM ESP32-S3 it crashes, exactly like the ESP32-WROVER.

It seems that there is a difference in the bootloader when PSRAM is Octal SPI.
It may have been configured to initialize the PSRAM in boot time.

Other than that, the explanation from https://github.com/espressif/arduino-esp32/issues/8529#issuecomment-1681423588 continues valid.

VojtechBartoska, 2024-09-12:
Hello, come up to this during issue clean up. Can you please @savagerex retest this against 3.1.0-RC1? If the issue is still valid? Thanks a lot!

closed 2025-01-10 by SuGlider