espressif/arduino-esp32 #8539: ESP32-S3 ESP32-S3 doesn't work and is not suitable for a decent production level projects

virtyaluk, 2023-08-17:
### Board

ESP32-S3-DevKitC-1 N32R8V

### Device Description

DevKitC from [here](https://www.aliexpress.com/item/1005003979778978.html?spm=a2g0o.order_list.order_list_main.71.15ca1802MmySrI)

### Hardware Configuration

USB connected to my laptop's USB through USB-to-USB-C adapter

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

macOS 13.5

### Flash frequency

QIO 80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

It is simple - the board doesn't work. It doesn't matter how hard I try it doesn't want to run my sketch. I've tried different Arduino IDE configs based on what I heard from other people on the Internet experiencing the same problems with this board, but nothing helps. The uploading process works fine, but the actual code won't execute. I have a dozen other boards like TTGO, Lolin D32, and regular ESP32 which work just fine. I simply that this board doesn't work and couldn't be used for a production-grade project.

### Sketch

```cpp
#include <Arduino.h>
#include <Wire.h>

void setup() {
  Wire.begin();
  Serial.begin(9600);

  while (!Serial)
  {
  }

  delay(500);
}

void i2c_scanner() {
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}

void loop() {
  i2c_scanner();
}
```

### Debug Message

```plain
02:11:34.174 -> assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)
02:11:34.174 -> 
02:11:34.174 -> 
02:11:34.174 -> Backtrace: 0x40377406:0x3fceb180 0x4037a70d:0x3fceb1a0 0x403801cd:0x3fceb1c0 0x42008ad6:0x3fceb2f0 0x40376d77:0x3fceb320 0x403cd823:0x3fceb350 0x403cdae2:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
02:11:34.174 ->
```

### Other Steps to Reproduce

My Arduino IDE setup:
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-18:
@virtyaluk - I think that the `ESP32-S3-DevKitC-1 N32R8V` is Octal Flash and Octal PSRAM.
Your configuration says `QIO 80MHz`, please change it to `OPI 80MHz` and in case you need to enable PSRAM, please select `OPI PSRAM`.

Please let me know if this new setting makes you board work.

SuGlider, 2023-08-18:
@virtyaluk - 

Confirmed. Looking into the table of parts, it says OT (Octal Flash and PSRAM). Please change it in your Arduino IDE configuration.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#ordering-information

<html>
<body>

ESP32-S3-DevKitC-1-N32R8V | ESP32-S3-WROOM-2-N32R8V | 32 MB OT | 8 MB OT | 1.8 V
-- | -- | -- | -- | --

</body>
</html>

SuGlider, 2023-08-28:
@virtyaluk - Please let me know if this issue has been solver and could be closed.

twinsvision, 2023-09-20:
> @virtyaluk - I think that the `ESP32-S3-DevKitC-1 N32R8V` is Octal Flash and Octal PSRAM. Your configuration says `QIO 80MHz`, please change it to `OPI 80MHz` and, in case you need to enable PSRAM, please select `OPI PSRAM`.
> 
> Please let me know if this new setting makes you board work.

GREAT!!! Works like a charm now! Been looking for this solution for days, thanks alot @SuGlider !

lbernstone, 2023-09-20:
> @virtyaluk - Please let me know if this issue has been solved and could be closed.

Do you now think the ESP32-S3 does work and is suitable for a decent production level project? :)

everslick, 2023-09-24:
@virtyaluk Please close the issue.

closed 2023-10-02 by VojtechBartoska