espressif/arduino-esp32 #8540: ESP32-S3  iOS Bluetooth transfer speed is only 5kb/s

xianengqi, 2023-08-18:
### Board

ESP32-S3

### Device Description

I am an iOS development engineer, the development of only one function, test bluetooth transfer speed, now encountered the problem is that a 300kb file, the transfer speed is only 7kb/s, troubled for several days, asked the GPT4 also did not optimize the bluetooth file transfer speed, so seek technical support. I will provide the complete SwiftUI code. 
```SwiftUI
import CoreBluetooth
import SwiftUI

struct ContentView: View {
  @State private var fileURL: URL? = nil
  @State private var status: String = "Not Connected"
  @StateObject private var bluetoothManager = BluetoothManager()

  var transferRateText: String {
    let transferRateKB = bluetoothManager.transferRate / 1024
    if transferRateKB < 1024 {
      return String(format: "%.2f KB/s", transferRateKB)
    } else {
      let transferRateMB = transferRateKB / 1024
      return String(format: "%.2f MB/s", transferRateMB)
    }
  }

  var body: some View {
    VStack {
      Button("Select File") {
        // TODO: Select file
        selectFile()
      }
      .padding()

      Text("Status: \(bluetoothManager.status)")
        .padding()
      Text("Transfer Rate: \(transferRateText)")
        .padding()

      Text("Total Time: \(bluetoothManager.totalTime, specifier: "%.2f") seconds")
        .padding()
      List(bluetoothManager.discoveredDevices, id: \.identifier) { device in
        HStack {
          Text(device.name ?? "Unknown")
          Spacer()
          Button("Connect") {
            bluetoothManager.connectToDevice(device)
          }
        }
      }
    }
  }

  private func selectFile() {
    bluetoothManager.presentFilePicker()
  }
}

#Preview {
  ContentView()
}

class BluetoothManager: NSObject, ObservableObject, CBCentralManagerDelegate, CBPeripheralDelegate {
  private var centralManager: CBCentralManager!
  private var peripheral: CBPeripheral?
  public var fileTransferCharacteristic: CBCharacteristic?
  @Published var status: String = "Not Connected"
  @Published var discoveredDevices: [CBPeripheral] = []
  private var startTime: Date?
  private var endTime: Date?
[... 388 lines cut ...]
      fileData = try Data(contentsOf: tempURL)

      // Send the file data
      sendFileData(fileData, to: fileTransferCharacteristic)
    } catch {
      print("Error reading file data: \(error)")
      status = "Error: \(error.localizedDescription)"
    }
  }
}
```

### Debug Message

```plain
*
```

### Other Steps to Reproduce

*

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

xianengqi, 2023-08-18:
[image]

xianengqi, 2023-08-18:
https://github.com/espressif/arduino-esp32/assets/30774189/a8490384-aa06-48b6-a3ba-11948a0f382d

mrengineer7777, 2023-08-18:
How is the ESP32 setup?  You aren't giving us anything from this repository to troubleshoot.

xianengqi, 2023-08-19:
Hello Teacher, I am new to IOT development, first time IOT developer, I am an iOS developer, I am really not sure about the ESP32 settings, but I suspect that it is due to the problem of my code, because I am writing Bluetooth transfer files for the first time, so the problem I want to solve is `How to speed up the bluetooth transfer of files to the ESP32-S3`.

xianengqi, 2023-08-21:
> How is the ESP32 setup? You aren't giving us anything from this repository to troubleshoot.

It's burned into the official sample project.``` https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server ```

mickeyl, 2023-09-23:
GATT has quite some overhead. I recommend using an L2CAP channel instead.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

mickeyl, 2025-01-15:
For anyone who comes over this ticket, I have done some extensive research on the topic which you can follow over @ https://github.com/espressif/esp-idf/issues/13637