espressif/arduino-esp32 #8542: CSS loading

bovineck, 2023-08-19:
### Board

ESP32 Dev Board

### Device Description

plain module on breadboard

### Hardware Configuration

GPIO 21 connected to an LED

### Version

v2.0.11

### IDE Name

Arduino IDE 1.8.15

### Operating System

Linux 5.4.0-156-generic #173-Ubuntu

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

CSS in SPIFFS not being loaded/recognised - used a manual workaround for the WebServer library which does work.

### Sketch

```cpp
Manual workaround:

void handleCSS() {
  server.sendHeader("Content-Type", "text/css");
  File cssFile = SPIFFS.open("/style.css", "r");
  if (cssFile) {
    server.streamFile(cssFile, "text/css");
    cssFile.close();
  }
}

...and in setup...

  server.on("/style.css", HTTP_GET, handleCSS);

Which does work.
```

### Debug Message

```plain
No debug message - just CSS styles not applied to the html file in SPIFFS
```

### Other Steps to Reproduce

Tried the following as suggested by ChatGPT:

If the style.css file in your SPIFFS is not affecting the style of the upload.html file, there could be a few reasons for this. Let's go through some troubleshooting steps to identify and resolve the issue:

Check File Paths:

Make sure both the upload.html and style.css files are placed in the root of the data folder within your Arduino sketch directory.
Confirm that the file names are spelled correctly and that they have the correct file extensions (html and css).
Verify SPIFFS Initialization:

Ensure that you're initializing the SPIFFS properly in the setup() function of your .ino sketch using SPIFFS.begin().

Check HTML Link to CSS:

In your upload.html file, ensure that the link to the style.css file is correct. It should be relative to the root of the SPIFFS and not include any leading slashes.

Check Browser Cache:

Browsers tend to cache CSS and other files. Clear your browser's cache or open the page in a private browsing/incognito window to ensure you're seeing the updated styles.
Inspect Network Requests:

Use your browser's developer tools to inspect the network requests. Make sure that the style.css file is being loaded correctly by the browser.

Debugging CSS:

Simplify your style.css file and test if any basic styles are being applied. For example, try changing the background color of the body element to see if it takes effect.

Content-Type Header:

Ensure that the server is sending the correct Content-Type header for the CSS file. It should be text/css.

Server Configuration:

Double-check the server configuration in your .ino sketch to ensure that the server is correctly serving files from the SPIFFS.

Serial Debugging:

Insert Serial.println() statements in your .ino sketch to confirm that the server is properly serving the style.css file. This will help you debug any issues related to the server setup.

Try Another Browser:

If you're still encountering issues, try accessing the page using a different browser to rule out any browser-specific problems.
PlatformIO vs Arduino IDE:

If you're using PlatformIO, ensure that the file paths and configurations match your setup. Sometimes, small differences can cause unexpected behavior.
By systematically going through these steps, you should be able to identify the root cause of the issue and resolve it. If you're still facing problems after trying these steps, feel free to provide more details, and I'll be happy to assist further.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-19:
Giving us the code that does work is not useful.  Post the code that fails.
Turn on verbose debugging, and try to retrieve only the css file.  Use the developer tools in the browser to watch what is going on (particularly the network stats).

bovineck, 2023-08-20:
#include <WiFi.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <ArduinoOTA.h>
#include "FS.h"
#include <SPIFFS.h>
#include "mymain.h"

// some variables

WebServer server(80);

void handleRoot() {
  File uploadPage = SPIFFS.open("/upload.html", "r");
  server.streamFile(uploadPage, "text/html");
}

// some other code including setup and loop

e.g.  server.on("/", HTTP_GET, handleRoot);

---------------------------------------

html code

<!DOCTYPE html>
<html>
<head>
    <title>ESP32 Firmware Update</title>
    <link rel="stylesheet" type="text/css" href="style.css">
</head>
<body>

// some html code 

</body>
</html>

---------------------------------------

css code

body {
    font-family: Arial, sans-serif;
    background-color: powderblue;
    margin: 0;
    padding: 0;
}

h1 {
    color: #333;
    text-align: center
}

form {
    margin: 20px;
    padding: 20px;
    border: 1px solid #ddd;
    background-color: powderblue;
    border-radius: 5px;
    text-align: center
}

input[type="file"], input[type="password"], input[type="submit"] {
    margin: 5px;
    padding: 8px;
    border: 1px solid #ccc;
    border-radius: 3px;
    text-align: center
}

The html file loads fine, all the code works, life is good. The CSS file is not loaded and has no effect on the style or format of the html file. UNLESS...I use the manual loading code below...

Manual workaround:

void handleCSS() {
  server.sendHeader("Content-Type", "text/css");
  File cssFile = SPIFFS.open("/style.css", "r");
  if (cssFile) {
    server.streamFile(cssFile, "text/css");
    cssFile.close();
  }
}

...and in setup...

  server.on("/style.css", HTTP_GET, handleCSS);

Which does work.

As for: "Turn on verbose debugging, and try to retrieve only the css file. Use the developer tools in the browser to watch what is going on (particularly the network stats)."

I don't know what that means or how to implement it...sorry. My code works so the incentive to dive deeper into your code is pretty low.

Jason2866, 2023-08-20:
Why still using the old deprecated SPIFFS? LittleFS is faster and maintained.

bovineck, 2023-08-20:
Yes I was wondering about that - is it likely that SPIFFS is the problem? I can swap that out for LittleFS - which I use for the ESP8266

bovineck, 2023-08-20:
OK I swapped over to LittleFS - still no joy with the CSS upload/recognition.

AAJAY5, 2023-08-25:
I think you missed concept.

After '/' request from client ESP sent root page. Now client load **index.html** and found that need to fetch **/style.css**. So, it will send GET request for the same. But you have not register any server.on() with "style.css" So browser can't load. 

You can use inspect>>Network to see which request  are done with status.
Additionally you can add NOT_FOUND(404) handler in code so you get to know which request need to add.

tyVandeZande, 2025-12-17:
This would be helpful to have re-opened! CSS as a seperate file would greatly increase the usability of an ESP32-hosted website!