espressif/arduino-esp32 #8546: Flashing any program in ESP32 S3 result in constant crash loop.

jcracine-cimeq, 2023-08-21:
### Board

ESP32 S3 WROOM 2

### Device Description

The main board is a DevKitC.
There is three components connected:
- [A SD card reader](https://www.sparkfun.com/products/544)
- a mems microphone breakout board
- a w5500 ethernet module

### Hardware Configuration

- GPIO 10, 11, 12 and 13 are connected to the w5500 module
- GPIO 1 and 2 are connected to the MEMS Microphone
- GPIO 37, 38, 39 and 48 are connected to the SD card reader

### Version

v2.0.11

### IDE Name

VS Code with Platform IO

### Operating System

Ubuntu 22.04 LTS

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

No matter what program I try to build and upload in my ESP32 S3, it constantly crashes when booting up. I dont know why but it reference files in a weird path when I look at the core dump. I tried resinstalling PlatformIO and its packages completely but it still didn't work.

### Sketch

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  
}
```

### Debug Message

```plain
Backtrace: 0x40377152:0x3fceb180 0x4037a779:0x3fceb1a0 0x40380081:0x3fceb1c0 0x42006a9c:0x3fceb2f0 0x40376ac3:0x3fceb320 0x403cd823:0x3fceb350 0x403cdae2:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

  #0  0x40377152:0x3fceb180 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037a779:0x3fceb1a0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40380081:0x3fceb1c0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x42006a9c:0x3fceb2f0 in do_core_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:320
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:441
  #4  0x40376ac3:0x3fceb320 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:630
  #5  0x403cd823:0x3fceb350 in ?? ??:0
  #6  0x403cdae2:0x3fceb380 in ?? ??:0
  #7  0x403c9929:0x3fceb4b0 in ?? ??:0
  #8  0x40045c01:0x3fceb570 in ?? ??:0
```

### Other Steps to Reproduce

My platformio.ini file looks like this:
```
[env:esp32s3]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.11
    espressif/toolchain-xtensa-esp32s3 @ 12.2.0+20230208
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-08-21:
You can not use a SPI based with ESP32 Arduino. Not supported. You will get what you encounter -> Crash

lbernstone, 2023-08-22:
Does it crash if you use the default platformio.ini (platform = espressif32, no packages)?

jcracine-cimeq, 2023-08-22:
@Jason2866 
> You can not use a SPI based Ethernet PHY with ESP32 Arduino. Not supported. You will get what you encounter -> Crash

At this point in time, even with just my devkit not connected to anything other than my usb cable it produce the behavior.

@lbernstone 
> Does it crash if you use the default platformio.ini (platform = espressif32, no packages)?

Yes, I changed the lines to have this in my file:
```
[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

Jason2866, 2023-08-22:
@jcracine-cimeq Probably the flash and RAM of the board does not fit to yours.
There is NOT just one `esp32-s3-devkitc-1` board. There are different ones.
Different flash size and type and PSRAM. If the settings are not exactly the ones your board has it will bootloop.

jcracine-cimeq, 2023-08-22:
> @jcracine-cimeq Probably the flash and RAM of the board does not fit to yours. There is NOT just one `esp32-s3-devkitc-1` board. There are different ones. Different flash size and type and PSRAM. If the settings are not exactly the ones your board has it will bootloop.

ok. I searched for the specific devkit I bought and it is the ESP32-S3-DevKitC-1-N32R8V. I looked online a little and i found someone that made a template for the N16R8V:
```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "psram_type": "opi",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "required_upload_port": true,
    "speed": 921600
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}
```

But it still doesn't work. What bugs me a lot is that the crash dump references the ESP32 S2 in its path while I'm compiling for a S3.

```
 #0  0x40377152:0x3fceb180 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037a779:0x3fceb1a0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40380081:0x3fceb1c0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4200699c:0x3fceb2f0 in do_core_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:320
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:441
  #4  0x40376ac3:0x3fceb320 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:630
```

Jason2866, 2023-08-26:
The boards.json you "found" is not the correct one for your board. Your board has OPI Flash and PSRAM regarding the datasheet. Your boards.json is made for QIO/OPI. This results in non working ;-)

Jason2866, 2023-08-26:
This one should work
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "opi_opi"
    },
    "boot": "opi",
    "core": "esp32",
    "extra_flags": [
      "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1-N16R8V (32 MB OPI, 8MB OPI PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.espressif.com/blabla",
  "vendor": "Espressif"
}
```

Jason2866, 2023-08-26:
But it is not optimal,since ArduinoIDE still does not provide a 32MB partition scheme!
So you can currently use only 16MB. If there would be a 32 MB partition scheme you still can't use the full 32 MB, since `esptool.py` can't flash more than 16MB on a S3.
The `esptool.py` bug hits ArduinoIDE and Platformio! It is used from both "under the hood".

VojtechBartoska, 2024-02-20:
Closing this as expired, if needed you can reopen.

closed 2024-02-20 by VojtechBartoska