espressif/arduino-esp32 #8547: HTTPClient/src/HTTPClient.h does not support ipv6

emmby, 2023-08-21:
### Board

esp32

### Device Description

adafruit esp32 feather v2

### Hardware Configuration

no

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

not sure

### PSRAM enabled

yes

### Upload speed

115200

### Description

HTTPClient works fine for IPv4, but does not work for IPv6
esp_http_client.h works fine on IPv6, but other modules built on top of HTTPClient (eg. HTTPUpdater) won't work

See https://arduino.stackexchange.com/questions/93628/internet-in-esp32-is-not-working-with-ipv6

### Sketch

```cpp
N/A
```

### Debug Message

```plain
-1 connection refused
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-08-22:
IPv6 support is not supported in core 2.0.x. It is planned for core 3.0.x

VojtechBartoska, 2023-08-22:
https://github.com/espressif/arduino-esp32/pull/7722

VojtechBartoska, 2023-08-22:
Closing as duplicate

closed 2023-08-22 by VojtechBartoska