espressif/arduino-esp32 #8548: Wifi error

savagerex, 2023-08-22:
### Board

ESP32 Wrover e

### Device Description

Sensor - iisdwb
external flash

### Hardware Configuration

NONE

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

First i start Ble
then i use task to connect wifi.

wifi always disconnect.

below is error message.

### Sketch

```cpp
too huge
```

### Debug Message

```plain
please connect Wifi
E (18886) wifi:Expected to init 4 rx buffer, actual is 1
Connecting to WiFi..
please connect Wifi
Connecting to WiFi..
Connecting to WiFi..
please connect Wifi
Connecting to WiFi..
Connecting to WiFi..
please connect Wifi
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-22:
@savagerex - Please provide a minimum sketch that can be use to replicate the issue and that demonstrates that the issue is due to a failure in the ESP32 Arduino Core code.

Arduino has an example that demonstrate that it works using BLE and WiFi at the same time:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiBlueToothSwitch/WiFiBlueToothSwitch.ino

The issue you have presented may be just a problem of lack of RAM  necessary to run all the layer and the application together.

SuGlider, 2023-08-22:
@savagerex - There is a related issue. Please read it : https://github.com/espressif/arduino-esp32/issues/8436

VojtechBartoska, 2023-10-04:
closing this as answered in mentioned issue above.

closed 2023-10-04 by VojtechBartoska