espressif/arduino-esp32 #8553: The Arduino Nano ESP32 lacks of support for FileSystem Configuration (IDFGH-10914)

mathertel, 2023-08-23:
### Is your feature request related to a problem?

For the Arduino Nano ESP32 configuration there are no configuration options.
Especially the File System type and size cannot be specified.

### Describe the solution you'd like.

The board is well specified so Flash Size, Flash Mode, PSRAM Size ... is fixed.
But like with other boards the Partition Scheme should be available for configuration.

as well: "Core Debug Level" and "Erase All Flash Before Sketch Upload" is meaningful.

### Describe alternatives you've considered.

A configuration of the generic ESP32S3 Dev Module might be helpful for experts - but there are very special pin re-assignments for this board (D2-D13, A0-A7)

### Additional context.

There seems to be a default FAT FileSystem defined in the partitions somewhere.

me-no-dev, 2023-08-23:
Please submit this to [Arduino.cc's fork](https://github.com/arduino/arduino-esp32)

mathertel, 2023-08-23:
The Arduino fork doesn't have the feature issue enabled.

pillo79, 2023-08-24:
Hello @mathertel and thanks for this request. As you correctly noted, many of the generic parameters are fixed by the board hardware. In addition, some user-friendly features we have implemented on this board further limit the options that we can leave as fully customizable. 

We will try to address some of your proposals in the next core version however, stay tuned!

VojtechBartoska, 2023-08-30:
Closing as mentioned PR above have been merged.

PR #8566 (2023-08-29): Nano ESP32 file system option
## Description of Change
This addresses issues raised by several Arduino forum users about not being able to use SPIFFS-based projects on their Nano ESP32, introducing the choice of data partition file system between FAT and SPIFFS. 

2578081 properly implements (as other ESP32 targets) the "Burn Bootloader" action to mean "Erase Flash". This is required when switching between file systems to avoid mounting errors.

## Tests scenarios
Tested on the Nano ESP32 with SPIFFS_Test and FFat_Test.

## Related links
First three commits are from #8565 to avoid conflicts in [... cut]
closed 2023-08-30 by VojtechBartoska