espressif/arduino-esp32 #8554: Can't use ESP Rainmaker with VSCode & PlatformIO (Arduino framework)

mottibz, 2023-08-23:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

Windows 11
VSCode 1.81.1
PlatformIO: Core 6.1.10·Home 3.4.4
Espressif 32 Platform 5.2.0

### Hardware Configuration

Nothing attached to the board

### Version

other

### IDE Name

VSCode & PlatformIO

### Operating System

Windows 11

### Flash frequency

Standard

### PSRAM enabled

no

### Upload speed

115200

### Description

For my development on ESP32, I am using VSCode and PlatformIO with the Arduino framework.
I also installed the Arduino IDE on my machine for testing.
I tried the ESP Rainmaker Switch sample code with the Arduino IDE. It complied, loaded to my ESP32 and run with no issues.
Before that, I tried to do the same with VSCode & PlatformIO and encountered the following issues:
- The code that took about 59% Flash using the Arduino IDE, took almost 92% with VSCode (after I played with the partitions table so that enough space for the code in flash).
- Some of the functions (e.g. the included App Insignts header, RMaker.enableScenes();, initAppInsights();, and RMaker.enableSystemService(SYSTEM_SERV_FLAGS_ALL, 2, 2, 2); - where not recognized in VSCode (produced compile errors). I commented these out, compiled, and uploaded to the ESP32, but nothing happend (no output at all to the terminal and no way to provision). Looks like I am doing something wrong or there is an issue with the Arduino-ESP32 package for VSCode & PlatformIO.
Can you please help in setting Rainmaker to be successfully used with VSCode and PlatformIO?
Thanks a lot!

### Sketch

```cpp
The sketch is the sample Rainmaker Switch code that is provided with the Arduino IDE ESP32 package
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Different code changes that did not help

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-08-23:
Hello, not sure if we will be able to help, no one from our Developers use PlatformIO IDE

SuGlider, 2023-08-23:
It is also in PIO Portal: https://community.platformio.org/t/setting-up-vscode-and-platformio-to-support-development-with-esp-rainmaker/35388

mottibz, 2023-08-23:
Vojtech: But Espressif also developed the ESP32 support for PlatformIO, isn't it? It is strange that Rainmaker looks like working in Arduino IDE and not in PlatformIO. Do you have any suggestion how to get this resolved? Thanks a lot!

SuGlider: Yes, I also put it in the PIO Portal but no answer yet.

SuGlider, 2023-08-23:
I think that I have seen some other issues questioning about the PIO Binary Flash size.
I guess that PIO may use different compilation/linking options when compared to the Arduino IDE .

With reagrd to RainMaker header files, it may also be an issue related to including the necessary folders in the compiler command line. 

I don't use PIO, therefore I can't help much.
Another way to go using VSCode would be creating an Arduino as Component project with RainMaker.
Not sure how easy it would be, but if necessary I have a repository with the general folder structure:

https://github.com/SuGlider/RainMaker_as_Arduino_Component  

This project is built using `idf.py build`. 
I use to built it manually, but I think that there is an IDF Extension for VSCode that may help and automate it.

SuGlider, 2023-08-23:
>But Espressif also developed the ESP32 support for PlatformIO, isn't it? It is strange that Rainmaker looks like working in Arduino IDE and not in PlatformIO.

Actually our Espressif Arduino Team only supports 2 building modes: Arduino IDE and Arduino as Component.
PIO is supported by PIO Team. They maintain all the tools, binary blobs, packages and scripts.
We just supply the releases for them.

mottibz, 2023-08-25:
Hello, I sent the PlatformIO guys a request for help. Waiting for their resolution. In the mean time, I am using the Arduino IDE which works OK. Will update when I get a solution. Waiting...

Jason2866, 2023-08-26:
Don't use such a old version `Espressif 32 Platform 5.2.0`
Try with 6.3.2 or even better use development platform. There are many changes since 5.2.0

VojtechBartoska, 2023-10-02:
There is not much to be done from our side, closing the issue.

closed 2023-10-02 by VojtechBartoska