espressif/arduino-esp32 #8555: ESPNOW CHANNEL SETTING CONFLICT WITH WIFI CONNECTION

GodayMax, 2023-08-23:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32

### Hardware Configuration

N/A

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

macOS 12

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have two of 「ESP32 Dev Module」communiting each other with ESPNOW ,  works like a charm.
these nodes running at channel 6.

THEN

1、pair one of there nodes to wifi [RUNNING AT CHANNEL 11], so this node will change its channel to 11
2、espnow fails beacause channel 6 cannot talk to channel 11

we cannot ask all of our customers to set their wifi to channel 6
so guys any practical solution?

thanks~

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-08-23:
The radio can be on one channel at a time, so if you use STA, ESP-NOW will be on that same channel. Same goes for AP when you have STA connected. If you need to be on two frequencies at the same time, you need two radios.

closed 2023-08-23 by me-no-dev

GodayMax, 2023-08-23:
I wonder if you guys have some tricky ways to avoid or to bypass this situation?

for example to let the other node know to change its channel or something.

GodayMax, 2023-08-23:
@me-no-dev @VojtechBartoska