espressif/arduino-esp32 #8556: esp_ble_gap_config_ext_adv_data_raw error

kensherman, 2023-08-23:
### Board

ESP32 C3

### Device Description

adafruit c3 development board

### Hardware Configuration

pin 20 and 21 connected to a gps

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to send an Bluetooth Extended Advertisement that changes and is not static.

### Sketch

```cpp
int transmit_ble2(uint8_t *ble_message,int length) {

  esp_err_t  ble_status;
  bool status;
  static int advertising = 0; 
  int advertno = 0;

  if(advertising)
  {
    Serial.println("BLE Stop Advert");
    const uint8_t inst[] = {0x00};
    status = advert.stop(0, inst);
    if(status){
      Serial.println("BLE Stop Advert Success");
    }
    status = advert.remove(0);
    if(status){
      Serial.println("BLE Stop Advert Remove Success");
    }
    status = advert.clear();
    // if(status){
    //   Serial.println("BLE Stop Advert Clear Success");
    // }
    advertising = 0;
    delay(1000);
  }
    delay(1000);
    Serial.println("BLE Start Advert");
    Serial.print("Message len:");
    Serial.println(length, DEC);
    advert.setAdvertisingParams(advertno, &ext_adv_params_1M);
    ble_status = esp_ble_gap_config_ext_adv_data_raw(0, length, ble_message);
    Serial.println(ble_status);
    //status = advert.setAdvertisingData(advertno, length, &ble_message[0]);
    if(ble_status){
      Serial.println("BLE Start Advert Data");
    }
    advert.setDuration(0);
    advert.setScanRspData(advertno, sizeof(raw_scan_rsp_data_coded), &raw_scan_rsp_data_coded[0]);
    advert.setInstanceAddress(advertno, addr_1m);
    status = advert.start(1,advertno);
    if(status){
      Serial.println("BLE Start Advert Success");
    }
    advertising = 1;
  
 
  //advert.start(1,0);

  Serial.println("BLE Transmitting");

  //return (int) ble_status;
  return 0;
}
```

### Debug Message

```plain
E (89916) BT_BTM: LE EA SetAdvData: cmd err=0x12
from this line
 ble_status = esp_ble_gap_config_ext_adv_data_raw(0, length, ble_message);
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-09-01:
It seems imcomplete.
What is the content of `ble_message`?

SuGlider, 2023-09-01:
Please check this example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE5_multi_advertising/BLE5_multi_advertising.ino

SuGlider, 2023-09-01:
More BLE 5 examples can be found at https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples

SuGlider, 2023-09-01:
Example:
``` cpp

uint8_t addr_1m[6] = {0xc0, 0xde, 0x52, 0x00, 0x00, 0x01};

esp_ble_gap_ext_adv_params_t ext_adv_params_1M = {
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE,
    .interval_min = 0x30,
    .interval_max = 0x30,
    .channel_map = ADV_CHNL_ALL,
    .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy = ESP_BLE_GAP_PHY_1M,
    .max_skip = 0,
    .secondary_phy = ESP_BLE_GAP_PHY_1M,
    .sid = 0,
    .scan_req_notif = false,
    .own_addr_type = BLE_ADDR_TYPE_RANDOM,
    .tx_power = EXT_ADV_TX_PWR_NO_PREFERENCE,
};

static uint8_t raw_adv_data_1m[] = {
        0x02, 0x01, 0x06,
        0x02, 0x0a, 0xeb,
        0x11, 0x09, 'E', 'S', 'P', '_', 'M', 'U', 'L', 'T', 'I', '_', 'A',
        'D', 'V', '_', '1', 'M'
};

// BLE 5 ADV Code:

    // 1M phy extend adv, Connectable advertising
    esp_ble_gap_ext_adv_set_params(0, &ext_adv_params_1M);
    esp_ble_gap_ext_adv_set_rand_addr(0, addr_1m);
    esp_ble_gap_config_ext_adv_data_raw(0, sizeof(raw_adv_data_1m), &raw_adv_data_1m[0]);
```

SuGlider, 2023-09-01:
You can also check this IDF example:

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble_50/multi-adv/main/multi_adv_demo.c

kensherman, 2023-09-02:
It was a configuration issue on my part. 
I needed to set up two adverts with two different adv_params , broadcast and update

closed 2023-09-02 by kensherman

JustDaveIII, 2023-09-28:
Ken, Is it possible to get the updated code posted here or somehow? I am experiencing the same problem as you - and writing the code for the same reason as you - I can tell this because I see parts of the original code base that we are both using.