espressif/arduino-esp32 #8558: Static IP connection only works on 1.0.6 version

phphengenharia, 2023-08-23:
### Board

ESP-WROOM-32

### Device Description

I tried on multiple chips.

### Hardware Configuration

Nothing is attached to board.

### Version

v2.0.11

### IDE Name

Arduino IDE 2.1.1

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

The attached code only works on 1.0.6 version. Why cannot get a static IP connection using 2.0.x versions? I am using CISCO AIR-CAP1602I-T-K9 acess point.

### Sketch

```cpp
#include <WiFi.h>
 
const char* ssid = "TCEGO-AUTOMACAO";
const char* password =  "password";
 
	IPAddress staticIP(172, 17, 86, 220);
	IPAddress gateway(172, 17, 86, 254);
	IPAddress subnet(255, 255, 255, 0);
	IPAddress dns(8, 8, 8, 8);
    
void setup(){
  Serial.begin(115200);
 
  if (WiFi.config(staticIP, gateway, subnet, dns, dns) == false) {
    Serial.println("Configuration failed.");
  }
    
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print("Connecting...\n\n");
  }
 
  Serial.print("Local IP: ");
  Serial.println(WiFi.localIP());
  Serial.print("Subnet Mask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway IP: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("DNS 1: ");
  Serial.println(WiFi.dnsIP(0));
  Serial.print("DNS 2: ");
  Serial.println(WiFi.dnsIP(1));

}
    
void loop(){}
```

### Debug Message

```plain
It only works on 1.0.6:

Local IP: 172.17.86.220
Subnet Mask: 255.255.255.0
Gateway IP: 172.17.86.254
DNS 1: 8.8.8.8
DNS 2: 8.8.8.8
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

phphengenharia, 2023-08-23:
With 2.0.11, no compiler error but it doesn't connect:

Connecting...

Connecting...

Connecting...

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Connecting...

Connecting...

Connecting...

Connecting...

Connecting...

Connecting...

Connecting...

Connecting...

Connecting...

SuGlider, 2023-08-24:
@phphengenharia - I just tested it and it works fine.
Please make sure that the SSID/PWD is correct, including using the precise upper/lower case symbols.

In order to help seeing WiFi debug messages, please turn the `Core Debug Level` to `"Debug"` in the Arduino IDE Tool/Board Menu.

When the SSID has just 1 letter in the wrong case, it won't connect.
You can use the WiFi Scan Arduino example to check the right SSID to connect.
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino

phphengenharia, 2023-08-24:
I am using the same code on two versions. It connects quickly using 1.0.6 version.

I get those events on 2.0.11 version:

[ 10028][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: TCEGO-AUTOMACAO, BSSID: 00:00:00:00:00:00, Reason: 201
[ 10029][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10037][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 10044][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
Connecting...

Connecting...

[ 14912][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: TCEGO-AUTOMACAO, BSSID: 00:00:00:00:00:00, Reason: 201
[ 14913][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 14921][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 14928][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
Connecting...

SuGlider, 2023-08-24:
>[ 2580][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: TCEGO-AUTOMACAO, BSSID: 00:00:00:00:00:00, Reason: 201
[ 2581][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 2589][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND

This messages (Verbose Debugging) indicate clearly that the SSID is not correct. 
As I said it must be in the right case (`TCEGO-AUTOMACAO` is not the same as `TCEGO-Automacao`).

List the SSID in your WLAN using the ESP32 with WiFi Scan example.
 https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino

phphengenharia, 2023-08-24:
The SSID e password are both correct. I am using exactly the same code on two versions.

I used that WiFiScan.ino for tests too:

Scan start
[ 71033][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 137, Status: 0, Results: 6
[ 71034][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
Scan done
6 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 | TCEGO-CORPORATIVA                |  -68 |  1 | WPA2-EAP
 2 | TCEGO-AUTOMACAO                  |  -68 |  1 | WPA
 3 | TCEGO-PUBLIC                     |  -69 |  1 | open
 4 | TCEGO-TVS                        |  -69 |  1 | WPA+WPA2
 5 | TCEGO-TELEFONIA                  |  -69 |  1 | WPA
 6 | DIRECT-seHL-L6402DW_BR5d77       |  -70 | 11 | WPA2

phphengenharia, 2023-08-24:
I don't know why it is working so fine on 1.0.x version but not on 2.0.x:

[image]

[image]

lbernstone, 2023-08-24:
Newer version do not allow connections to WPA by default.
https://github.com/espressif/arduino-esp32/blob/master/docs/source/troubleshooting.rst#wi-fi

phphengenharia, 2023-08-24:
I can understand now. It connected normally with "WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK);". 

Thanks you!

mrengineer7777, 2023-08-24:
@phphengenharia For your own safety I suggest changing your network option "Layer 2 Security" to WPA2.  Using `WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK);` is a reasonable workaround for backwards compatibility.

closed 2023-08-24 by mrengineer7777