espressif/arduino-esp32 #8560: digitalPinToInterrupt doesn't accept pin 48

sparvanders, 2023-08-24:
### Board

bpi_leaf_s3, and other ESP32-S2 and ESP-S3 variants

### Device Description

Not relevant

### Hardware Configuration

A pushbutton on IO48.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Any

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

5000000

### Description

In many variants for ESP32S2 and ESP32S3, digitalPinToInterrupt is defined as:

#define digitalPinToInterrupt(p)    (((p)<48)?(p):-1)
But pin 48 is also valid as interrupt source.
Without a fix, this crashes:

attachInterrupt(digitalPinToInterrupt(48), Callback, FALLING);

The line should read:
#define digitalPinToInterrupt(p)    (((p)<=48)?(p):-1)

Furthermore, returning -1 doesn't play nice with attachInterrupt() that accepts an uint8_t. attachInterrupt lacks a bounds check for array access so this gives undefined behavior (for me, a crash).

### Sketch

```cpp
// Not a working sketch

void IRAM_ATTR Callback() {
}

void setup() {
  //This does not work:
  attachInterrupt(digitalPinToInterrupt(48), Callback, FALLING);
  //This works:
  attachInterrupt(48, Callback, FALLING);
}
```

### Debug Message

```plain
Crash
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-24:
@sparvanders - Thanks for reporting this bug! It shall be fixed as you wrote.

VojtechBartoska, 2023-09-06:
For S3 SoCs, issue is solved in [2.0.12 release](https://github.com/espressif/arduino-esp32/releases/tag/2.0.12). I am closing this issue, for other boards this will be addressed in 3.X release.

PR #8726 (2023-10-06): Master5.1 s3 gpio48 and platform txt review
## Description of Change
This PR follows #8600 and #8562 changes for all necessary variant files / boards.txt.
It also fixes the #8560 issue in the Arduino Core 3.0.0
The PR moves common macros and defines to `Arduino.h`, not necessary in `pins_arduino.h` file anymore.
It also adds NEO_LED macros and constants in order to allow using it in a `digitalWrite()` for Blinking as in the example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/BlinkRGB/BlinkRGB.ino

## Tests scenarios
Passed boards CI https://github.com/ [... cut]
closed 2023-09-06 by VojtechBartoska