espressif/arduino-esp32 #8568: esp_wifi_set_channel does not work when a device is connected to the SoftAP

Cancro29, 2023-08-26:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module (ESP32-D0WD-V3) , 4 MB of SPI Flash

### Hardware Configuration

SSD1306 attached to GND,3.3, GPIO 22, GPIO 21

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm unable to change WiFi channel using esp_wifi_set_channel() when a device is connected to the SoftAP. Serial monitor is just showing ESP_FAIL. When I disconnect the device, esp_wifi_set_channel() works as intended

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 2   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED

// Set these to your desired credentials.
const char *ssid = "1";
const char *password = "12345678";
unsigned long ch_change = 0;

WiFiServer server(80);

extern "C" {
#include "esp_wifi.h"
  esp_err_t esp_wifi_set_channel(uint8_t primary, wifi_second_chan_t second);
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.mode(WIFI_AP); 
  WiFi.softAP(ssid, password,1);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }

  if (millis() - ch_change >= 2000){
      int ch = random(1,13);

      esp_err_t result = esp_wifi_set_channel(ch, WIFI_SECOND_CHAN_NONE);
      if (result == ESP_OK) {
        Serial.println("Channel changed to " + String(ch));
      }
      else {
        Serial.println(result);
        Serial.println(esp_err_to_name(result));
        Serial.print(" while changing to channel " + String(ch));
      }
      ch_change = millis();
    }
}
```

### Debug Message

```plain
07:17:14.301 -> ESP_FAIL
07:17:14.301 ->  while changing to channel 9-1
```

### Other Steps to Reproduce

This issue happens in all ESP32 based board

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-26:
Any connected devices will need to reconnect to the new channel, so I don't think this is an error.  You are best off stopping the AP (alerting the clients), then making the change.

Cancro29, 2023-08-26:
> Any connected devices will need to reconnect to the new channel, so I don't think this is in error. You are best off stopping the AP (alerting the clients), then making the change.

Thank you for your reply Ibernstone.
I understand that changing channels can cause clients to disconnect so they need to reconnect. But if I change the channel quickly enough (in milliseconds timespan) and then return to original channel, the clients will still work as usual. This is what I did on my project using ESP8266 board. But in ESP32, this behavior is preventing me to do that. I don't know if this is a bug if not as in ESP8266 wifi_set_channel() works even if a device is connected to the SoftAP

SuGlider, 2023-08-26:
The explanation about it can be found in the documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_channel7uint8_t18wifi_second_chan_t

[image]

Cancro29, 2023-08-26:
Thank you for your explanation SuGlider. I will close this.

closed 2023-08-26 by Cancro29