espressif/arduino-esp32 #8571: SPIFFS mount mount failed, -10025

fatsheep2020, 2023-08-27:
### Board

ESP32S3BOX

### Device Description

coding in main:
==================
#include <Arduino.h>
#include <SPIFFS.h>

void setup()
{
  Serial.begin(115200);
  if(SPIFFS.begin()) {
     Serial.println("ffs mount success");
  }
}
==================
debug result:
E (888301) SPIFFS: mount failed, -10025
[888388][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1

### Hardware Configuration

ESP32-S3-WROOM-1-N8R8

### Version

latest master (checkout manually)

### IDE Name

VSCode +PlatformIO

### Operating System

Ubuntu

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

coding in main:
==================
#include <Arduino.h>
#include <SPIFFS.h>

void setup()
{
  Serial.begin(9600);
  if(SPIFFS.begin()) {
     Serial.println("ffs mount success");
  }
}
==================

### Sketch

```cpp
code in main.cpp:
#include <Arduino.h>
#include <SPIFFS.h>

void setup()
{
  Serial.begin(9600);
  if(SPIFFS.begin()) {
     Serial.println("ffs mount success");
  }
}
```

### Debug Message

```plain
debug result:
E (888301) SPIFFS: mount failed, -10025
[888388][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-27:
ESP32-S3-WROOM-1 is quad flash (qpi)
ESP32-S3-WROOM-2 is octal flash (opi)
8MB PSRAM modules are octal
2 & 4MB PSRAM modules are quad

Jason2866, 2023-08-27:
As @lbernstone wrote, this error happens when flash/psram type is not configured correctly to the module used.

HareshPrajapati, 2023-08-28:
look into [this](https://community.platformio.org/t/vscode-esp-idf-framework-esp32-spiffs-error/10817)

VojtechBartoska, 2023-08-28:
Hello, I think this can be closed.

closed 2023-08-28 by VojtechBartoska

fatsheep2020, 2023-08-28:
> look into [this](https://community.platformio.org/t/vscode-esp-idf-framework-esp32-spiffs-error/10817)

i check the link and add csv file as the following, but the test result still failed. is the csv file not correct? it come from 
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_8MB.csv
[image]

Jason2866, 2023-08-29:
This [boards.json](https://github.com/platformio/platform-espressif32/blob/develop/boards/lilygo-t-display-s3.json) meets exactly what your need for your S3Box.

fatsheep2020, 2023-08-30:
> This [boards.json](https://github.com/platformio/platform-espressif32/blob/develop/boards/lilygo-t-display-s3.json) meets exactly what your need for your S3Box. Do not specify a partition scheme here. The boards Definition includes a correct one.

Thanks! but how to use this json file?

fatsheep2020, 2023-08-30:
i'm fix this issue! actually not csv configure or json configure. the root cause is a spiffs.begin(), not put parameter. it should be:
spiffs.begin(true), then the file system will be format, and file can be create/write/read.