espressif/arduino-esp32 #8573: HardwareSerial::end() won't detach RX/TX from IO MUX when it is executed isolated

SuGlider, 2023-08-28:
Using, for instance, ESP32S3 GPIO 43 and/or GPIO44 for OneWire:

GPIOs 43 and 44 are default UART pins.
The issue seems related to being able to detach the GIPOs 43 and 44 from IO Mux and UART.
The execution of `Serial0.end()` shall do it, but I read the code again and realized that because, it has never been initialized, it also doesn't run the code to detach the IO MUX.

Therefore, @teastainGit, please try you code in this way:

``` cpp
void setup() {
  // Serial0 is the UART0 when the S3 CDC is enabled
  Serial0.begin(115200); // this will initialize the UART with GPIOs 43 and 44, marking those pins to be detached
  Serial0.end(); // this will detach GPIOs 43 and 44
  
  // code as usual...
  Serial.begin();
  delay(200);
  Serial.println(" setup ");
  delay(200);
  sensors.begin();
}
```

_Originally posted by @SuGlider in https://github.com/espressif/arduino-esp32/issues/8324#issuecomment-1695543739_

VojtechBartoska, 2023-08-28:
**Task update:**
- should be handled also by Peripheral Manager

teastainGit, 2023-08-28:
So, thanks guys for all your support, but it still does not 'work' !
Shown below is my LilyGO LilyGO T-Display S3 breadboard test setup, with the DS18B20 'onewire'.
[image]
I uploaded your sample code and as before it did not return the temperature:
I get 13:56:36.242 -> -127.00ºC
When I use pin1 in the sketch and move it over to the test position to the right with the coloured wires I get:
(pin1 is yellow in both examples)
14:10:31.317 -> 24.62ºC
Both a TMP102 I2C temp sensor and a I2C SHTx sensor work on both positions!
[observation] when uploading your suggested sketch the tiny LED on the DS18B20 flickers, in a way that looks like the visual representation of serial info.
Neither 43 or 44 work with the onewire device.
I hope this extra info is helpful.
-Terry

closed 2023-10-06 by me-no-dev

teastainGit, 2023-10-07:
The final solution was to edit the helper header file in OneWire library and change to <33 pin number limit maximum to any number.

GPIO 43 and 44 now work with the dallas semiconductor temperature sensor!

I notified the authors of OneWire with An Issue and offered to make a PR, but no response.

thanks for all of your help, it is resolved.
-Terry