espressif/arduino-esp32 #8576: Missing source files "class/hid/hid.h" && "class/hid/hid_device.h"

DoganM95, 2023-08-28:
### Board

esp32 doit dev board

### Device Description

Doit v1 modole with a cp2102 onboard

### Hardware Configuration

None, only usb cable attached.

### Version

latest master (checkout manually)

### IDE Name

VSCode (PlatformIO)

### Operating System

Windows 11

### Flash frequency

115200

### PSRAM enabled

yes

### Upload speed

115200

### Description

```
#include errors detected. Please update your includePath. IntelliSense features for this translation unit
(C:\Users\User\OneDrive\Documents\PlatformIO\Projects\IoT-Win-Waker\src\main.cpp) will be provided 
by the Tag Parser.C/C++(1696) 
cannot open source file "class/hid/hid.h" (dependency of "USBHIDKeyboard.h")C/C++(1696)
```

The main issue here seems, that some files are missing. Jumping into "USBHIDKeyboard.h", the line
`#include  "USBHID.h"` is marked with the error 
```
#include errors detected. Please update your includePath. IntelliSense features for this translation unit (C:\Users\User\OneDrive\Documents\PlatformIO\Projects\IoT-Win-Waker\src\main.cpp) will be provided by the Tag Parser.C/C++(1696)
cannot open source file "class/hid/hid.h" (dependency of "USBHID.h")C/C++(1696)
```

Then, jumping into  "USBHID.h" source, the following two lines
```
#include "class/hid/hid.h"
#include "class/hid/hid_device.h"
```

are showing the errors

```
cannot open source file "class/hid/hid.h"C/C++(1696)
cannot open source file "class/hid/hid_device.h"C/C++(1696)
```

Summed up, the last two files seem missing in the whole library, please correct me if i am wrong and thanks for any help.

### Sketch

```cpp
#define CONFIG_TINYUSB_HID_ENABLED 1
#define ARDUINO_USB_MODE 1
#define CFG_TUD_HID 1

#include "USB.h"
#include "USBHIDKeyboard.h"

USBHIDKeyboard Keyboard;

setup(){...}...
```
```

### Debug Message

```plain
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Blynk @ 1.3.0
|-- AsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.0.0
|-- USB @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\BlynkRpcCRC8.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\BlynkRpcClientWeakImpl.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\BlynkRpcInfra.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\BlynkRpcInfraUart.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\BlynkRpcUartFraming.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\MessageBuffer.c.o
Compiling .pio\build\esp32dev\lib4a3\BlynkNcpDriver\MessageWriter.c.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\libed6\Blynk\utility\BlynkDebug.cpp.o
Compiling .pio\build\esp32dev\libed6\Blynk\utility\BlynkHandlers.cpp.o
Compiling .pio\build\esp32dev\libed6\Blynk\utility\BlynkHelpers.cpp.o
Compiling .pio\build\esp32dev\libed6\Blynk\utility\BlynkTimeUtils.cpp.o
Compiling .pio\build\esp32dev\libed6\Blynk\utility\BlynkTimer.cpp.o
Compiling .pio\build\esp32dev\libed6\Blynk\utility\utility.cpp.o
Compiling .pio\build\esp32dev\lib561\AsyncTCP-esphome\AsyncTCP.cpp.o
In file included from C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/USB/src/USBHIDKeyboard.h:24,
                 from src/main.cpp:9:
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/USB/src/USBHID.h:22:10: fatal error: class/hid/hid.h: No such file or directory
 #include "class/hid/hid.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp32dev\lib843\FS\FS.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
```

### Other Steps to Reproduce

- Install PlatformIO extension in VSCode
- Set up a new project with the following settings:
```
platform: espressif32
board: esp32dev
framework: arduino
```
- Paste the exmaple code, e.g. https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/Keyboard/KeyboardSerial/KeyboardSerial.ino
- Build it using platformIO

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-08-28:
esp32 does not have native USB.  Those files are available for variants that have such (S2/S3).

DoganM95, 2023-08-29:
Alright, thanks for the hint!
But lets assume i plugged in an esp32-s2, which supports it.
The same issue would persist.

closed 2023-09-20 by DoganM95