espressif/arduino-esp32 #8588: New sketch Compilation error in Mac (Arduino IDE + ESP 32)

Allwingit, 2023-08-30:
### Board

ESP32 Dev Board 

### Device Description

No external devices attached

### Hardware Configuration

No Configuration done

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Mac OS 10.15.2

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

92160MHz

### Description

Can't compile a new sketch without any code added.

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
bash: line 1:  1824 Segmentation fault: 11  "/Users/allwins/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1"/esptool --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB -o "/private/var/folders/y6/s32mrsz52j35qslyhj05lf3w0000gn/T/arduino/sketches/DA42FA776494081623186D35034D1879"/sketch_aug30b.ino.bootloader.bin "/Users/allwins/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32"/bin/bootloader_qio_40m.elf

exit status 139

Compilation error: exit status 139
```

### Other Steps to Reproduce

- Installed Arduino IDE latest version
- Open IDE & Create new sketch
- Configure board & USB port
- Arduino -> Preference . added this data additional board manager url : https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
- Installed Latest ESP32 2.0.11 from board manager 
- Restarted Arduino IDE
- Getting above mentioned error 

Occurrence rate: 100%

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-08-31:
Mac OS 10.15 is not supported

Allwingit, 2023-08-31:
Hi @me-no-dev ,

Thanks for the update. Will it be fine if we use Mac OS 12 ?

Jason2866, 2023-08-31:
The Github Actions CI test are running under macOS 12.6.7

Allwingit, 2023-08-31:
@Jason2866 , Thank you. I will try it in macOS 12.6.8

Allwingit, 2023-08-31:
Working fine in mac OS 12.6.8. Hence closing this issue.