espressif/arduino-esp32 #8593: Can't update LittleFS via nothing.

zekageri, 2023-09-01:
### Related area

LittleFS

### Hardware specification

not depends

### Is your feature request related to a problem?

Currently there is no possibility to update LittleFS.

### Describe the solution you'd like

**I would imagine something like this**

`HTTPUpdateResult HTTPUpdate::updateLittleFS(WiFiClient& client, const String& url, const char *partitionLabel = "littlefs", const String& currentVersion, HTTPUpdateRequestCB requestCB)`

So we could pick which partition to update too.

### Describe alternatives you've considered

There is no alternatives. Currently i can only update my files one by one. This is not ideal.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

atanisoft, 2023-09-01:
You should be able to use `Updater.updateSpiffs(...)` to have it download and update LittleFS partitions, they are handled the same as SPIFFS (even partition type!).

zekageri, 2023-09-01:
So it should update my file system if it can download a firmware.elf file? Why does it say that it is full?
I have a custom partition table and i want to update only one of my partitions.

zekageri, 2023-09-01:
Oooohhh, maybe the "Build Filesystem Image" builds it for the full size and my esp gets the full firmware.elf but wants to update that one partition and the full elf file does not fit inside....

is this possible?

atanisoft, 2023-09-01:
firmware.elf wouldn't ever reach the ESP32, it's not in a usable format. You would need to send the firmware.bin file instead.

The Build FS image target in PIO is just to package the "data" folder (or whatever you might name it) into a bin file that would be uploaded to the filesystem partition.

lbernstone, 2023-09-01:
Not sure how you send an elf file as a filesystem image.  The filename in the arduino plugin is sketchname.spiffs.bin.  I'm sure the pio team uses a similar naming scheme.  You can also directly build the image file using [mklittlefs](https://github.com/espressif/arduino-esp32/blob/master/package/package_esp32_index.template.json#L415).

zekageri, 2023-09-01:
I can build a file system image via pio too. But i dont know where it puts it now... will search for it

lbernstone, 2023-09-01:
bytes 9-16 in a littlefs image should be "littlefs"

zekageri, 2023-09-04:
Okay, the littlefs.bin is at the same dir as the firmware.bin.
It is in pio/build/esp-wrover-kit/littlefs.bin

The problem is that if i update it, it says that i have no space left.

```cpp
WiFiClientSecure _wifi_client;
Updater.updateSpiffs(_wifi_client, url)
```

```
[Updater] - Starting file system update
[286314][V][HTTPClient.cpp:252] beginInternal(): url: https://@raw.githubusercontent.com/{name}/{repo}/main/littlefs.bin
[286328][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: 
raw.githubusercontent.com port: 443 url: /{name}/{repo}/main/littlefs.bin
[286793][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 0

[286793][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 169996
[286797][V][ssl_client.cpp:68] start_ssl_client(): Starting socket      
[286826][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[286827][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[286831][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[286839][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[286847][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...
[287456][V][ssl_client.cpp:290] start_ssl_client(): Verifying peer X.509 certificate...
[287457][V][ssl_client.cpp:298] start_ssl_client(): Certificate verified.
[287460][V][ssl_client.cpp:313] start_ssl_client(): Free internal heap after TLS 128324
[287468][D][HTTPClient.cpp:1170] connect():  connected to raw.githubusercontent.com:443
[287476][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 596 bytes...
[287796][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[287797][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Connection: close'
[287800][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Length: 7192576'
[287807][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Cache-Control: max-age=300'
[287815][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline'; sandbox'  
[287828][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Type: application/octet-stream'
[287837][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'ETag: "d4ec9fb85f52cb1bbdf110230a94e864b35f2daf8fc86612a0f755bfef90f3eb"'
[287849][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Strict-Transport-Security: max-age=31536000'
[287858][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Content-Type-Options: nosniff'
[287866][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Frame-Options: deny'
[287874][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-XSS-Protection: 1; mode=block'
[287882][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-GitHub-Request-Id: A0DA:0E62:1369C99:1444488:64F5754B'
[287892][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
[287900][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Date: Mon, 04 Sep 2023 06:24:27 GMT'
[287908][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Via: 1.1 varnish'
[287915][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Served-By: cache-vie6335-VIE'
[287924][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Cache: HIT'
[287930][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Cache-Hits: 0'
[287937][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Timer: S1693808667.095831,VS0,VE174'
[287946][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Vary: Authorization,Accept-Encoding,Origin'
[287955][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Access-Control-Allow-Origin: *'
[287964][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Cross-Origin-Resource-Policy: cross-origin'
[287973][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Fastly-Request-ID: 4e0d317a38357613ef17527783be3534b78aeb22'
[287984][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Expires: Mon, 04 Sep 2023 06:29:27 GMT'
[287993][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Source-Age: 0'
[287999][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: ''
[288005][D][HTTPClient.cpp:1321] handleHeaderResponse(): code: 200
[288011][D][HTTPClient.cpp:1324] handleHeaderResponse(): size: 7192576  
[288017][D][HTTPClient.cpp:642] sendRequest(): sendRequest code=200     

[288023][D][HTTPUpdate.cpp:241] handleUpdate(): Header read fin.

[288029][D][HTTPUpdate.cpp:242] handleUpdate(): Server header:

[288035][D][HTTPUpdate.cpp:243] handleUpdate():  - code: 200

[288041][D][HTTPUpdate.cpp:244] handleUpdate():  - len: 7192576

[288046][D][HTTPUpdate.cpp:250] handleUpdate(): ESP32 info:

[288052][D][HTTPUpdate.cpp:251] handleUpdate():  - free Space: 2097152

[288058][D][HTTPUpdate.cpp:252] handleUpdate():  - current Sketch Size: 
1369136

[288305][E][HTTPUpdate.cpp:270] handleUpdate(): spiffsSize to low (5242880) needed: 7192576

[288311][D][HTTPClient.cpp:395] disconnect(): tcp stop
[288315][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[Updater] - HTTP_UPDATE_FAILED Error (-100): Not Enough space
[Updater] - File system update was unsuccessful
ets Jul 29 2019 12:21:46
```

And for some reason i can't update my firmware.bin either.
It says that everything is ok, and when it reboots i got the last firmware and not the new.

zekageri, 2023-09-04:
I think my problem relies in that the updater searches for the first occourence of spiffs partition.
I want to update my second partition labeled as "system" for me.

My "user" partition is roughly 5mb and the updater sees that, i assume from the log

`[602663][E][HTTPUpdate.cpp:270] handleUpdate(): spiffsSize to low (5242880) needed: 7192576`

I have these two partitions

```
user,	    data,	spiffs,     0x410000,   0x500000,   #5mb
system,	    data,	spiffs,     0x910000,   0x6E9000,   #6.9mb
```

The funny thing in that is that the platformIO "Upload Filesystem Image" command uploads my data folder into my "system" partition ( so to the last ) but the http updater wants to upload it to my first spiffs partition.

How to select partition?

zekageri, 2023-09-04:
Here it is in the HTTPUpdate.cpp starting at line  262

```cpp
if(spiffs) {
  const esp_partition_t* _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);
  if(!_partition){
      _lastError = HTTP_UE_NO_PARTITION;
      return HTTP_UPDATE_FAILED;
  }
  
  if(len > _partition->size) {
      log_e("spiffsSize to low (%d) needed: %d\n", _partition->size, len);
      startUpdate = false;
  }
}
```

zekageri, 2023-09-04:
I have changed the call `esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);` in HTTPUPdate.cpp to `esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "system");` so it will update the partition with the label "system" and now it says that it is too large instead of too low.

```cpp
[Updater] - HTTP update process started
[152097][D][HTTPUpdate.cpp:306] handleUpdate(): runUpdate spiffs...

[152097][E][Updater.cpp:157] begin(): too large 7192576 > 5242880
[152098][E][HTTPUpdate.cpp:411] runUpdate(): Update.begin failed! (Bad Size Given)

[152105][E][HTTPUpdate.cpp:359] handleUpdate(): Update failed
```

What is happening

zekageri, 2023-09-04:
### Okay so it seems to work in the following ways

In `HTTPUpdate.cpp` at the method `HTTPUpdateResult HTTPUpdate::handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs, HTTPUpdateRequestCB requestCB)` i had to change this line `const esp_partition_t* _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);` to `const esp_partition_t* _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "system");`  ( "system" as my partition label ) and in the method `bool HTTPUpdate::runUpdate(Stream& in, uint32_t size, String md5, int command)` i had to change this line `if(!Update.begin(size, command, _ledPin, _ledOn)) {` to `if(!Update.begin(size, command, _ledPin, _ledOn,"system")) {`  ( "system" as my partition label ) and it seems to work. My firmware.bin does get's downloaded but for some reason it does not get replaced on reboot. Will investigate it further. And i have to test the littlefs updater too. ( it downloads but i did not confirmed if it updates it too yet )

zekageri, 2023-09-04:
The best solution would be for us to be able to pass a partition label to the `handleUpdate` method in the HTTPUpdate.cpp

zekageri, 2023-09-04:
I have opened a new feature request here: https://github.com/espressif/arduino-esp32/issues/8598 so we could track my most recent findings.

closed 2023-09-04 by zekageri