espressif/arduino-esp32 #8596: How to include Custom ESP-IDF in Arduino-esp32?

spilz87, 2023-09-03:
### Board

Esp32c3

### Device Description

Custom board esp32c3

### Hardware Configuration

Esp32c3 

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Because of some issue with deep sleep, GPIO wake up and internal resistors, I found the lines I want to change in ESP32 IDF. 

I would like to try this modification but I don’t know the steps to include this modification into my arduino-ESP32 

Someone knows how to do it ?
do I need to compile esp IDF then to include in arduino-ESP32 folder ?

Thanks in advance for your help

### Sketch

```cpp
None
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-09-03:
You can use the [arduino lib-builder](https://github.com/espressif/esp32-arduino-lib-builder). Make sure to pull the 4.4 release
https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

SuGlider, 2023-09-03:
The second way to do it is by using Arduino as IDF Component in an IDF project.

SuGlider, 2023-09-03:
You can also verify if IDF 5.1 fit your needs.
There is an Arduino Core branch based on IDF 5.1 at https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs

spilz87, 2023-09-03:
> You can use the [arduino lib-builder](https://github.com/espressif/esp32-arduino-lib-builder). Make sure to pull the 4.4 release
> https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Thanks, I’m going to try it tomorrow

spilz87, 2023-09-03:
> The second way to do it is by using Arduino as IDF Component in an IDF project.

All my projects are on Arduino for now, so I will stay like this, but good to know 👍

spilz87, 2023-09-03:
> You can also verify if IDF 5.1 fit your needs.
> There is an Arduino Core branch based on IDF 5.1 at https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs

I checked, IDF 5.1 has the same issue with deep sleep mode. 

So I will first try with 4.4 as I know the réussit to compare, then I will try 5.1 later

spilz87, 2023-09-04:
> You can use the [arduino lib-builder](https://github.com/espressif/esp32-arduino-lib-builder). Make sure to pull the 4.4 release https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Hello
I'm sorry for my dummy question : 

I need to modify a file of the IDF (`/espressif/esp-idf/tree/release/v4.4/components/esp_hw_support/sleep_modes.c` to be precise).
So I guess I have to compile it again or to do something to be able to use it 
and then I have to include it in arduino-esp32
and then I have to use arduino lib-builder to include all of it arduino environement 

is it these ?
I will need some help, I'm a little lost with this mix of all these different environnement :(

spilz87, 2023-09-04:
I created a fork with the modification : [v4.4 Spilz87](https://github.com/spilz87/esp-idf/tree/release/v4.4)

If I understand well, I may directly use **arduino lib-builder** with `build.sh` and add parameters like `-i` ou `-I` to link to my fork.
is it true ?
I don't really get how to add this parameters (what should it looks like) neither where it's going to be installed 

thanks for your help

spilz87, 2023-09-04:
> Once it has cloned IDF into the Lib Builder folder, you may change the IDF file (sleep_modes.c) and build it again in order to create the libraries and Arduino sdk structure.

I can not directly point to my fork ?

SuGlider, 2023-09-04:
>I can not directly point to my fork ?

You may link it, but the Lib Builder script will try, at the first run, to clone IDF from the Espressif repository.
If it is already there, I think it may just leave it as it is... but I'm not sure.

SuGlider, 2023-09-04:
>I copy the result into my Arduino15 folder -> which folder ? it will earse other "normal" arduino-esp folder so ?

For the version 2.0.x, the folder shall be Arduino15/packages/esp32/hardware/esp32/2.0.x/tools/sdk/

SuGlider, 2023-09-04:
>I run again ./build.sh to include the modification (it will compile with the local files by default ?)

Yes, it should build the whole `sdk/` folder structure using whatever is in the Lib Builder folder, including the components, which includes the Arduino component folder.

SuGlider, 2023-09-04:
>I modify the file I want in the IDF files -> it say "Your branch is up to date with 'origin/idf-release/v4.4" but I don't find where it download the files

Maybe it is using the folder from the current IDF installation.
It will be found in the IDF_PATH environment variable.

spilz87, 2023-09-04:
> > I modify the file I want in the IDF files -> it say "Your branch is up to date with 'origin/idf-release/v4.4" but I don't find where it download the files
> 
> I think it shall be under Lib Builder folder.... but it haven't run it for a year or so, thus it may have changed.

I'm in the folder  **esp32-arduino-lib-builder** where is `build.sh` but I don't find the download files in any folder around :(

SuGlider, 2023-09-04:
you can use Windows `set`command or Linux `env | grep IDF` command

spilz87, 2023-09-04:
I got this : 

```
your branch is up to date with 'origin/idf-release/v4.4'.
warning: redirecting to https://github.com/espressif/arduino-esp32.git/
remote: Enumerating objects: 117, done.
```

I feel like as it's the last release, it directly download arduino-esp32 files (that I find in esp32_arduin_lib_builder folder) instead of doing everything from idf v4.4

spilz87, 2023-09-04:
[image]
when I run `build.sh` I get this error about MSYS2 and PATH.
I tryed to install and different way to add it, I'm out of idea :(

SuGlider, 2023-09-04:
I think that it won't work from Windows .
I only use Linux or MacOS.

SuGlider, 2023-09-04:
With Win11, you can also use Windows Subsystem for Linux (WSL)

spilz87, 2023-09-05:
Just be carefull to run `./build.sh -A idf-release/v4.4 -I release/v4.4 -i v4.4.5` to be able to compile with the good version of IDF

SuGlider, 2023-09-05:
@spilz87 - I have just installed the `esp32-arduino-lib-builder` using Ubuntu WSL, from scratch, using these steps:
```
sudo apt-get update
sudo apt-get install git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
cd ~
git clone https://github.com/espressif/esp32-arduino-lib-builder.git
cd esp32-arduino-lib-builder/
git checkout release/v4.4
./build.sh
====> at this point, it may fail with some python error... in this case just execute the following commands:
cd esp-idf
./install.sh
. ./export.sh
idf.py --version              ====> shall report 4.4.5 when Arduino Component is on Master (2.0.12)
cd ..
./build.sh
==> here all python 3.9 packages shall be installed and the build process shall move forwar with no problems.
==> All SoC libraries shall be built in the folder ~/esp32-arduino-lib-builder/build/
```

After all is executed (it takes some time), you should have this folder: ~/esp32-arduino-lib-builder/esp-idf/
This folder has the IDF code used to build the Arduino Libraries. You can change the `sleep_modes.c` file there.

After changing IDF code as desired, you may run `./build.sh` again.
The final /sdk folder used in Arduino IDE can be found at 

~/esp32-arduino-lib-builder/out/tools/sdk/
This should replace Arduino15/packages/esp32/hardware/esp32/2.0.12/tools/sdk/ files (Arduino IDE installation folder).

You can also check the distribuiton tar.gz files at ~/esp32-arduino-lib-builder/dist

SuGlider, 2023-09-05:
> > After that, you can copy those file to the Arduino15 folder or your sketchbook board folder (when using the git checkout from master to create a sketchbook board in the Arduino IDE).
> 
> @SuGlider How do you put in your sketchbook board folder ? ( I guess it help to not impact other projects as Arduino 15 folder will do, right ?)

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation

In a Nutshell:
- go to you arduino sketch folder (in windows it usually is Documents/Arduino) 
- create this folder strucuture: ./hardware/espressif
- git clone ESP32 Arduino into esp32 folder
- go to ./hardware/espressif/esp32/tools/ folder and execute `get.exe` to download the Toolchains.
- restart your Arduino IDE and you will find a new Sketchbook Board in IDE->Menu->Tools

VojtechBartoska, 2023-10-04:
Closing this as answered.

closed 2023-10-04 by VojtechBartoska