espressif/arduino-esp32 #8597: ESP IDF(Arduino as component)not identifying BluetoothSerial.h  (IDFGH-10958)

Abhay-2412, 2023-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

it should be able to identify Arduino library "BluetoothSerial.h" and should be able to complie the code

### What is the actual behavior?

 error: 'BluetoothSerial' does not name a type
 BluetoothSerial SerialBT;

### Steps to reproduce.

i had followed documentation by espressif and added arduino as component
in sdkconfig file i have changed CONFIG_FREERTOS_HZ to 1000
i had run the command idf.py menuconfig and selected "Start with loop and serup function"
After that i had wrote a simple classic bluetooth code and its not at all identifying it

### Build or installation Logs.

```plain
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:7:1: error: 'BluetoothSerial' does not name a type
 BluetoothSerial SerialBT;
 ^~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp: In function 'void connectToBluetooth()':
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:19:9: error: 'SerialBT' was not declared in this scope
     if (SerialBT.available())
         ^~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:19:9: note: suggested alternative: 'Serial2'
     if (SerialBT.available())
         ^~~~~~~~
         Serial2
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:47:7: error: 'connectToWiFi' was not declared in this scope
       connectToWiFi(ssid.c_str(), password.c_str());
       ^~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp: In function 'void setup()':
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:89:1: error: 'SerialBT' was not declared in this scope
 SerialBT.begin("ESP32_BT_1");
 ^~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:89:1: note: suggested alternative: 'Serial2'
 SerialBT.begin("ESP32_BT_1");
 ^~~~~~~~
 Serial2
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp: In function 'void loop()':
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:98:5: error: 'SerialBT' was not declared in this scope
     SerialBT.println("Enter credentials");
     ^~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:98:5: note: suggested alternative: 'Serial2'
     SerialBT.println("Enter credentials");
     ^~~~~~~~
     Serial2
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:105:5: error: 'SerialBT' was not declared in this scope
     SerialBT.println("Connected successfully");
     ^~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:105:5: note: suggested alternative: 'Serial2'
     SerialBT.println("Connected successfully");
     ^~~~~~~~
     Serial2
```

### More Information.

please help me through it as soon as possible

lbernstone, 2023-09-03:
Are you using the [sdkconfig from arduino-esp32](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig)?
BluetoothSerial is dependent on a couple of those [config settings](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.h#L20).

Abhay-2412, 2023-09-08:
yep i figured out in menuconfig i enabled bluetooth option and chose classical bluetooth code is working fine

closed 2023-09-08 by Abhay-2412