espressif/arduino-esp32 #8598: Allow user to pass partition label to the handleUpdate method of the HTTPUpdate class.

zekageri, 2023-09-04:
### Related area

HTTPUpdate

### Hardware specification

Every esp32

### Is your feature request related to a problem?

I'm facing an issue where i have custom partitions and i want to update only one of them with HTTPUpdate.
My idea is that i want a "user" partition and a "system" partition.
The system partition would hold the web related files, the user partition would hold user configuration and things like that.
On a firmware update, the esp would download a new firmware.bin and a new littlefs.bin but it would spaw only my system partition so the user configuration will remain intact.

### Describe the solution you'd like

Currently there is no possibility to update only one particular partition. The HTTPUpdater picks the last partition labeled as "spiffs" and wants to put the downloaded bin file to this partition.

It would be good if we could pass a partition label to updateSpiffs method.

Current: `Updater.updateSpiffs(_wifi_client, url)`
Possible: `Updater.updateSpiffs(_wifi_client, url,"system")`

### Describe alternatives you've considered

Alternatively i'm doing this right now:

In **HTTPUpdate.cpp** at line **183** there is a method **handleUpdate**. 
In this method i replaced this line ` const esp_partition_t* _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL); ` with this line: ` const esp_partition_t* _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "system"); ` So it will search for my custom partition. 

After that still in **HTTPUpdate.cpp** at line **398** there is a method **runUpdate**. 
In this method i replaced this line `if(!Update.begin(size, command, _ledPin, _ledOn)) { ` with this line: ` if(!Update.begin(size, command, _ledPin, _ledOn,"system")) { ` so i passed my custom partition to the begin method of the Update class.

Now if in my sketch if i call `Updater.updateSpiffs(_wifi_client, url);` it will update my custom partition.

### Additional context

Right now it says that it downloads both of my firmware.bin and littlefs.bin but i see no change after rebooting.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

zekageri, 2024-01-25:
Can this be implemented? I just updated the core and i had to do these changes again.

lbernstone, 2024-01-25:
Can you submit a PR with your changes?
I recommend not using a partition for your read-only files, but rather including them in your firmware as C-string arrays.  You can use `xxd -i` to convert files in the file system, and script a method to include them as header files. [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/OTAWebUpdater/html.h#L38)

lbernstone, 2024-01-25:
Ok.  Whoever has to support this is going to looove you.

zekageri, 2024-01-25:
Well, it was already a bad design from the start.
If we can create multiple partitions why can't we update them individually from the start?
I mean, the option is there but the implementation left it out.

lbernstone, 2024-01-25:
I don't mean the feature request, which seems like a fine idea, but your system, where the html app is going to be able to get out of sync with the firmware version.

zekageri, 2024-01-25:
Ah it cant. With each update it downloads both files at once from a server :D

zekageri, 2024-01-25:
Each version comes with a backend and a frontend.

lbernstone, 2024-01-25:
How about if we supercede the updateSpiffs idea, and if you call HTTPUpdate(const char* partition), it will search for that partition, attempt to determine the partition type, and call the appropriate Update?

zekageri, 2024-01-25:
That would be nice

zekageri, 2024-01-25:
That way we dont have to modify the existing functions

zekageri, 2024-01-25:
It isnt even spiffs anymore so

lbernstone, 2024-01-25:
Looks like where I am headed is making a completely alternate route in Update::begin, with a command called U_AUTO, which will use the label name. That way it is a non-breaking change.  If it is good enough at the end of the day, it may be able to be a full replacement.

zekageri, 2024-01-25:
I had to specify the label at two different places in order to work. Will check on it tomorrow

zekageri, 2024-01-26:
`updateSpiffs(WiFiClient& client, const String& url, const String& currentVersion, HTTPUpdateRequestCB requestCB)` calls `handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs, HTTPUpdateRequestCB requestCB)` which then calls `runUpdate(Stream& in, uint32_t size, String md5, int command)`. I don't see what you see in here.

zekageri, 2024-01-26:
If we create a new function like `update(HTTPClient& http,const char* partitionLabel,HTTPUpdateRequestCB requestCB)` we still need to call `handleUpdate` and `runUpdate` ?!

zekageri, 2024-01-26:
What i can see here is that we must modify the handleUpdate and runUpdate methods like this

```cpp
t_httpUpdate_return handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs = false, const char* partitionLabel = NULL, HTTPUpdateRequestCB requestCB = NULL);
bool runUpdate(Stream& in, uint32_t size, String md5, int command = U_FLASH, const char* partitionLabel = NULL,);
```

That way we can make a method which a user can call like this

```cpp
HTTPUpdateResult HTTPUpdate::updatePartition(WiFiClient& client, const String& url, const char* partitionLabel, HTTPUpdateRequestCB requestCB)
{
    HTTPClient http;
    if(!http.begin(client, url))
    {
        return HTTP_UPDATE_FAILED;
    }
    return handleUpdate(http, currentVersion, true, partitionLabel, requestCB);
}
```

zekageri, 2024-01-26:
This would looke like that:

**Methods**
```cpp
t_httpUpdate_return handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs = false, const char* partitionLabel = NULL, HTTPUpdateRequestCB requestCB = NULL);
bool runUpdate(Stream& in, uint32_t size, String md5, int command = U_FLASH, const char* partitionLabel = NULL);
HTTPUpdateResult updatePartition(WiFiClient& client, const String& url, const char* partitionLabel, HTTPUpdateRequestCB requestCB);
```

**updatePartition**

```cpp
HTTPUpdateResult HTTPUpdate::updatePartition(WiFiClient& client, const String& url, const char* partitionLabel, HTTPUpdateRequestCB requestCB)
{
    HTTPClient http;
    if(!http.begin(client, url))
    {
        return HTTP_UPDATE_FAILED;
    }
    return handleUpdate(http, currentVersion, true, partitionLabel, requestCB);
}
```

**handleUpdate**

```cpp
/**
 *
 * @param http HTTPClient *
 * @param currentVersion const char *
 * @return HTTPUpdateResult
 */
HTTPUpdateResult HTTPUpdate::handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs, const char* partitionLabel = NULL, HTTPUpdateRequestCB requestCB)
{

    HTTPUpdateResult ret = HTTP_UPDATE_FAILED;

    // use HTTP/1.0 for update since the update handler not support any transfer Encoding
    http.useHTTP10(true);
    http.setTimeout(_httpClientTimeout);
    http.setFollowRedirects(_followRedirects);
    http.setUserAgent("ESP32-http-Update");
    http.addHeader("Cache-Control", "no-cache");
    http.addHeader("x-ESP32-STA-MAC", WiFi.macAddress());
    http.addHeader("x-ESP32-AP-MAC", WiFi.softAPmacAddress());
    http.addHeader("x-ESP32-free-space", String(ESP.getFreeSketchSpace()));
    http.addHeader("x-ESP32-sketch-size", String(ESP.getSketchSize()));
    String sketchMD5 = ESP.getSketchMD5();
    if(sketchMD5.length() != 0) {
        http.addHeader("x-ESP32-sketch-md5", sketchMD5);
    }
    // Add also a SHA256
    String sketchSHA256 = getSketchSHA256();
    if(sketchSHA256.length() != 0) {
      http.addHeader("x-ESP32-sketch-sha256", sketchSHA256);
    }
    http.addHeader("x-ESP32-chip-size", String(ESP.getFlashChipSize()));
    http.addHeader("x-ESP32-sdk-version", ESP.getSdkVersion());

    if(spiffs) {
        http.addHeader("x-ESP32-mode", "spiffs");
    } else {
        http.addHeader("x-ESP32-mode", "sketch");
    }

    if(currentVersion && currentVersion[0] != 0x00) {
        http.addHeader("x-ESP32-version", currentVersion);
    }
    if (requestCB) {
        requestCB(&http);
[... 222 lines cut ...]
// To do: the SHA256 could be checked if the server sends it

    if(Update.writeStream(in) != size) {
        _lastError = Update.getError();
        Update.printError(error);
        error.trim(); // remove line ending
        log_e("Update.writeStream failed! (%s)\n", error.c_str());
        return false;
    }

    if (_cbProgress) {
        _cbProgress(size, size);
    }

    if(!Update.end()) {
        _lastError = Update.getError();
        Update.printError(error);
        error.trim(); // remove line ending
        log_e("Update.end failed! (%s)\n", error.c_str());
        return false;
    }

    return true;
}
```

zekageri, 2024-01-26:
The problem with this is that in the rest of the `handleUpdate` calls we have to modify from that

`handleUpdate(http, currentVersion, true, requestCB);`

to that

`handleUpdate(http, currentVersion, true, NULL, requestCB);`

closed 2025-01-15 by Parsaabasi