espressif/arduino-esp32 #8601: OTA update has no effect after reboot.

zekageri, 2023-09-05:
### Board

ESP32 Wrover

### Device Description

I2C,Ethernet,PSRAM

### Hardware Configuration

Nothing serious.

### Version

latest master (checkout manually)

### IDE Name

Platformio

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have managed to do OTA update from Github releases.
I plan to update both firmware.bin and littlefs.bin from Github private repo.

Downloads are going fine, the logs says success on both firmware and littlefs download.

After reboot, i did not get the new sketch. Confirmed both several times.

For the file system partition update, i had to modify several things in HTTPUpdate.cpp in order to work with custom partitions.
I wanted to update only a specific partition labeled as "system" for me.

When my ESP restarts, it prints the current version. If i modify this version string and the update is success, i should see the updated version number. But after a reboot, i see the previous version number printed.

### Sketch

```cpp
#include <Arduino.h>
#include <Update.h>
#include <HTTPUpdate.h>

WiFiClientSecure _wifi_client;

void begin(){
_wifi_client.setCACert(github_certificate);

    Updater.rebootOnUpdate(false);
    Updater.onStart([this]() {
        updating = true;
        Serial.printf("[Updater] - HTTP update process started\n");
    });
    Updater.onEnd([this]() {
        updating = false;
        Serial.printf("\n[Updater] - HTTP update process finished\n");
    });
    Updater.onProgress([this](int currentlyReceiced, int totalBytes) {
        Serial.printf("\r[Updater] - Progress: %.2f %%", 100.0 * currentlyReceiced / totalBytes);
    });
    Updater.onError([this](int err) {
        updating = false;
        Serial.printf("[Updater] - HTTP update fatal error code %d\n", err);
    });
    Updater.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
}

void processDownload() {
    if (debug) {
        logger.write("[Updater] - Starting firmware update\n");
    }
    auto result = updateFirmware(FIRMWARE_URL);
    if (debug) {
        printUpdateResult(result);
        Serial.printf("[Updater] - Firmware update was %s\n", result == HTTP_UPDATE_OK ? "successful" : "unsuccessful");
    }

    if (result == HTTP_UPDATE_OK && debug) {
        Serial.printf("[Updater] - Starting file system update\n");
    }
    if( result != HTTP_UPDATE_OK ) { return; }
    

    result = updateFileSystem(FS_URL);
    if (debug) {
        printUpdateResult(result);
        Serial.printf("[Updater] - File system update was %s\n", result == HTTP_UPDATE_OK ? "successful" : "unsuccessful");
    }

    if (result == HTTP_UPDATE_OK) {
        if( debug ){ Serial.printf("[Updater] - Rebooting...\n"); }
        ESP.restart();
    }
}

HTTPUpdateResult updateFileSystem(String url) {
    return Updater.updateSpiffs(_wifi_client, url);
}

HTTPUpdateResult updateFirmware(String url) {
    _wifi_client.setInsecure();  // Running OOM when using second certificate
    return Updater.update(_wifi_client, url);
}

void printUpdateResult(HTTPUpdateResult result) {
    switch (result) {
        case HTTP_UPDATE_FAILED:
            Serial.printf("[Updater] - HTTP_UPDATE_FAILED Error (%d): %s\n", Updater.getLastError(), Updater.getLastErrorString().c_str());
            break;
        case HTTP_UPDATE_NO_UPDATES:
            Serial.printf("[Updater] - HTTP_UPDATE_NO_UPDATES\n");
            break;
        case HTTP_UPDATE_OK:
            Serial.printf("[Updater] - HTTP_UPDATE_OK\n");
            break;
    }
}
```

### Debug Message

```plain
[Updater] - Found version is v0.0.35, Current version is v0.0.34      
[Updater] - New version is available.
[Updater] - Starting firmware update
[Updater] - HTTP update process started
[Updater] - Progress: 68.49 %[Events] - Client connected!
[Updater] - Progress: 100.00 %
[Updater] - HTTP update process finished
[Updater] - HTTP_UPDATE_OK
[Updater] - Firmware update was successful
[Updater] - Starting file system update
[Updater] - HTTP update process started
[Updater] - Progress: 100.00 %
[Updater] - HTTP update process finished
[Updater] - HTTP_UPDATE_OK
[Updater] - File system update was successful
[Updater] - Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zekageri, 2023-09-05:
### Here is a detailed log with -DCORE_DEBUG_LEVEL=5

```cpp
[Updater] - New version is available.
[Updater] - Starting firmware update
[ 80685][V][HTTPClient.cpp:252] beginInternal(): url: https://@raw.githubusercontent.com/{user}/{repo}/main/firmware.bin
[ 80700][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: raw.githubusercontent.com port: 443 url: /{user}/{repo}/main/firmware.bin
[ 81615][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 0

[ 81628][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 170204
[ 81628][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[ 81657][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[ 81658][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS 
structure...
[ 81661][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[ 81670][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[ 81678][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS 
handshake...
[ 82290][V][ssl_client.cpp:290] start_ssl_client(): Verifying peer X.509 certificate...
[ 82290][V][ssl_client.cpp:298] start_ssl_client(): Certificate verified.  
[ 82293][V][ssl_client.cpp:313] start_ssl_client(): Free internal heap after TLS 128728
[ 82301][D][HTTPClient.cpp:1170] connect():  connected to raw.githubusercontent.com:443
[ 82309][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 
596 bytes...
[ 82630][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'HTTP/1.1 200 
OK'
[ 82631][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Connection: close'
[ 82634][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Length: 1368016'
[ 82641][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Cache-Control: max-age=300'
[ 82649][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline'; sandbox'
[ 82662][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Type: application/octet-stream'
[ 82671][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'ETag: "d9cb3fbbc22b84dc7c5bb3b155cb39b266e88116d94843271b0391b0a61da176"'
[ 82683][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Strict-Transport-Security: max-age=31536000'
[ 82692][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Content-Type-Options: nosniff'
[ 82700][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Frame-Options: deny'
[ 82708][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-XSS-Protection: 1; mode=block'
[ 82716][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-GitHub-Request-Id: 8D08:0C8D:DEC977:E8E356:64F58BB4'
[ 82726][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
[ 82734][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Date: Mon, 04 Sep 2023 08:00:04 GMT'
[ 82742][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Via: 1.1 varnish'
[ 82749][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Served-By: 
cache-vie6359-VIE'
[ 82757][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Cache: HIT'[ 82764][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Cache-Hits: 1'
[ 82771][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Timer: S1693814405.594440,VS0,VE179'
[ 82780][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Vary: Authorization,Accept-Encoding,Origin'
[ 82789][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Access-Control-Allow-Origin: *'
[ 82798][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Cross-Origin-Resource-Policy: cross-origin'
[ 82807][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'X-Fastly-Request-ID: 28d19b38ef1c9026ebc1cb3b03dab27486619840'
[ 82818][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Expires: Mon, 04 Sep 2023 08:05:04 GMT'
[ 82827][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Source-Age: 0'
[ 82833][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: ''
[ 82839][D][HTTPClient.cpp:1321] handleHeaderResponse(): code: 200
[ 82845][D][HTTPClient.cpp:1324] handleHeaderResponse(): size: 1368016
[ 82851][D][HTTPClient.cpp:642] sendRequest(): sendRequest code=200        

[ 82857][D][HTTPUpdate.cpp:241] handleUpdate(): Header read fin.

[ 82863][D][HTTPUpdate.cpp:242] handleUpdate(): Server header:

[ 82869][D][HTTPUpdate.cpp:243] handleUpdate():  - code: 200

[ 82874][D][HTTPUpdate.cpp:244] handleUpdate():  - len: 1368016

[ 82880][D][HTTPUpdate.cpp:250] handleUpdate(): ESP32 info:

[ 82886][D][HTTPUpdate.cpp:251] handleUpdate():  - free Space: 2097152     

[ 82892][D][HTTPUpdate.cpp:252] handleUpdate():  - current Sketch Size: 1368016

[Updater] - HTTP update process started
[... 68 lines cut ...]
[106106][D][HTTPUpdate.cpp:243] handleUpdate():  - code: 200

[106111][D][HTTPUpdate.cpp:244] handleUpdate():  - len: 7192576

[106117][D][HTTPUpdate.cpp:250] handleUpdate(): ESP32 info:

[106123][D][HTTPUpdate.cpp:251] handleUpdate():  - free Space: 2097152     

[106129][D][HTTPUpdate.cpp:252] handleUpdate():  - current Sketch Size: 1368016

[Updater] - HTTP update process started
[106477][D][HTTPUpdate.cpp:306] handleUpdate(): runUpdate spiffs...

[Updat[175915][D][HTTPUpdate.cpp:346] handleUpdate(): Update ok

[175917][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.[175918][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[Updater] - Progress: 100.00 %
[Updater] - HTTP update process finished
[175935][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[175941][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.[Updater] - HTTP_UPDATE_OK
[Updater] - File system update was successful
[Updater] - Rebooting...
```

zekageri, 2023-09-05:
I can download these files with curl and they are legit.

```curl
curl https://{token}@raw.githubusercontent.com/{username}/{repo}/main/firmware.bin --output C:\Users\Pc\OneDrive\Dokumentumok\test2\firmware.bin
```

This curl is putting the firmware.bin into the specififed directory. The github server is good, the files are good.

zekageri, 2023-09-05:
I have a custom partition table but it should work since i did not modify the app0 and app1 partitions

```csv
# Name,	    Type,	SubType,    Offset,	Size,	 Flags
nvs,        data,	nvs,	    0x9000,	0x5000,
otadata,    data,	ota,        0xE000,	0x2000,
app0,	    app,	ota_0,	    0x10000,	0x200000,
app1,	    app,	ota_1,	    0x210000,	0x200000,
user,	    data,	spiffs,     0x410000,   0x500000,
system,	    data,	spiffs,     0x910000,   0x6DC000,
coredump,   data,	coredump,   0xFEC000,	0x10000,
```

zekageri, 2023-09-06:
I just can't figure out what is happening.

lbernstone, 2023-09-06:
It's not updating either partition?  The SPIFFS update should happen immediately.  If you can SPIFFS.end() and then begin again, you should immediately see changes in the contents.  The firmware update may be failing for some reason, and in that case OTA will fail back to the old (valid) image.

zekageri, 2023-09-06:
Yeah, weird. The log says that everything is okay. But it does not update either.

zekageri, 2023-09-06:
Even more strange that my sketch indeed boots from another partition. I confirmed this by printing the boot partition address. And it is changing between updates.

zekageri, 2023-09-07:
I tested more and more and the littlefs is updating fine now. I did nothing unusual.
But my sketch does not update.

zekageri, 2023-09-07:
So i found out that after update, my sketch is rolling back to a previous version.

Before update

```cpp
********** Firmware Info **********
[SYSTEM] - Firmware version: v0.0.45
[SYSTEM] - Serial baud rate: 115200
[SYSTEM] - Partition address: 65536
[SYSTEM] - Partition size: 2097152
********** ****** **********
```

After update:

```cpp
********** Firmware Info **********
[SYSTEM] - Firmware version: v0.0.42
[SYSTEM] - Serial baud rate: 115200
[SYSTEM] - Partition address: 2162688
[SYSTEM] - Partition size: 2097152
********** ****** **********
```

Something is really reversed.

zekageri, 2023-09-07:
Now i try to hijack the verifyRollbacklater() function and see what comes out!

```cpp
/*
 *   This function is used to overwrite espressif's verifyRollbackLater() function.
 *   The espressif's version is returning false, and performing a verify before the
 *       app can start.
 */
extern "C" bool verifyRollbackLater() {
    return true;
}
```

Now i do not verify my firmware but i verify it myself after i tested unverifyed with this

```cpp
void Sys::verifyFirmware() {
    if (debug) {
        Serial.printf("[SYSTEM] - Verifying firmware\n");
    }
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            if (esp_ota_mark_app_valid_cancel_rollback() == ESP_OK) {
                if (debug) {
                    Serial.printf("[SYSTEM] - Firmware verified. Rollback disabled.\n");
                }
            } else {
                if (debug) {
                    Serial.printf("[SYSTEM] - Failed to disable rollback\n");
                }
            }
        } else {
            if (debug) {
                Serial.printf("[SYSTEM] - OTA partition has no record in OTA data\n");
            }
        }
    }
}
```

zekageri, 2023-09-07:
So nothing happens if i do not verify the firmware

```cpp
[Updater] - Checking for update
[Updater] - Found version is v0.0.49, Current version is v0.0.48
[Updater] - New version is available.
[Updater] - Updating Started. 1/2
[Updater] - Stage 1/2 Progress: 100.00 %
[Updater] - Update finished
[Updater] - HTTP_UPDATE_OK
[Updater] - Updating Started. 2/2
[Updater] - Stage 2/2 Progress: 100.00 %
[Updater] - Update finished
[Updater] - HTTP_UPDATE_OK
[Updater] - Rebooting...
[Config] - Creating configuration file...
[FileSystem] - (createPath) Path already exists: /config.json
[Config] - Configuration file saved...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00    
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

********** Firmware Info **********
[SYSTEM] - Firmware version: v0.0.48
[SYSTEM] - Serial baud rate: 115200
[SYSTEM] - Partition address: 2162688
[SYSTEM] - Partition size: 2097152
********** ****** **********
```

zekageri, 2023-09-07:
**Starting manual verify test ( after 10 sec of running )**

```cpp
********** Firmware Info **********
[SYSTEM] - Firmware version: v0.0.49
[SYSTEM] - Serial baud rate: 115200
[SYSTEM] - Partition address: 2162688
[SYSTEM] - Partition size: 2097152
********** ****** **********

[SYSTEM] - Verifying firmware
[SYSTEM] - Firmware verified. Rollback disabled.
[Updater] - Checking for update
[Updater] - Found version is v0.0.50, Current version is v0.0.49
[Updater] - New version is available.
[Updater] - Updating Started. 1/2
[Updater] - Stage 1/2 Progress: 100.00 %
[Updater] - Update finished
[Updater] - HTTP_UPDATE_OK
[Updater] - Updating Started. 2/2
[Updater] - Stage 2/2 Progress: 100.00 %
[Updater] - Update finished
[Updater] - HTTP_UPDATE_OK
[Updater] - Rebooting...

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00    
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

********** Firmware Info **********
[SYSTEM] - Firmware version: v0.0.49
[SYSTEM] - Serial baud rate: 115200
[SYSTEM] - Partition address: 65536
[SYSTEM] - Partition size: 2097152
********** ****** **********

[SYSTEM] - Verifying firmware
[SYSTEM] - Firmware verified. Rollback disabled.
```

zekageri, 2023-09-07:
I have also swapped the firmware and filesystem update calls. 
Now it is downloading the littlefs.bin file first and the firmware.bin last.

But it changed nothing.
Manual firmware verify changed nothing.

zekageri, 2023-09-07:
```cpp
esp_ota_img_states_t ota_state;
if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
  Serial.printf(
      "[SYSTEM] - OTA state: %s\n",
      ota_state == ESP_OTA_IMG_PENDING_VERIFY? "ESP_OTA_IMG_PENDING_VERIFY" :
      ota_state == ESP_OTA_IMG_VALID? "ESP_OTA_IMG_VALID" :
      ota_state == ESP_OTA_IMG_INVALID? "ESP_OTA_IMG_INVALID" :
      ota_state == ESP_OTA_IMG_ABORTED? "ESP_OTA_IMG_ABORTED" :
      ota_state == ESP_OTA_IMG_UNDEFINED? "ESP_OTA_IMG_UNDEFINED" :
      "Unknown"
  );
}
```

It logs out this

**[SYSTEM] - OTA state: ESP_OTA_IMG_UNDEFINED**

zekageri, 2023-09-08:
So it turns out that when the log is `ESP_OTA_IMG_UNDEFINED` that means that i have uploaded via cable.
If i upload it via http the state is `ESP_OTA_IMG_PENDING_VERIFY`

And i do verify it after 10 sec of running but it just does not update.

zekageri, 2023-09-08:
If i restart the esp before it can verify the "new" firmware it saays that `ESP_OTA_IMG_VALID` even if i did not verify.

zekageri, 2023-09-08:
I have tried to update with raw http instead of Updater class but it fails just like the Updater.

```cpp
boolean HsH_Updater::updateFirmware(){
    HTTPClient http;
    http.setReuse(false);
    http.begin(_wifi_client,FIRMWARE_URL);
    int httpCode = http.GET();
    if ( httpCode <= 0 || httpCode != HTTP_CODE_OK ) {
        firmwareUpdateEnd("Failed to start firmware update.",true,http);
        return false;
    }
    int contentLen = http.getSize();
    if (!Update.begin(contentLen)) {
        firmwareUpdateEnd(Update.errorString(),true,http);
        return false;
    }

    updating = true;
    updateStage++;
    Serial.printf("[Updater] - Updating Started. %d/2\n",updateStage);

    size_t written = Update.writeStream(_wifi_client);
    if (written != contentLen) {
        firmwareUpdateEnd(Update.errorString(),true,http);
        Update.end();
        return false;
    }
    if (!Update.end()) {
        firmwareUpdateEnd(Update.errorString(),true,http);
        return false;
    }
    if (!Update.isFinished()) {
        firmwareUpdateEnd(Update.errorString(),true,http);
        return false;
    }
    firmwareUpdateEnd(Update.errorString(),false,http);
    return true;
}

void HsH_Updater::firmwareUpdateEnd(const char* errorString, boolean hasError,HTTPClient &http){
    if (hasError) {
        Serial.printf("[Updater] - Update failed: %s\n",errorString);
    }
    else {
        Serial.printf("[Updater] - Update finished\n");
    }
    updating = false;
    http.end();
}
```
It has no effect after reboot as well.

zekageri, 2023-09-08:
I can't belive no one has ever had this problem before.

closed 2023-09-12 by zekageri

joaocelson, 2024-06-30:
Hello friend, did you manage to solve this problem?