espressif/arduino-esp32 #8602: Can't receive fragmanted 802.11 packets

razgino11, 2023-09-05:
### Board

ESP32 wroom

### Device Description

Using the esp32 wroom, but I dont think it will matter which hardware

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10 / WSL / Ubuntu

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am receiving a video stream from a drone, and i noticed most of the packets were not receiving, after doing a wireshark analysis (with my network card in monitor mode) i have noticed that packets that are fragmanted on the 802.11 layer do not get received on the esp32 (presists also on esp8266 if that matters)

On the official documation it says the defragmanation is supported, but that doesn't seem to be the case at least for the arduino sdk

### Sketch

```cpp
void setup()
{    
 Serial.begin(115200);
  Serial.println();
  Serial.println(PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  //wifi_set_sleep_type(NONE_SLEEP_T); //LIGHT_SLEEP_T and MODE_SLEEP_T
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println(" connected");
}

void loop()
{
    // Receive a video packet if available
    char buffer[MAX_PACKET_SIZE] = {0};

    int packetSize = udpVideo.parsePacket();
    if (packetSize) {
        Serial.printf("Received packet of size %d from %s:%d\n \n", packetSize, udpVideo.remoteIP().toString().c_str(), udpVideo.remotePort());

        // read the packet into packetBufffer
        int n = udpVideo.read(buffer, MAX_PACKET_SIZE);
        buffer[n] = 0;
    }
```

### Debug Message

```plain
Only packets that are not fragmanted get recevied
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-09-05:
Have you tried using [asyncUDP](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPClient/AsyncUDPClient.ino)? It is better suited for fast streaming.

razgino11, 2023-09-06:
> Have you tried using [asyncUDP](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPClient/AsyncUDPClient.ino)? It is better suited for fast streaming.

Yes, i have tried writing te same thing using asyncUDP and it has the same problem, also rewrote it using the idf sdk and the same problem presists :(

SuGlider, 2023-09-21:
Have you tried it using IDF directly? Arduino Networwing is just a layer on top of IDF LwIP.

razgino11, 2023-09-21:
@SuGlider Yes, the problem presists there as well, also tried recently on the esp32-s3, the problem is there as well

SuGlider, 2023-09-21:
In that case, I'd say that this could be a current limitation of the ESP32 LwIP implementation.

Another possibility would be that it lacks RAM to assemble the package and it would need to use PSRAM.

There may be some specific configuration for it.

I'd suggest asking in the IDF Forum and/or GitHub IDF Issues.

razgino11, 2023-09-21:
@SuGlider In my opinion, the problem isn't in the LwIP implemention, since the LwIP is a tcp/ip stack, and im claiming the problem is in refragmantion on the 802.11 layer, which is below that, in my opnion the problem lies within the closed source libraries (libnet80211.a for example) that the arduino implemention that is based on idf is using

I have also opened issues on the placed you mentioned,

altought just to verify, can you guide me on what you ment, how do i make lwip use the PSRAM for refragmanation?

SuGlider, 2023-09-21:
Arduino has **IP Reassembly disabled** by default.
It could be set with menuconfig using Arduino as IDF Component, if necessary.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-ip4-reassembly

https://github.com/espressif/arduino-esp32/blob/d135e8a7e1e12160ebf68330254ecaa270266fb9/tools/sdk/esp32c3/sdkconfig#L1255-L1258

razgino11, 2023-09-21:
@SuGlider Yes Im farmilier with that, but I'm talking about something different, IP layer fragmention is not my problem, I'm talking about 802.11 layer fragmantion

SuGlider, 2023-09-21:
How do you know, using Arduino, that the problem is within 802.11 layer?!

Please provide evidences.

razgino11, 2023-09-21:
I'm using platform io, there you can write build flags to compile LWIP with reassembly at the udp level and the problem persists, further more, the packets I have issues with are not fragmented on the udp layer, they are fragmented however on the 802.11 layer, which is a different kind of fragamnation

closed 2025-01-15 by Parsaabasi