espressif/arduino-esp32 #8604: ESP32S2 native USB - serial port visible only in dfu mode.

androot2, 2023-09-06:
### Board

ESP32S2

### Device Description

D1 mini, ESP32S2

### Hardware Configuration

S2 mini board connected only to USB.

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Win7. Win10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

After uploading and restart, serial port gone. Downgrade ESP32 boards to older version (2.0.8 work for me) solve problem.

### Sketch

```cpp
Empty
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-09-07:
Hello, can you please retest with latest version 2.0.12 please?

androot2, 2023-09-07:
The same behaviour. Port doesn't exists.

hpax, 2023-09-19:
I can confirm this problem with 2.0.12 and 2.0.13.

hpax, 2023-10-01:
For what it's worth, this is not host-OS specific (I run Linux.)
The ACM port *does* work in the bootloader (forcing bootloader entry via GPIO reset.)

hpax, 2023-10-13:
2.0.14 seems to fix this problem.

jekoehler, 2023-10-30:
Same issue here.

Version:  2.0.14
OP system: macOS 13.3.1
Arduino IDE 2.2.1

A serial port connection is just available in DFU mode after pulling down GPIO0 and pressing RST button.
Uploading code via DFU mode works fine. Soft reset isn't working. Pressing RST button to get into working mode makes the device disappear from serial port.

Sometimes a hard reset results in a working connection for a few seconds, then serial port appears and disappears again for a few seconds. This will happen three or four times and then stops with no serial port connected.

jekoehler, 2023-10-31:
I tried different versions and different upload codes.

Version >2.0.8
- uploading simple code without WiFi enabled works fine
- serial port gets recognized after hard reset

Version >2.0.8
- uploading code with WiFi enabled (station and/or AP mode) works fine
- serial port gets recognized for a few milliseconds and prints out some `Serial.print` messages that are written in code before `WiFi.mode(WIFI_MODE_STA);`
- serial port connection gets lost when the `WiFi.mode(WIFI_MODE_STA);` code gets executed
- then the system is trying to connect serial port four or five times with the same behavior as before but doesn't succeed
- finally serial port gets NOT recognized after reset at all
- uploading any further code requires enabling DFU mode via (GPIO0 and RST) button press

For some unknown reason each time when WiFi is going to be established the serial port connection gets lost.

A possible cause could be insufficient power for WiFi. As far as I know MacBook COM ports provide at least 500mA which is enough to provide the required peak current of 310mA for the ESP32's WiFi connection.

closed 2025-01-15 by Parsaabasi