espressif/arduino-esp32 #8605: Esp32 - W5500 TCP Server

Impavido85, 2023-09-06:
### Board

Esp32 Dev module

### Device Description

DevKit

### Hardware Configuration

GPIO

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921000

### Description

I'm using a w5500 with the esp32 and I need to get both wifi and ethernet working, as a client and as a TCP server. I have not found any examples. If I try to use the Arduino's "Ethernet.h" library, it works as a TCP client but not as a TCP server. When compiling I get the error "virtual void Server::begin(uint16_t)" How can I do? thank you all

### Sketch

```cpp
#include <WiFi.h>
EthernetServer serverTcpEth(portServerTCP);
```

### Debug Message

```plain
error: cannot declare variable 'serverTcpEth' to be of abstract type 'EthernetServer'
note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^~~~~~~~~~~~~~
note: 	'virtual void Server::begin(uint16_t)'
     virtual void begin(uint16_t port=0) =0;
                  ^~~~~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2023-09-06:
Hello @Impavido85 

this (or similar) issue has been reported before. See [here](https://github.com/espressif/arduino-esp32/issues/6058) and [here](https://github.com/espressif/arduino-esp32/issues/3580).

Thanks
Felix

Impavido85, 2023-09-07:
Hi Felix, 
Thanks for the reply. I've read, but I haven't found any solutions.
In my specific case, I need an ethernet TCP server and it doesn't compile either.

felmue, 2023-09-07:
Hello @Impavido85 

it looks like it is in the works. See this [PR](https://github.com/espressif/arduino-esp32/pull/8575).

Thanks
Felix

VojtechBartoska, 2023-09-07:
Yes, this is something which will be investigated.

kapyaar, 2023-09-07:
@Impavido85 I saw somewhere else to change the server.h line from

virtual void begin(uint16_t port=0) =0;

to 

void begin() {};

You can give it a try and see if it works.

me-no-dev, 2023-09-07:
> Yes, this is something which will be investigated.

This is an external library that will be made obsolete with 3.0.0 and the ETH updates we have ready there

Impavido85, 2023-12-29:
Are there news? Thanks

VojtechBartoska, 2024-02-20:
Relates to https://github.com/espressif/arduino-esp32/pull/8760

me-no-dev, 2024-04-03:
W5500 is natively supported in Arduino v3.0.0 You can use it the same way you use WiFi

closed 2024-04-03 by me-no-dev