espressif/arduino-esp32 #8611: gen_esp32part.py 517：UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal

1217IGOR, 2023-09-07:
### Board

ESP32

### Device Description

ESP32 Dev Modeule

### Hardware Configuration

Haven't connected the development board yet, just compiling the code

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

windows11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I just finished building the ESP32 development environment on ArduinoIDE 2.21, I have selected the development board as ESP32 Dev Module, and used the basic example "blink" to compile and test. Then the IDE reports an error:UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal. 
And there was an error during debugging: "debug.cfg not found in scripts"

### Sketch

```cpp
void setup() {
  pinMode(5, OUTPUT);
}

void loop() {
  digitalWrite(5, HIGH);  
  delay(1000);                      
  digitalWrite(5, LOW);   
  delay(1000);                    
}
```

### Debug Message

```plain
the IDE reports an error:UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal. 
and there was an error during debugging: "debug.cfg" not found in scripts.
```

### Other Steps to Reproduce

This problem has been bothering me for two weeks now and I'm very anxious about it, so I'm hoping the big boys will help me!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

xkhannx, 2023-12-10:
Got the same problem, anyone able to fix it?

Makas-Marakas, 2024-03-04:
Hello, have you found the solution?

Unao1011, 2024-05-01:
Hello, I got the same problem. I also haven't connected the board yet, so maybe that's the issue.

lbernstone, 2024-05-01:
If this is happening during compile, then it sounds like an environment issue.  Do you have unicode characters in your home directory path, or are you building the firmware on OneDrive?

per1234, 2024-05-23:
This was fixed incidentally by https://github.com/espressif/arduino-esp32/pull/9340

This is the line of the script that produced the warning:

https://github.com/espressif/arduino-esp32/blob/5502879a5b25e5fff84a7058f448be481c0a1f73/tools/gen_esp32part.py#L517

The warning was produced when compiling sketches under the following conditions:

- The [`build.path` path passed to the "output" argument to the script](https://github.com/espressif/arduino-esp32/blob/d164df89de350aa945ccb0cb4261f56306bfdb3c/platform.txt#L156) contains unicode characters.
- The compilation is performed on a Windows machine.

The reason it was Windows-specific is because the warning is not produced by newer versions of Python ([probably >=3.0](https://docs.python.org/3.0/whatsnew/3.0.html#:~:text=8%2Dbit%20strings.-,All%20text%20is%20Unicode,-%3B%20however%20encoded%20Unicode)). On Windows, the standalone executable [`gen_esp32part.exe` is used](https://github.com/espressif/arduino-esp32/blob/d164df89de350aa945ccb0cb4261f56306bfdb3c/platform.txt#L24) and apparently the version of Python that was used to produce the version of the executable prior to the rebuild done for https://github.com/espressif/arduino-esp32/pull/9340 was one of the Python versions that produced the warning. On Linux and macOS machines, the [system Python 3.x is used to execute the script](https://github.com/espressif/arduino-esp32/blob/d164df89de350aa945ccb0cb4261f56306bfdb3c/platform.txt#L23).

Here is a minimal script that produces the warning when executed using Python 2.x (tested with 2.7.18):

```python
from __future__ import unicode_literals
import sys
sys.argv[1] == '-'
```

The warning can be safely ignored. The line in the `gen_esp32part` script that generates the warning is for a feature that allows you to set the `<output>` argument to `-` if you want the script to print the output instead of writing it to a file. That that feature is never used during the compilation process, so users of the platform don't need to worry about it at all. Even for those who use the script directly from the command line and do use the feature, there is also nothing to worry about because by its very nature there can never be a case where they will be passing a unicode character as this argument when intending to cause the script to print the output.

VojtechBartoska, 2024-05-23:
thanks for nice and extensive clarification @per1234! 

I am closing the ticket.