espressif/arduino-esp32 #8612: ESP_ARDUINO_VERSION_PATCH is still 11  should have been 12..

dagnall53, 2023-09-07:
### Board

ESP32 devboard

### Device Description

Dev board

### Hardware Configuration

nothing special!

### Version

latest master (checkout manually)

### IDE Name

2.2.1 

### Operating System

win10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Just updated ESP32 from arduino from 2.0.11 to 2.0.12.
Noted 
Failure to update **ESP_ARDUINO_VERSION_PATCH** from  11 to 
ESP_ARDUINO_VERSION_PATCH = 12   in esp_arduino_version.h

!!  
I know probably few people use this.. but I do!! 

### Sketch

```cpp
example: 
webserver.sendContent("<br><center>Compiled with:"+String(ESP_ARDUINO_VERSION_MAJOR)+"."+String(ESP_ARDUINO_VERSION_MINOR)+"."+String(ESP_ARDUINO_VERSION_PATCH)+ " (based on ESP-EDF:"+String(esp_get_idf_version())+")</center>");
```

### Debug Message

```plain
Reports: 
Compiled with:2.0.11 (based on ESP-EDF:v4.4.5)
```

### Other Steps to Reproduce

checked the source and it looks like updating was simply missed before release.. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-09-07:
> @me-no-dev Please help with triage, thanks!

nothing we can do right now, or we will need to make a new release just for that.

SuGlider, 2023-09-07:
I just installed the Arduino Core 2.0.12 and I can see that:
1- `esp_arduino_version.h` says it is version 11 `#define ESP_ARDUINO_VERSION_PATCH   11`
2- `platform.txt` has no version tag  `version=` 
3- `package.json` has `"version": "2.0.11"`

Using branch `relesase/v2x` the file `platform.txt` has `version=2.0.11`

`2.0.12` is already released... Not sure if it can be replaced by another ZIP file.
The JSON file in the release are pointing to 2.0.12. 
The ZIP files are as mentioned above.

SuGlider, 2023-09-07:
Maybe we could just update the file in order to make it have the latest version number.

SuGlider, 2023-09-16:
fixed in 2.0.13.

PR #8615 (2023-09-13): Set ref to 2.0.13
## Description of Change
Update Arduino Core version to reflect latest 2.0.12

## Tests scenarios
None

## Related links
Fixes #8612
closed 2023-09-16 by SuGlider